iNavFlight/inav #2211: Implement a proper message bus

fiam, 2017-09-26:
With the implementation of the new OSD_MESSAGES element, it has become evident that a system wide message bus would be very useful to do even more than displaying different messages in the OSD. Buzzer and LED systems would become message listeners too.

API should allow for both messages with specific duration (e.g. "Autotrim updated" or "Settings Saved" might last 2 seconds) as well as permanent messages that will be either cleared or updated by the message sender ("Unable to arm, no GPS fix" should stay until a fix is acquired).

User notification/feedback systems (OSD, buzzer, LEDs, OLED, telemetry...) should react to messages rather than polling the system status (current OSD implementation) or being explicitly called (current buzzer implementation).

digitalentity, 2017-09-26:
Cool! A "message" should be defined as:

- Message ID - mandatory
- Message Text - optional
- Beeper pattern - optional
- LED pattern - optional

digitalentity, 2017-09-26:
Other possibilities include sending messages via telemetry to allow LTM listener or Lua script to log them.

digitalentity, 2018-05-14:
@fiam ?

fiam, 2018-05-14:
It’s something useful that I’d like to get done. I might have time for this in the next cycle, I don’t think I will for 2.0.

closed 2023-03-06 by DzikuVx