iNavFlight/inav #2215: Omnibus F4 V2 glitches OSD and gyro on firmware 1.7.2-1.7.3

a6j, 2017-09-26:
The board is bought on BG.
On this board, the OSD indicators are displayed with a delay and with a low response. Appear artifacts when moving text across the screen.
All indications are displayed with a delay.
[image]
[image]
[image]
In addition, the gyro constantly swims, does not hold the horizon.
Even if the firmware is in the configuration window, this is the "full erase" check box, the controller does not blink, hangs when "erased". If you disable "full removal", the firmware is poured, but the settings are not erased. In the betaflight configurator, firmware 1.7.3 with "full erase" enabled is filled without problems.
In the versions 1.7.0-1.7.1 there are no such problems with OSD and the horizon holds normally.
Where to dig?

fiam, 2017-09-26:
@a6j All the OSD issues you mentioned have been fixed in the development branch, not sure about the gyro drift. Are you able to compile from git?

a6j, 2017-09-26:
Yes I can. If there is time, I'll try it today.

fiam, 2017-09-26:
@a6j If you're flashing INAV on new board which previously had something else (either nothing or another firmware like ßF or CF) you should enable the "Full Chip Erase" checkbox.

When upgrading INAV versions, it depends on how much has change between them. Sometimes you can upgrade without erasing. Sometimes you will need to dump your configuration, erase the whole chip and restore your configuration using the CLI. It's usually mentioned in the release notes.

a6j, 2017-09-26:
So I do, I turn on the "Full Chip Erase" checkbox and the firmware process hangs on erase.
I wrote about this above. Other controllers in the iNav configurator are flushed without problems, regardless of whether the "Full Chip Erase" checkbox is turned on or not. The only problem with OmnibusF4PRO.
Compiled firmware 1.7.4, OSD works like without comment. I can test the gyroscope only on weekends.
I also have to solve the problem with erasing the flash.

fiam, 2017-09-26:
@a6j I'm glad to know the OSD is working properly on 1.7.4.

Regarding the "Full Chip Erase", you'll need to use the configurator from git or wait for the next release. It hangs because of a bug in the DFU driver, which was [recently updated](https://github.com/iNavFlight/inav-configurator/commit/df9abf057b4d562d740b55d90729238af26c1da5).

Please, let us know when you're able to test the gyro.

a6j, 2017-09-26:
The problem in the accelerometer was wrong. While re-tuning the controller, the accelerometer began to show incorrect values. In the configurator, the aircraft rotates with a large roll clockwise in space.
Calibration helps for a while, then the problem can be repeated anew.

digitalentity, 2017-09-27:
@a6j boards from banggood are known for their issues with accelerometer. Most likely they use ultrasonic cleaning to wash off the residual flux from the board - ultrasonic process damages accelerometer internal mechanics. If accelerometer calibration drifts over time - this board is simply unsafe to use.

a6j, 2017-09-27:
I already understood that. Today will be the time I solder accelerometer.
By results I'll let you know.

a6j, 2017-09-30:
АНГЛИЙСКИЙ
I replaced the accelerometer from the old CC3D controller. Accelerometer worked as it should, the horizon does not float away. The problem is solved. iNav flies smartly, before that 5 years flying to Eagletree. iNav super, flying smartly, thank you so much!
 iNav супер, летит шикарно, огромное спасибо!

a6j, 2017-10-01:
Yesterday no problems were revealed.
Today the problems are again manifested. The problem with OSD is partially left. When you exit the statistics screen on the third screen above.
The replacement of the accelerometer also only partly solved the problem, a slight deviation was tired.

digitalentity, 2017-10-02:
@a6j check if accelerometer in sensor tab shows (0;0;1g) when the board is flat. Issues with OSD are already fixed in development code - will be part of next release.

---
Акселерометр должен показывать X=0, Y=0, Z=1g на вкладке Sensors, если плата лежит ровно. Если это не так и значения меняются с течением времени - акселерометр поврежден. Проблемы с OSD уже починили - войдут в следующий релиз.

a6j, 2017-10-02:
В том то и дело что иногда долго ровно, иногда "уплывает" сразу при включении. MPU6000 я взял с CC3D там такой проблемы не было, на нем год отлетал на коптере. Неохота покупать и ждать новый полетник, да и интерфейс омнибуса наиболее удобен для мелколета. Есть еще NAZE32 с MPU6500 по даташиту выводы немного отличаются от 6000. Возможно ли заменить на 6500?

a6j, 2017-10-21:
We collected the firmware from the development branch 1.8.0 + did support the led strip on pwm5. Problems with the accelerometer are not noticed. Today, 6 batteries were flying off, the horizon worked stably and did not float away. The problem is probably not hardware, but software. In vain the money spent on the new MPU6000, the parcel will soon arrive.

digitalentity, 2017-10-22:
Closing the issue then.

closed 2017-10-22 by digitalentity