iNavFlight/inav #2216: Omnibus F4/F7 Servo Tilt/CamStab

cristy6100, 2017-09-26:
As there is no info on the Wiki regarding Servo Tilt+Cam Stab mode on newer board (F4-F7), I am asking here, which of the Omnibus boards support this feature, or do all of them? I want to use a single servo to control tilt on my quad using the Omnibus F7, and reassign LED pin output as servo control for tilt then enable CamStab mode, is this possible yet?

cristy6100, 2017-09-27:
It should also be possible to use a timer for the servo by using the UART 3/6 RX pin on the Omnibus F7 V1 and even UART 7 on the V2, how would one acomplish this? Or do we need to wait until Inav 1.8?

AntonVoltchok, 2017-09-27:
This is my exact issue as well, I need to trigger a servo for a dropper mechanism, if I just plug the servo wire directly into my receiver, would that work? Are there any solutions to this? (I'm on v2 version)

cristy6100, 2017-09-27:
It depends on the receiver, some FlySky receivers can send IBUS/SBUS and PWM at the same time, the catch is that it sends them both for the same channels, so if you use IBUS you need to wire your servo on a PWM channel that is not used by IBUS on the flight controller or it will trigger them both at the same time, I tested this with a FlySky i10 Tx and IA6B Rx. Now I'm using a Fkysky Sbus only receiver and wiring a servo is impossible for me, so waiting on input from developers regarding this

AntonVoltchok, 2017-09-28:
Thanks cristi, that's good news since I'm using flysky at the moment for this, so say I have first 5 channels in use on quadcopter, in inav I just don't assign the 6th channel and that's what I plug the servo into right?

cristy6100, 2017-09-28:
Yes if I remember right that should work, but you need to test and see if it does on your configuration

cristy6100, 2017-09-29:
Oh well, it seems the developers of INAV don't care about answering its users questions, time to go to Betaflight at least its developers are more user friendy.

digitalentity, 2017-09-29:
@cristy6100 do you expect to have the issue be resolved instantly? Unlikely to happen. this is not a paid support channel, developers have the life to live and their number is smaller than in i.e. Betaflight.

Your suggestion about reusing LED as a servo output is a good idea and it will be considered (and will likely be implemented). Bear with us.

cristy6100, 2017-09-29:
Of course not, and I fully understand that, I just asked the active developers for a confirmation that its not just me who can't get it to work. I expect nothing more than it is offered, and fully understand the Open-Source nature

digitalentity, 2017-09-29:
@cristy6100 yes, this is current firmware limitation. Omnibus F7 is very limited at the moment, and for Omnibus F4 we only have M6 to work as a servo. 

https://github.com/iNavFlight/inav/issues/2117 is also a relevant issue.

cristy6100, 2017-09-29:
I understand, yes indeed its limited, reading on the BF threads (I know they are not really related) one of the developers said the Omnibus F7 target does support LED pin and RX3/RX6 reassign as servo, maybe that could work on the F4 also, just an ideea, but its good to know sometime it will be possible, until then time to program a separate mpu6050 with arduino for the servo :))

digitalentity, 2017-09-29:
@cristy6100 so you are particulary interested in Omnibus F7? Just recently I've received Omnibus F7 from Airbot, intend to spend some time improving support for this board in INAV next week.

cristy6100, 2017-09-29:
Yes, I almost finished my quad and ordered two pieces of the Omnibus f7, they will arrive sometime next month if any testing will be needed I will gladly assist then.
I was checking my firmware options in advanced and I would use INAV because it seems more refined, I dont need dshot and all the latest, mostly interested in the auto and gps assisted modes

digitalentity, 2017-09-29:
Your help in testing and refining Omni F7 will be much appreciated.

cristy6100, 2017-09-29:
When they arrive I will post here and I can test builds if available to check for possible problems, thank you for your time and help :)

ins2d, 2017-10-23:
Hey guys! Really looking forward to use camstab/servotilt in my Airbot Omnibus F4v3 project. At least pitch axis.

Now testing 1.8-RC where finally dedicated LED pin seems to be used, possibly opening ch5/ch6 for servos so we don't even have to sacrifice LEDs for tilting? 

I'd love to help testing on my board once a solution is ready for that :)

ins2d, 2017-11-06:
So I've been playing around with some tiny hardware (1.7g servo) lately and came up with a simple and compact cog-wheeled tilt mount like this:

https://www.youtube.com/watch?v=ZHJeG3ivDO0

So eager to try this out on my (successful so far) INAV build with Omni F4v3 board.

Just making sure this topic is still alive ))

daspunner, 2017-11-27:
Hello!

Is there anyway to find out if this issue is going to solved for the omnibus boards?

/Dave

nahrwoldinternet, 2018-04-21:
Are there any updates on the possibility of resource remapping the currently non functional led pinout on the v2 to be used as a servo output?

digitalentity, 2018-04-21:
@nahrwoldinternet on Chinese clones LEDSTRIP connector is routed to a CPU pin that don't have assignable timer. This makes it impossible to use for anything that requires pulse generation - motor/servo/ledstrip etc

DzikuVx, 2018-05-09:
INAV 2.0 will solve this problem as long as target will have usable servo outputs. 
Channel forwarding and servo_tilt will be done on mixer level, not specially configurable, target dependent features. 

See https://github.com/iNavFlight/inav/blob/development/docs/Channel%20forwarding.md and  https://github.com/iNavFlight/inav/blob/development/docs/Servo%20Gimbal.md

Omnibus F4 will allow to use PWM5 and 6 for multirotors. Omnibus F7 does not have usable servo outputs for multirotors I'm affraid

Because so, closing this one. Thanks

gtoumie, 2021-07-22:
Hi @DzikuVx , the links for the documents are not working for me, i'm trying to setup servo gimbals 
can you refer me to relevant documentation please