iNavFlight/inav #2221: FSK modem in software

OpticalNZ, 2017-09-27:
For sending LTM telemetry down the FPV audio channel without special modem hardware. Potentially using either a PWM output (with two base frequencies) or DAC output?

Implemented by the Aeroquad project developed code to generate the FSK output in software. Code here:
https://github.com/AeroQuad/AeroQuad...AQ_SoftModem.h

fiam, 2017-09-28:
@OpticalNZ I've been looking at exactly this feature recently. The problem that the DAC pins are not arbitrary. You need to use either PA4 or PA5 (or both for stereo applications). In most boards those pins are either already used for something else or not broken out. This means few users would benefit from this feature.

Note that you can already connect a simple FSK modem to a serial port and then pipe its output to the audio signal.

fiam, 2017-09-28:
@OpticalNZ You can find how to build such a modem here https://www.rcgroups.com/forums/showthread.php?2963807-FSK-Modem-in-RC-Obsolete-or-a-hidden-gem

wx4cb, 2018-01-17:
so question. if you was to use a DDS oscillator you would only need a couple of pins, plus there are those that work on I2C, you wouldn't need to do much coding, just tell it whether it's a high or a low. but there are also FSK modem chips you can use such as AD5700. personally. I use XBEE modules on 900 meg for mine as i'm not using it for long range yet. either way would be fine because of a 433 control link

PiManASM, 2018-05-20:
Do you need a DAC? Seems like PWM would be fine, with the Arduino FSK Library you can just use digital IO pins

Nandox7, 2018-05-22:
Don't think a DAC is required since in the past other implementations were not using it.
i.e.: https://github.com/KipK/minoposd_audiomodem

fiam, 2018-05-22:
@Nandox7 The problem with PWM from any non-DAC pin is that you probably need additional external components to reduce the voltage (PWM will be 3.3V, while audio expects 2.5Vpp) and maybe to smooth the PWM into a sine wave.

To me, it seems like a lot of work for a small return, since those HW FSK modems are small and work quite well (I've built a few myself since this issue was opened). You could make them even smaller by using SMD components.

Nandox7, 2018-05-22:
I agree, plus you'd need a steady signal and that would assume you still have free timers in the FC.

I do plan to do that, go the HW way potentially as SMD to make it small enough to fit small lightweight sized wings.

closed 2021-08-23 by DzikuVx