iNavFlight/inav #2222: support for FrSky XSRF3O

janbolt, 2017-09-27:
I would like to see support for FrSky XSRF3O.

DzikuVx, 2017-09-29:
If FrSky would be willing to contribute by donating boards to dev team it might happen

janbolt, 2017-09-29:
is it possible to steal the code from betaflight ?

giacomo892, 2017-09-29:
@janbolt of course it is. but it is not stealing. Betaflight is opensource and you can port code to iNAV with "ease"

DzikuVx, 2017-09-29:
Only you can not just "copy-paste" since it will not work. Additional work and testing is required

janbolt, 2017-10-03:
what about cleanflight ? Can the code be "copied & pasted" from there ?

digitalentity, 2017-10-04:
Cleanflight = Betaflight in many aspects, including code portability. No, code can't be directly copy-pasted from there. Required changes are usually minor, but require some understanding of STM32 hardware.

DzikuVx, 2017-10-30:
With work in progress in #2430 I'm closing this one. Expect this target in release after 1.8

closed 2017-10-30 by DzikuVx