iNavFlight/inav #2229: Matek F405 AIO inaccurate current sensor readings

sammichael25, 2017-09-28:
Suggested scale of 165 by manufacture gives extremely high current reading even when motors are disarmed.

stronnag, 2017-09-28:
Dumping out the raw and processed values for the MSP_ANALOG intPowerMeterSum and amperage values (sitting unarmed on the bench). Current seems somewhat random. 
````
   Powermeter	  Current
raw	mAh	raw	Amps
5527	5.527	5969	59.69
5579	5.579	6327	63.27
5620	5.62	6806	68.06
5678	5.678	7593	75.93
5711	5.711	3266	32.66
5733	5.733	1309	13.09
5743	5.743	951	9.51
5750	5.75	672	6.72
5824	5.824	16460	164.6
5920	5.92	11763	117.63
5963	5.963	12636	126.36
6071	6.071	16121	161.21
6170	6.17	9987	99.87
6242	6.242	10587	105.87
6318	6.318	11284	112.84
6408	6.408	17181	171.81
6505	6.505	12206	122.06
6593	6.593	12175	121.75
6668	6.668	10006	100.06
6715	6.715	5400	54.0
6765	6.765	8236	82.36
6854	6.854	17327	173.27
6948	6.948	11630	116.3
7027	7.027	11000	110.0
7105	7.105	11036	110.36
7218	7.218	14587	145.87
7301	7.301	11654	116.54
7381	7.381	11418	114.18
7461	7.461	11315	113.15
7539	7.539	11327	113.27
7618	7.618	11278	112.78
7700	7.7	12787	127.87
7793	7.793	13315	133.15
7885	7.885	12709	127.09
7977	7.977	12848	128.48
8166	8.166	15593	155.93
8278	8.278	16381	163.81
8391	8.391	16854	168.54
8512	8.512	17521	175.21
8638	8.638	18103	181.03
8764	8.764	18987	189.87
8898	8.898	19466	194.66
9038	9.038	19993	199.93
9174	9.174	19939	199.39
9314	9.314	19993	199.93
9454	9.454	19993	199.93
9594	9.594	19872	198.72
9728	9.728	19993	199.93
9868	9.868	19993	199.93
10006	10.006	19454	194.54
10141	10.141	19993	199.93
10282	10.282	19993	199.93
````

sammichael25, 2017-10-01:
@digitalentity  Any chance you can help me plzz  :)

digitalentity, 2017-10-02:
Unfortunately I don't have an AIO board, but I'll check on Matek F405 OSD board I have.

sammichael25, 2017-10-02:
Much appreciated....its hard to predict flight time with only voltage

digitalentity, 2017-10-04:
Can't reproduce on MATEKF405 OSD. Current readings are correst. @stronnag @sammichael25 can you verify that ADC channel mapping is correct? It should be:
```
set vbat_adc_channel = 1
set rssi_adc_channel = 3
set current_adc_channel = 2
set airspeed_adc_channel = 0
```

stronnag, 2017-10-04:
As here:
````
airspeed_adc_channel = 0
vbat_adc_channel = 1
current_adc_channel = 2
rssi_adc_channel = 3
````
βF does the same, bizarre amps reading.

digitalentity, 2017-10-04:
Ok, thanks. There must be some difference between AIO and OSD boards. Time
to get Matek involved.

sammichael25, 2017-10-04:
My ADC channel mapping is correct
 `set vbat_adc_channel = 1
set rssi_adc_channel = 3
set current_adc_channel = 2
set airspeed_adc_channel = 0`

MATEKSYS, 2017-10-06:
Tested it just now, current sensor works normal with fw_1.7.3.
Pls send some clear wiring & board photos to service@mateksys.com

digitalentity, 2017-10-07:
@MATEKSYS thanks for stepping in.

sammichael25, 2017-10-08:
Matek helped me solve the problem.Turns out my board came without a specific resistor. After soldering on one and performing scale calibration it solved the issue. 
Thanks again for the help guys :) 👍 

[image]

digitalentity, 2017-10-08:
Well done @MATEKSYS! This is how a good board manufacturer should act! Closing.

closed 2017-10-08 by digitalentity

bhuism, 2017-10-09:
What value should the SMD resistor be? So I can fix mine as well

dc914337, 2018-08-26:
Great thread! I had wrong battery voltage reads and calibration didn't help. The problem appeared near the resister shown on the picture above. Cleaning the board resolved the issue.
I was running betaflight but it's not relevant as it's hardware problem. 

Hope my comment helps someone having wrong voltage reads. This thread appeared on the first page in Google when I was searching for a solution.