iNavFlight/inav #2230: Mag HMC5883 of REVO board not working with INAV 1.7.3

kahira, 2017-09-28:
After flashing 1.7.3 the HMC5883 mag. of the Revo Board isn't recognized anymore  ( painted red in the top ribon of the inav configurator)  witch makes  1.7.3 completely unusable.
Everything is back to normal flashing back INAV 1.7.2
Here is the diff all of the faulty setup

Entering CLI Mode, type 'exit' to return, or 'help'
# diff all
# version
# INAV/REVO 1.7.3 Sep  9 2017 / 10:40:06 (974440df)
# reset configuration to default settings
defaults noreboot
# resources
# mixer
# servo
# servo mix
# feature
feature -RX_SERIAL
feature PWM_OUTPUT_ENABLE
# beeper
# map
# name
# serial
# led
# color
# mode_color
# aux
# adjrange
# rxrange
# master
set acc_hardware = MPU6000
set mag_hardware = HMC5883
set baro_hardware = MS5611
set pitot_hardware = NONE
# profile
profile 1
# profile
profile 2
# profile
profile 3
# restore original profile selection
profile 1
#

Please advise. / Thanks

digitalentity, 2017-09-28:
Onboard compass is unsafe to use. Use external compass which works in 1.7.3 on REVO.

kahira, 2017-09-29:
Here we go again mister   SHARLAIMOV

First--onboard compass for fixed wings are in no way unsafe (no magnetic surrounding nuisances) .

Beyond that the Revo (the minI  version) is used for a thermal glider AND there is  no place to mount an external mag.

Further more others have also complained about this bug  https://github.com/iNavFlight/inav-configurator/issues/245

Finally why maim  in    1.7.3 what was working in 1.7.2 …..and doing so without making user aware of it .

Truly….the  issue deserve to be addressed   in a less dismissive way

digitalentity, 2017-09-29:
Internal compass is unsafe - this fact has been proven multiple times on both planes and quads. Some people (a small fraction of users) are able to use built-in compass with success, the rest (vast majority) see the onboard compass as a potential cause of a fly-away and prefer to stay on the safe side.

Regarding lack of space to add external compass - I assume you are going to use the GPS in which case you can use compass sensor that is available on many GPS units at zero extra space requirement.

Regarding your current issue - you are more than welcome to contribute to #1816 which will allow you to switch between internal and external compass.

Regarding your attitude - I should have remembered that you expect your issues to be resolved instantly and without question as if you have paid for support. Fine, this is your way of expressing yourself. But please, don't expect that such attitude will increase speed of resolving your issues - more like the opposite.

kahira, 2017-09-29:
Replica in italics

Internal compass is unsafe - this fact has been proven multiple times on both planes and quads. Some people (a small fraction of users) are able to use built-in compass with success, the rest (vast majority) see the onboard compass as a potential cause of a fly-away and prefer to stay on the safe side.

These are at best only  postulates for justification and certainly with very doubtful validity for planes especially motor less gliders  .Without the numbers to support your claims those are not facts but only your very  personal view .Anyway this shouldn’t allow you -  even more so without prior  notification to the user community at large -to prejudice opposite views by peremptorily/disruptively switching off onboard functionalities of  platforms officially supported until 1.7.2 by INAV 

Regarding lack of space to add external compass - I assume you are going to use the GPS in which case you can use compass sensor that is available on many GPS units at zero extra space requirement.

Obviously you have no understanding of the packaging of GPS boards. Indeed there is an increase of the footprint by a factor of  3 to 4  between  those with or  without a mag. The very reason I have to, and indeed do, use  a BN-180 and not a BN-800.(picture of layout available)

Regarding your current issue - you are more than welcome to contribute to  <https://github.com/iNavFlight/inav/issues/1816> #1816 which will allow you to switch between internal and external compass.

Why should I contribute to something which is required only as result of your knee jerk decision to  all of a sudden support only external mag  and deny disruptively   the use of internal ones although  included on platform you pretend to support (but actually maim them).

Regarding your attitude - I should have remembered that you expect your issues to be resolved instantly and without question as if you have paid for support. Fine, this is your way of expressing yourself. But please, don't expect that such attitude will increase speed of resolving your issues - more like the opposite.

Again short of factual arguments you derail on personal matters and allow yourself based on convoluted earlier disagreement to pass judgement on what you call  my attitude .Well what about your own  prima donna posturing so typical of system programmers renowned in IT departments for their moody  ,user dismissive and apodictic exuberations.

digitalentity, 2017-09-29:
@kahira REVO (both Revolution and Revo Mini) is less than 2% of used boards. Airplane & Custom airplane mixers are just ~11% of all setups. There are only a handful of users who use both REVO and airplane - much less on a motorless glider. If there are other users who are affected by disabling onboard compass on REVOs - they are welcome to chime in and have this issue reopened.

closed 2017-09-29 by digitalentity