iNavFlight/inav #2231: Oplink Modem Revolution

ValCx, 2017-09-29:
I would like to be able to use the oplink modem on the revolution board.
Anyone try to implement this module?

Thanks

Val

DzikuVx, 2017-09-29:
I'm not aware of any attempts to port Oplink modem into CF or INAV

fiam, 2017-09-29:
I have one of those boards available, but I don't have much interest in adding support for it since I don't have anything to interact with it from the ground.

atomiclama, 2017-10-07:
Hi, well I'm currently working on getting the RFM22 working within BF. Mainly for my own interest as I didn't think there was any interest out there.
I have a proof of concept setup running. 2-way serial data transmission that has been tested with Mavlink MSP and RX_Serial protocols.

It is not 100% reliable, contains no frequency hopping or configuration. But it is a start. Once I'm have it a bit further along I'll do a PR and see if I can get some others using it.

Any interest?

fiam, 2017-10-07:
@atomiclama I have no objections to a PR adding support for it. Could you let me know which hardware would I need to test it? Thanks!

DzikuVx, 2017-10-07:
@atomiclama Yes!

atomiclama, 2017-10-07:
Blimey that was a quick response, nice to know there is some interest. 

OK when I'm back home I'll put together the information / HW etc. I don't get much time for my hobby so please don't expect fast progress.
I'll also have a look to see how much iNav is different from BF in the RX_SPI area.

Are there any recommendations for GC SW that can handle just Mavlink telemetry? Currently using Mission planner but seams a bit old and clunky. QGround Control I can't get to handle just telemetry. Don't have an Android so can't use the aps for that.

DzikuVx, 2018-04-04:
I belive this is dead issue now. Original board is discontinued and not popular after all. Closing, thanks for participating

closed 2018-04-04 by DzikuVx

sfjuocekr, 2021-04-09:
It might be a little late, but would you still be able to share the code?

I've got a Revolution sitting idle here and was looking into what it would take to get it working!

It is kind of sad that this board never catched on, it was way ahead of its time...

atomiclama, 2021-04-09:
@sfjuocekr I never did get the modem up and running in iNav and did not continue the project in BF.
I did a stand alone RX instead as it was easier to develop and debug outside BF. 
Code available in my repos if you are interested. 
But I'm not working on it any more as no free time.

sfjuocekr, 2021-04-09:
I was looking into just using it as a receiver as well for BF.

My goal is to add a SPI_RX driver to BF to handle the OP Revo and then use the OPLM as a receiver.

I've been reading through most of the code from dRonin, LibrePilot and the datasheet on the RFM22b to get my head around the problem. It looks like implementing a receiver driver BF would be feasible, but it has been quite a while messing with embedded projects so I'm just looking for a place to start off :)