iNavFlight/inav #2232: F1 - MSP not working on Softserial if other Softserial Baudrate is different

Schalonsus, 2017-09-29:
Have a HC-05 connected to Softserial 1 and MSP 9600 baud.
When i select Softserial 2 to be LTM and 9600 baud everything is fine.
But if i change LTM to 1200 baud, Softserial 1 MSP is not working anymore.

DzikuVx, 2017-09-29:
I suppose this is one of SS limitations we are facing. This was never working very reliably when more than 1 SS was used

DzikuVx, 2017-09-29:
And this is not only F1 issue, so I'm changing topic

stronnag, 2017-09-29:
Not sure how you expect it to work. You would also need to change the rate of every other device; BT, radios, ground station. Not an fc issue as LTM at 1200 works with correctly configured hardware.

stronnag, 2017-09-29:
Ignore me, misread the question. OTOH, isn't it a SS limitation that the rates are the same for each ?

giacomo892, 2017-09-29:
That's stated also in the documentation. SS ports must all run at the same baud rate. Limiting but a reality.

DzikuVx, 2017-09-29:
So, close? I suppose...

closed 2017-09-29 by DzikuVx

giacomo892, 2017-09-29:
If someone wants to ammend this is very welcome. SS needs a total rework

Schalonsus, 2017-09-29:
ok thanks for the information, then i have to live with it

fiam, 2017-09-29:
@DzikuVx , @digitalentity How feasible would to allow different speeds on multiple SS?