iNavFlight/inav #2236: Failsafe RTH does not work on 1.7.3 - shuts down motors

stefan73, 2017-09-30:
INAV 1.7.3 on SP Racing F3 in a FPV Racer quadcopter; PWM RX; UBlox GPS
RTH works fine when activated over remote. 
Failsave set to RTH.

--> Switch off remote. Motors immerdiately off and copter drops.

fiam, 2017-09-30:
@stefan73 It might be a configuration error. Please, post a config so we can make sure everything is configured correctly.

stefan73, 2017-09-30:
Here is the config. Looks fine at least in the configurator. As said before: Flies perfectly. RTH & Postion Hold ok. Remote off --> Drop instead of RTH.

# version
# INAV/SPRACINGF3 1.7.3 Sep  9 2017 / 10:20:09 (974440df)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature MOTOR_STOP
feature GPS
feature RX_PARALLEL_PWM
feature LED_STRIP
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
[... 422 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 60
set pitch_rate = 60
set yaw_rate = 60
set tpa_rate = 0
set tpa_breakpoint = 1500

stronnag, 2017-09-30:
If you get rid of the (insane) MOTOR_STOP feature does it work?

Certainly works for me (cyan in manual, yellow is RTH, big red text show why it's in RTH).
[image]

stefan73, 2017-10-01:
Thanks a lot. Confirmed. Failsafe RTH works ONLY if MOTOR_STOP is disabled. i.e. Motors must spin when armed and throttle min. Otherwise copter just drops.
I consider this as a major BUG. It is shocking to see that this bug seems to be around for more than a year now. I somewhen enabled MOTOR_STOP because I use an older RC without arm-switch on that model. It nearly costed me the copter that failsafe turned to drop silently.

If this is unfixable or difficult in INAV, at least the INAV configurator should actively block this configuration and warn about this configuration.

digitalentity, 2017-10-01:
@stefan73 interesting. We have code to disable motor stop automatically when failsafe activates. 
Can you show us screenshots of **Receiver** tab in Configurator when your TX is ON and when your remote TX is OFF?

stefan73, 2017-10-01:
Sure. I can also describe this. Failsafe on the "cheap" RC that I used for the tests puts everything on center and throttle to low, when you switch off. Bad thing is that this is unchangable on a lot of cheaper RCs.
Let me know if you still need the screenshot.

digitalentity, 2017-10-01:
@stefan73 yes, I still need the screenshots. I think I know where the bug lurks, but I'm not sure if it's fixable at all. Screenshots will confirm.

stefan73, 2017-10-01:
Made two screenshots:
RC-on - copter on, RC on & connected
RC-off - after switching RC off
[image]
[image]

digitalentity, 2017-10-01:
Does your copter correctly recognize that RC is off?

[image]

Also, I see that when your RC is turned off AUX2 goes to 2000. Is this how failsafe is configured on the receiver?

stefan73, 2017-10-01:
The upper images are mixed up. Aux1 is at 1100 when RC is off. As said not configurable on this RC.
When I hook the copter up to the laptop, attach battery, bind RC and disarm, the props spin now. When I turn the RC off, the red parachute does not come. Maybe I do something wrong here.
When I fly with MOTOR_STOP off and turn off the RC, it comes back home. So I guess it recognizes the the RC is off.

digitalentity, 2017-10-01:
Ok, so when your RC is off, throttle channel goes down to approx. 889.
This is on the edge of detecting invalid signal (which may explain why sometimes it works).

Try setting `set rx_min_usec = 900` in CLI and don't forget to do `save`. This will ensire that when your RC is off, throttle channel goes to a invalid value that will trigger failsafe.

In general - you need to make sure that red parachute lights up when you turn off the transmitter. Also, using receivers that don't have programmable failsafe is not a good idea.

stefan73, 2017-10-01:
I see the point. Thanks. I do agree that the cheap remote might be too cheap because you cannot set failsafe.
Still I am wondering, why it reproducibly flies homes with MOTOR_STOP disabled and falls from the sky with MOTOR_STOP enabled.

fiam, 2017-10-02:
I think we should investigate this one further. It doesn't make sense that RTH via FS worked without MOTOR_STOP and dropped with MOTOR_STOP.

Also, if the problem was that THR = 889us wasn't triggering FS, the copter should have gone to the ground rather than RTH when MOTOR_STOP was disabled.

digitalentity, 2017-10-04:
I agree, this needs to be investigated. Changing the lablel to "Bug"

stefan73, 2017-10-14:
Bad news in terms of reproducability. I killed the remote that was involved into this. This made me switch to an rc with arm switch and ibus communication. So I cannot reproduce it any more. Sorry for that.

digitalentity, 2017-10-15:
@stefan73 same here. Unable to reproduce neither on Taranis (with variety of receivers) nor on FlySky (IBUS, SBUS & PPM tested)

digitalentity, 2017-10-18:
We are disabling MOTOR_STOP for RTH & WP - might mitigate issues with throttle stick being at zero when failsafe happens or RTH is activated. Closing.

closed 2017-10-18 by digitalentity