iNavFlight/inav #2237: OS: Windows, Chrome: 61.0.3163.100, Configurator: 1.7.3 omnibusf4v3 INAV version: 1.7.2-3 Board: OB43 OSD

lavaed, 2017-09-30:
[image]

When configuring icons in the configurator on the osd screen, the icons in the fpv image are cloned. The dragged icon remains in the old place and on the new one.

fiam, 2017-09-30:
@lavaed This has been already fixed and will be part of 1.8

fiam, 2017-09-30:
Just added the 1.8 milestone because I I discovered this one myself and I think I  fixed it without opening an issue. This way we won’t forget to mention it in the release notes.

closed 2017-09-30 by fiam

skaman82, 2017-09-30:
@fiam does your fix also includes the stats screen still being there (partly) after the throttle stick is moved up to hide the stats? The screen is only being fully refreshed after rearm.

fiam, 2017-09-30:
@skaman82 AFAIK I fixed that one too, but on a different commit.

fiam, 2017-09-30:
@skaman82 Fixed in https://github.com/iNavFlight/inav/commit/c5cfe4ac845351016d6fd233b48403bdc2c1b4a0 and then reworked in a better way in https://github.com/iNavFlight/inav/pull/2163