iNavFlight/inav #2240: omnibusf4 pro v3 voltage reading fault 1.7.3

robschii, 2017-10-01:
hello 

a full battery is showing only 0,9V in gui and osd

in betaflight it is 16,5V   and it is okay

is it the wrong target i flash?      omnibusf4 pro   oder omnibusf3v3???

digitalentity, 2017-10-01:
Try doing a configuration reset.

robschii, 2017-10-01:
nothing help  here is the dump

# version
# INAV/OMNIBUSF4PRO 1.7.3 Sep  9 2017 / 10:35:55 (974440df)

# resources

# mixer
mixer FLYING_WING

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 960 2160 1604 -100 -1 
servo 4 830 1920 1273 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature TELEMETRY
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
[... 601 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

digitalentity, 2017-10-01:
This is clearly not firmware defaults. In your dump channels for VBAT and current sensor are mixed up:
```
set vbat_adc_channel = 1
set current_adc_channel = 2
```

Try setting this:
```
set vbat_adc_channel = 2
set current_adc_channel = 1
```

robschii, 2017-10-01:
okay thanks it works now

digitalentity, 2017-10-01:
Great. Closing.

closed 2017-10-01 by digitalentity