iNavFlight/inav #2242: Display alarm on the OSD if altitude is below a threshold (negative one is the most useful case)

giacomo892, 2017-10-01:
It would be very helpful to display an alarm if the altitude drops under a certain threshold.

fiam, 2017-10-06:
How should we implement this one? A new configurable alarm for negative altitude or just hardcode it to blink below, let's say, -10m?

giacomo892, 2017-10-08:
I think we can made it configurable with a default of -10m.

Warning the user going under the landing altitude is important to prevent radio / video loss and the possibility of hitting objects along the route.

fiam, 2017-10-08:
Configurable with a default of -10m it is 👍

digitalentity, 2017-10-09:
Good thing. @giacomo892 you know first hand why this warning is required :laughing:

giacomo892, 2017-10-09:
@digitalentity Yes. After the second rescue mission the wing still missing. RIP :)

fiam, 2017-10-09:
Let us take a moment of silence for that Wing Wing. It will always be remembered as the wing who brought us low altitude alarms!

giacomo892, 2017-10-09:
I'll place a marble commemorative stone where I lost it. You are all invited to the ceremony.

digitalentity, 2017-10-12:
Fixed via 2296

PR #2296 (2017-10-12): Add OSD alarms for distance and negative altitude. Refactor alarms to use SI units internally.
Distance alarm makes the OSD blink when distance to home goes
above a given threshold. Default is 1000m and it always works
on meters to avoid performing unnecessary arithmetic operations
for displaying the distance indicator. Unit conversion is done
in the configurator instead. Zero disables the alarm.

Negative altitude alarm blinks when negative altitude is below
the given threshold. Note that the alarm is a positive number,
but will trigger with -ALT > NEG_ALT_ALARM. Default value is 5
(which mea [... cut]
commit a9081d9b (2017-10-12): Add OSD alarms for distance and negative altitude

Distance alarm makes the OSD blink when distance to home goes
above a given threshold. Default is 1000m and it always works
on meters to avoid performing unnecessary arithmetic operations
for displaying the distance indicator. Unit conversion is done
in the configurator instead. Zero disables the alarm.

Negative altitude alarm blinks when negative altitude is below
the given threshold. Note that the alarm is a positive number,
but will trigger with -ALT > NEG_ALT_ALARM. Default value is 5
(which means -5 meters). Again, conversion is pushed t [... cut]
closed 2017-10-12 by digitalentity