iNavFlight/inav #2243: Board orentation didnt work in any stabilized

JDU1, 2017-10-01:
hi,

INAV 1.7.3 on Omnibus F3 V2 board, fixed wing.
board is build in 90° CW turned. board alignment is 90° in configurator.
Set everything that the "brick" in configurator moves in the right way. 

If I turn any stabilized mode on, the FC thinks it dosn´t build in 90° CW and act like that.

Now  I went back to INAV 1.6.1 there is everythinh fine.

digitalentity, 2017-10-02:
If you have set alignment correctly different servo behavior in passthrough mode vs stabilized modes is caused by a combination of incorrect servo reversals or connections. You need to make sure your sticks are not reversed and then connect and configure servo reversals in such way that ailerons/elevator respond correctly to stick motion in all modes and airplane motion in stabilized modes.

JDU1, 2017-10-02:
sticks aren`t reversed and on inav 1.6.1 it worked fine with the exact same settings e.g. mixer. are there any differences?

digitalentity, 2017-10-04:
@JDU1 please try doing a configuration erase and re-configure from the start. There were some changes to servo mixer, old settings might be incompatible.

JDU1, 2017-10-05:
I will give it a try.

closed 2018-05-27 by stale[bot]