iNavFlight/inav #2251: Simplify launch mode

brat002, 2017-10-03:
Some time ago I've asked about an implementation of "simple launch assistant". After hard troubles with my planes and my friend's planes I am asking again to do that again.

How this should work:
1. Enable launch mode
2. Arm the plane
3. Motor starts to rotate with desired thrust. Elevons is up to maintain predefined angle 5-30.
4. Throw the plane.
5. After reach predefined altitude plane should switch to main flying mode.

Please. Current implementation is not clear and not safe. Each launch has LOT of actions to do something incorrect and to bend your new plane about the earth or damage equipment.

michaelvanderhoek, 2017-10-04:
i'm in for for this. I quit using launchmode because my wing hits the dirt more then it flies. 
No i just use a mixer on my transmitter so it gives a bit of up elevator and i throw the wing on full throttle.

fiam, 2017-10-04:
What we need is better documentation. Current launch mode can be configured to act like you describe minus the target altitude (which I’d find dangerous). Also, the default values for the auto launch setting need to be safe for all types of planes, which makes them not optimal for flying wings thrown by the side. What I do in my wings is set the idle THR while launch mode is enabled to 1200-1300, so the motor is already spinning before throwing it, and I lower the waiting times since my hand is never close to the prop. Note that this would be unsafe for airplanes thrown by the bottom with a pusher prop, that why we don’t want to make it the default.

brat002, 2017-10-04:
Not default certain, but as option - for flying wings this will be much more convenient than current implementation.

fiam, 2017-10-04:
This the configuration I have in my S800 with some comments (default settings omitted). Give it a try and let us know.

```
# Forward velocity the plane needs to reach for the launch sequence to be detected in cm/s.
# 3 m/s seems like a good compromise between false triggers and the force required to throw
# the plane. You can go lower if your hands are never close to the prop. Note that by default
# applying more than 2G of acceleration in the forward direction will trigger launch too
# (nav_fw_launch_accel).
set nav_fw_launch_velocity = 300
# How long (ms) horizontal velocity needs to be over nav_fw_launch_velocity until the launch
# system detects the plane has been launched. You can go lower if your hands are never close to the prop.
set nav_fw_launch_detect_time = 30
# I think the default is 1700 here? Depending on your power train you might want a different value.
set nav_fw_launch_thr = 1800
# THR when idle set to 1200 so the motor starts spinning before being thrown. Makes it
# almost impossible to miss a throw.
set nav_fw_launch_idle_thr = 1200
# Time (ms) since launch is detected until motor is started. If your hands are never close to the prop
# You can make this as low as you want.
set nav_fw_launch_motor_delay = 50
# Time (ms) since launch sequence starts until motor reaches nav_fw_launch_thr. You probably don't
# want to go lower than 100 since sudden motor speed changes cause stress in the ESC and motor
set nav_fw_launch_spinup_time = 100
# 10 seconds timeout. Default is 5. After this interval, the auto launch system will release control
# even if you haven't touched the sticks.
set nav_fw_launch_timeout = 10000
# Climb angle using during launch. This is 18 by default, but it highly depends on the plane. You
# can go higher if you want.
set nav_fw_launch_climb_angle = 23
```

brat002, 2017-10-04:
set nav_fw_launch_timeout = 10000 <- most of my concern is about that option. You don't know if iNav still handles the plane in launch mode or not. If you throw the plane on 11th second - it will fly to the ground. With default 5s timeout - plane will throw in the ground almost every time.

Please return us normal way to launch plane without all of this excess magic. Current implementation is for catapults. How much people are using catapult to launch their plane with iNav?

giacomo892, 2017-10-04:
nav_fw_launch_timeout is for telling iNAV after how much time stopping climbing. Not how much after the launch option is disabled on the ground.

brat002, 2017-10-04:
I don't know why, but after short climb the plane strongly dives to the earth.

digitalentity, 2017-10-04:
The sole purpose of launch mode is to prevent PID controller from winding up when plane is not flying.

When you pull the pitch stick on the ground with motor running PID controller won't be able to satisfy climb angle bevause FC has no control. However, since FC is unaware that the plane is not flying it will try as hard as possible by saturating the servos. Then, when you finally throw the plane you may end up with weird behavior. 

The key feature of launch mode is **detection of the launch**. Without it this mode is no different from a pitch mix on your TX. Therefore I don't see a point in adding requested behavior to the code.

digitalentity, 2017-10-04:
@brat002 do you have strong dive down when using launch mode or when using manual launch?

digitalentity, 2017-10-04:
In addition to my comment above:

I do, however, think that we should make launch detection user-selectable option to allow user chose from `THROW`, `SWING` or `GPS` speed.

Pairan, 2017-10-04:
@digitalentity ... that would help a lot while thinking of greater planes! I can't swing a full loaded heavy Skywalker X6 :) ... thats more a THROWing there and hope the prop wont shredd the hand !

brat002, 2017-10-04:
>@brat002 do you have strong dive down when using launch mode or when using manual launch?
Both. My previous plane was completely destroyed due that, but my friend's plane almost does same every launch. Just luck to restore control above half meter above earth.

>I do, however, think that we should make launch detection user-selectable option to allow user chose from THROW, SWING or GPS speed.
Or NONE. In this case plane should be ready to stabilise and climb just after enabling launch mode. Value of throttle should be an user select for each launch.

I will cry of happiness if this would appear.

digitalentity, 2017-10-04:
Ok, let's make it happen in 1.8 then :smile:

brat002, 2017-10-04:
Sure!

Pairan, 2017-10-04:
The S800 Sky Shadow will surely try all methods ;)

oleost, 2017-10-04:
Why would you want to choose between different launch detections?
What if you want to to trigger on which ever scenario is met first?

Maybe a simpler GUI for selecting the different parameters with explanations is all we need?

Current implementation is very powerful and allows the user to set it up exactly as he wants.

Pairan, 2017-10-05:
> Current implementation is very powerful and allows the user to set it up exactly as he wants.
That might be true, but people need to get "educated/guided" here, as many seem to toss their craft and hope for the best! ( _"iNav will do this for me" ... NOT! :D_ ...) 

I got it working on mine but of course ended up in fails sometimes ... 

Perhaps something like presets for the parameters (like in the type of craft presets) could help here and leave it open to the users while still being able to tweak on everything?!

Optimized basics for a swing, a overhead toss, basic throw (uh nasty with wrong params btw) or something like that. Allthough this could be hard to create matching profiles but I assume it to be even harder to hard code those movements/values.

How about that @oleost @digitalentity ?

Schalonsus, 2017-10-05:
IMHO the people should just read the documentation before using iNav, but that seems to be the hardest part for most of them. If we make more options too choose, most people will be more confused than now

brat002, 2017-10-05:
Just try to keep this mode as simple as possible. Current implementation is
really not simple to use.
Many options - many mistakes, hard to test and use it.

digitalentity, 2017-10-05:
Current implementation launch mode is as simple as it could possibly be. Defaults are safe and reliable for small planes (up to 80-100cm wingspan). Generally you need to set up only a few parameters:

1. Launch throttle value that is used to climb (`nav_fw_launch_thr`)
1. For bigger planes you might want to lower the acceleration threshold (`nav_fw_launch_accel`).
1. For classic planes you may want to keep props spinning at some throttle value while waiting for launch (`nav_fw_launch_idle_thr`)
1. You may also want to adjust delay between throw and motor startup (`nav_fw_launch_motor_delay`).

If you want to do it simpler - configure pitch mix on your TX and don't use launch mode.

brat002, 2017-10-05:
Mix won't disable I term of PID.

Pairan, 2017-10-05:
Looks like, we'll need more education on HOW IT'S DONE. Okay ... maybe I could put some video up on that by time if we get a weather slot.

digitalentity, 2017-10-05:
Nothing will allow us to zero I-term on the ground except proper throw detection - which requires some configuration and also some understanding how it's done. We definitely need some tutorials about how to configure launch mode :smile:

oleost, 2017-10-05:
Mhm, only thing is that CLI is not user friendly for the average user.

If we get the values shown in GUI in the configurator with tooltip explanation it would probably be better.

nav_fw_launch_velocity - Used to detect launch in when you `SWING` launch your Fixed-wing. Lower this value if you intend to use Swing launch. Example 200

nav_fw_launch_accel - Used to detect launch when you throw your Fixed-wing straigh. Should be lowered if you intend to use this. Around 1700 is a good starting point.

nav_fw_launch_max_angle - Used to limit the angle when an forward launch is consider safe to do. If you intend to throw example above or other non straight throw you might want to increase this over the default value, set to 180 to disable it.

nav_fw_launch_detect_time - Maybe just dont show this value in GUI?

nav_fw_launch_thr - Launch throttle - throttle to be set during launch sequence (pwm units)

Not time to suggest the rest..

|  nav_fw_launch_idle_thr       | 1000  | Launch idle throttle - throttle to be set before launch sequence is initiated. If set below min_throttle it will force motor stop or at idle throttle (depending if the MOTOR_STOP is enabled). If set above min_throttle it will force throttle to this value (if MOTOR_STOP is enabled it will be handled according to throttle stick position)	|
|  nav_fw_launch_motor_delay    | 500 | Delay between detected launch and launch sequence start and throttling up (ms) |
|  nav_fw_launch_spinup_time    | 100 | Time to bring power from min_throttle to nav_fw_launch_thr - to avoid big stress on ESC and large torque from propeller |
|  nav_fw_launch_timeout  | 5000 | Maximum time for launch sequence to be executed. After this time LAUNCH mode will be turned off and regular flight mode will take over (ms) |
|  nav_fw_launch_climb_angle  | 18 | Climb angle for launch sequence (degrees), is also restrained by global max_angle_inclination_pit |

digitalentity, 2017-10-05:
GUI page for airplane-related configs?

brat002, 2017-10-05:
How to trust all of this magic? I haven't ten of spare planes to test which
values are better.

digitalentity, 2017-10-05:
It's simple - if you don't trust launch mode - don't use it. People are flying perfectly fine without it.

There is a very simple way to test that launch detection works. Remove the prop. Give your plane a jerk as you would do when launching it but don't let it go. Motor should start. If it doesn't - you are either doing something wrong (forgot to raise throttle stick etc) or have too strict thresholds for your plane.

digitalentity, 2017-10-05:
GUI page for setting up launch mode would be good. Maybe with some nice pictures illustrating parameters relevant to different ways of throwing the plane.

brat002, 2017-10-05:
>People are flying perfectly fine without it.

Somebody - maybe. If PIds are tuned with autotune - you should have a lot of luck to throw your plane as usually. It always flies into the ground almost without reaction to pitch.

Here is my friend's config of launch parameters:
```
set nav_fw_launch_velocity = 300
set nav_fw_launch_accel = 1400
set nav_fw_launch_max_angle = 45
set nav_fw_launch_detect_time = 30
set nav_fw_launch_thr = 1920
set nav_fw_launch_idle_thr = 1200
set nav_fw_launch_motor_delay = 10
set nav_fw_launch_spinup_time = 50
set nav_fw_launch_timeout = 20000
set nav_fw_launch_climb_angle = 23
```
Why does the plane make strongly dive into the ground exactly after short climb in launch mode?

digitalentity, 2017-10-05:
For troubleshooting issues you need to provide dump, logs & detailed description of the setup. Might be setup, might be tuning, might be a bug, no way to tell

DzikuVx, 2017-10-05:
I'm also against changing Launch. It works just fine and is well documented. I'm not using it ATM, but works just fine

Dronek, 2017-10-05:
I have been launching my aircraft almost exclusively in launch mode since it was released. It failed very few times, mostly in the beginning. But after some adjustment of parameters accordingly to match my personal throwing style and my aircraft it has been a rock solid mode for many successful launches. 

The possibility to adjust launch mode as it is now is actually very powerful and once you understand all parameters (which isn't that hard, actually) it exhibits a completely transparent, reliable and logical behavior.

Just make sure you stay away from the steering stick until aircraft has climbed enough.

I think this here is mostly a documentation issue rather than a functionality/feature issue.

brat002, 2017-10-06:
Please re-read this thread before any comments. It is not replace of
current launch mode.

5 окт. 2017 г. 23:26 пользователь "Dronek" <notifications@github.com>
написал:

I have been launching my aircraft almost exclusively in launch mode since
it was released. It failed very few times, mostly in the beginning. But
after some adjustment of parameters accordingly to match my personal
throwing style and my aircraft it has been a rock solid mode for many
successful launches.

The possibility to adjust launch mode as it is now is actually very
powerful and once you understand all parameters (which isn't that hard,
actually) it exhibits a completely transparent, reliable and logical
behavior.

Just make sure you stay away from the steering stick until aircraft has
climbed enough.

I think this here is mostly a documentation issue rather than a
functionality/feature issue.

—
or mute
the thread

brat002, 2017-10-07:
This is what happens when auto launch is enabled https://www.youtube.com/watch?v=A-46gI5Q-2I&feature=youtu.be. Strong dive exactly after short climb. The plane shouldn't allow dive in any case, but you see shat happens.

digitalentity, 2017-10-07:
Not possible to troubleshoot by video alone. Please record a log

oleost, 2017-10-07:
Cli Dump and an log please

brat002, 2017-10-31:
Your plane just has big wing surface. Try something like FunJet 1.5KG AUW with APC 4.5x4.5 prop and 6s and you will feel my pain. 
There are another thing, that make safe launch almost impossible. It is current TPA implementation on launch. These problems are related.

danarrib, 2017-10-31:
@brat002 , the video you linked above (https://www.youtube.com/watch?v=A-46gI5Q-2I&feature=youtu.be) seams to be a Small Flying wing, not a Funjet.

My flying wing have 120cm wingspan and weights about 1.5kg.

I guess the funjet you described is difficult to launch even in manual mode... I know a Friend who have a funjet that spins when launched due to the torque of the motor (and slow airspeed of course). On this aircraft, an autolaunch sequence must gradually increase throttle to reduce this effect, but must reach full throttle very quickly. Also, it must start it's engine as soon as it detects the throw (I guess 50 to 80ms is ok).

Follow digitalentity's advice and record a blackbox log. It's the only way to figure out what's happening with your aircraft.

linuxras, 2017-11-04:
I have also been depending on launch mode with my smaller planes 1.3m or less, it has failed me exactly one and it was my fault for not bumping nav_fw_launch_max_angle as I wanted to launch a wing at high alpha. 
Please leave the main settings as they are and make `SWING`/`THROW` just adjust those values to some basic defaults.

Landflieger, 2018-05-13:
why we have all the parameters to understand and to tune ... why can autolaunch be working like with an MFD Autopilot. Tune climbing angle, activate Launchmode, put your throttle up to the desired power an throw your plane ... is kind of working well, without tuning all the parameters ...

digitalentity, 2018-05-13:
@Landflieger parameters are about detecting a launch. You don't have to use launch mode at all. Add pitch up mix on a switch, arm, activate, throw.

closed 2018-07-27 by stale[bot]

padcom, 2019-07-12:
For my setup, where I launch the plane first, then pick up the goggles, put them on and in the end I grab the transmitter a setting of

```
set nav_fw_launch_timeout = 60000
```

does the trick. It basically means _you have 60 seconds to get everything sorted out - after that time the plane is yours_. Of course I never need that much time so the moment I have everything on I just touch the sticks and the plane is mine to control. Well sort of because I usually fly in 3D Cruise mode :)