iNavFlight/inav #2255: 1.7.3 osd.c compile failure for OMNIBUSF4PRO

SPlissken, 2017-10-04:
Compiling is failing on osd.c, please see the error below.
Two other non-OSD targets (NAZE and SPRACINGF3) have compiled with no errors.
No changes were made to osd.c only frsky.c which compiles fine.
Thanks.

make TARGET=OMNIBUSF4PRO
%% osd.c 
./src/main/io/osd.c: In function 'osdConvertDistanceToUnit':
./src/main/io/osd.c:128:37: error: expected ')' before ';' token
         FALLTHROUGH;
                                     ^
./src/main/io/osd.c:128:9: warning: 'fallthrough' attribute not followed by ';' [-Wattributes]
         FALLTHROUGH;
         ^~~~~~~~~~~~ 
./src/main/io/osd.c:127:10: error: label at end of compound statement
     case OSD_UNIT_UK:
          ^~~~~~~~~~~
./src/main/io/osd.c:124:5: warning: enumeration value 'OSD_UNIT_METRIC' not handled in switch [-Wswitch]
     switch (osdConfig()->units) {
     ^~~~~~
./src/main/io/osd.c: In function 'osdFormatDistanceStr':
./src/main/io/osd.c:153:37: error: expected ')' before ';' token
         FALLTHROUGH;
                                     ^
./src/main/io/osd.c:153:9: warning: 'fallthrough' attribute not followed by ';' [-Wattributes]
         FALLTHROUGH;
         ^~~~~~~~~~~~ 
./src/main/io/osd.c:152:10: error: label at end of compound statement
     case OSD_UNIT_UK:
          ^~~~~~~~~~~
./src/main/io/osd.c:144:5: warning: enumeration value 'OSD_UNIT_METRIC' not handled in switch [-Wswitch]
     switch (osdConfig()->units) {
     ^~~~~~
./src/main/io/osd.c: In function 'osdConvertVelocityToUnit':
./src/main/io/osd.c:176:37: error: expected ')' before ';' token
         FALLTHROUGH;
                                     ^
./src/main/io/osd.c:176:9: warning: 'fallthrough' attribute not followed by ';' [-Wattributes]
         FALLTHROUGH;
         ^~~~~~~~~~~~ 
./src/main/io/osd.c:175:10: error: label at end of compound statement
     case OSD_UNIT_UK:
          ^~~~~~~~~~~
./src/main/io/osd.c:174:5: warning: enumeration value 'OSD_UNIT_IMPERIAL' not handled in switch [-Wswitch]
     switch (osdConfig()->units) {
     ^~~~~~
./src/main/io/osd.c:174:5: warning: enumeration value 'OSD_UNIT_METRIC' not handled in switch [-Wswitch]
./src/main/io/osd.c:172:49: warning: unused parameter 'vel' [-Wunused-parameter]
 static int32_t osdConvertVelocityToUnit(int32_t vel)
                                                 ^~~
./src/main/io/osd.c: In function 'osdFormatVelocityStr':
./src/main/io/osd.c:194:37: error: expected ')' before ';' token
         FALLTHROUGH;
                                     ^
./src/main/io/osd.c:194:9: warning: 'fallthrough' attribute not followed by ';' [-Wattributes]
         FALLTHROUGH;
         ^~~~~~~~~~~~ 
./src/main/io/osd.c:193:10: error: label at end of compound statement
     case OSD_UNIT_UK:
          ^~~~~~~~~~~
./src/main/io/osd.c:192:5: warning: enumeration value 'OSD_UNIT_IMPERIAL' not handled in switch [-Wswitch]
     switch (osdConfig()->units) {
     ^~~~~~
./src/main/io/osd.c:192:5: warning: enumeration value 'OSD_UNIT_METRIC' not handled in switch [-Wswitch]
./src/main/io/osd.c:190:40: warning: unused parameter 'buff' [-Wunused-parameter]
 static void osdFormatVelocityStr(char* buff, int32_t vel)
                                        ^~~~
./src/main/io/osd.c:190:54: warning: unused parameter 'vel' [-Wunused-parameter]
 static void osdFormatVelocityStr(char* buff, int32_t vel)
                                                      ^~~
At top level:
./src/main/io/osd.c:172:16: warning: 'osdConvertVelocityToUnit' defined but not used [-Wunused-function]
 static int32_t osdConvertVelocityToUnit(int32_t vel)
                ^~~~~~~~~~~~~~~~~~~~~~~~
make: *** [Makefile:961: obj/main/OMNIBUSF4PRO/io/osd.o] Error 1

stronnag, 2017-10-04:
Known bug, fixed in the development branch.
Other options:
* Downgrade the compiler to 6.3*
* Add the missing bracket ./src/main/common/utils.h
````
#if __GNUC__ >= 7
#define FALLTHROUGH __attribute__ ((fallthrough))  // <-------------- missing )
#else
#define FALLTHROUGH do {} while(0)
#endif
````

SPlissken, 2017-10-04:
Thanks a lot stronnag!

closed 2017-10-04 by SPlissken