iNavFlight/inav #2256: FrSky telemetry sending number of sats in Celsius

SPlissken, 2017-10-04:
In function sendSatalliteSignalQualityAsTemperature2() in frsky.c, line 218, the data is sent back as if it were Fahrenheit converted to Celsius. This happens if 'frsky_unit = IMPERIAL'.

As the data is either signal quality or the number of satellites and never temperature then converting it makes no sense, in fact if 'frsky_unit = METRIC' then it is left untouched and correct.

Would it be possible to replace this:

    if (telemetryConfig()->frsky_unit == FRSKY_UNIT_METRICS) {
        serialize16(satellite);
    } else {
        float tmp = (satellite - 32) / 1.8f;
        //Round the value
        tmp += (tmp < 0) ? -0.5f : 0.5f;
       serialize16(tmp);
    }

with just this:

serialize16(satellite);

Thanks.

fiam, 2017-10-09:
Not sure if this is a bug per se. Maybe OpenTX is performing the inverse celsius/farenheit conversion since we're using a sensor intended for sending a temperature. Have you tested how the sensor data is displayed when the the radio is configured with Units = Imperial? (1st page in the OpenTX radio configuration)

teckel12, 2017-10-11:
I have my FC set to `frsky_unit = IMPERIAL` and there's nothing funny going on.  Everything is displayed correctly on my transmitter.

The reason it's set to degrees Celsius is because that's what the `Tmp2` telemetry sensor defaults to.  Maybe you changed the telemetry sensor on your transmitter from degrees Celsius to degrees Fahrenheit?

Also, are you running an older transmitter with frsky telemetry or is it a newer X9D/X9D+/QX7 which uses the SmartPort telemetry instead?  There's a bit of confusion on this sometimes.  In any case, I'm running SmartPort on QX7 and the `Tmp2` telemetry sensor works perfect.

SPlissken, 2017-10-11:
I first noticed this after an upgrade from 1.6.1 to 1.7.3 left me with -13 satellites, I know this is now a post-fact world but I still found negative satellites hard to swallow, what next, imaginary satellites?

No matter the units that my transmitter was set in I still received negative satellites.
With 'frsky_unit = METRIC' entered on the CLI everything looked normal again so I had a poke around in the code.
I recompiled with my suggested change and everything was fixed and the satellites and HDOP were reported without error again.

The setup is:
D4R-II with 27ms CPPM firmware
Q X7 with OpenTX 2.2.0 final
SP Racing F3 clone

The relevant code that I am referring to is in FrSky telemetry as handled in telemetry/frsky.c and not SmartPort.

My problems with this are:

1) In OpenTX the units of tmp2 need to be null otherwise the HDOP and number of satellites are displayed and spoken as degrees so converting the raw value makes no sense as no conversion is done anyway in a properly configured OpenTX.

2) The IF statement is basically saying that if 'frsky_unit = IMPERIAL' then send the data as converted to Celsius/Centigrade  which is just plain wrong.

3) The conversion rounds the result to make it an integer so presumably it is possible to get the wrong result when converting back, thereby possibly sending a false number of satellites. I haven't checked this particular point mathematically so it might be complete nonsense.

SPlissken, 2017-10-13:
I've checked the code in telemetry/smartport.c which does a similar thing by sending HDOP and number of satellites as temperature in FSSP_DATAID_T2, line 696. 
There is no conversion going on there at all and there is no switch based upon METRIC/IMPERIAL which lends further weight to the argument that the conversion in sendSatalliteSignalQualityAsTemperature2() in telemetry/frsky.c is wrong.

teckel12, 2017-10-14:
@SPlissken I can confirm that it works correctly with SmartPort as I just totally re-worked `smartport.c`. It's correct that it sends as Celsius, but there shouldn't be any conversion from metric to imperial as it's not sending a temperature, just using a telemetry sensor that defaults to Celsius. 

So I would agree `frsky.c` is wrong. No imperial/metric conversion should happen with that telemetry sensor. I don't have any D-series receivers to test the code myself or I'd submit the code change.

teckel12, 2017-10-14:
@SPlissken As part of this change, we could also get the FrSky and SmartPort telemetry in sync. This would help avoid confusion with sensor data differences and allow my INAV Lua script to work with FrSky D-series receivers as a bonus. I could look at doing the code changes if someone could test the results for me.

SPlissken, 2017-10-14:
@teckel12 I have had a think about that, and the first thing that occurred to me was that I have renamed a lot of my sensors so that they appear with decent names on my own telemetry screens, I would have thought that other people might have done this too, so using ID's rather than names would be a worthwhile extension to the code I feel.

Also there is no distance sensor in frsky.c so that might make the code a bit more complex.

SPlissken, 2017-10-14:
@teckel12 I'll definitely help with the testing but if you are talking about a rewrite of frsky.c then there is a lot of redundant and buggy code for the various voltage sensors. I use my own version of frsky.c which fixes these problems which I update myself and recompile at every release. I haven't submitted anything here because I felt that messing with people's voltage sensors might annoy those with a setup based upon the existing idiosyncratic sensors. 
It also seems that some of these bugs might have been there since baseflight days maybe even multiwii.
Let me know if you do plan a full rewrite and I'll send over my work on the voltages.

If it helps then I've been flying with my suggested Tmp2 change for a couple of weeks and everything seems fine to me independent of which units are used.

teckel12, 2017-10-15:
@SPlissken I did a lot of work with the SmartPort telemetry to also support 2-way communication. I don't think a total rewrite for FrSky telemetry would be required to be more consistent with SmartPort, increase functionality and still not mess up people that have long-standing scripts or settings. I'll look at the FrSky code tomorrow and see what the potential issues are. Also, it sounds like maybe a good time to incorporate your voltage changes. Looking at the latest Clean/Betaflight FrSky telemetry may give some direction also.

SPlissken, 2017-10-15:
@teckel12 I have attached my frsky.c version with comments next to the code that I've changed in lines 378-431.
All the changes have been tested and working for me for about 9 months.

[frsky_markup.txt](https://github.com/iNavFlight/inav/files/1385787/frsky_markup.txt)

teckel12, 2017-10-16:
@SPlissken First off, making the FrSky telemetry return results like S.Port would break all legacy stuff so I think that's out of the question.

But, the satellite count logic is WAY off for sure.  The `cellVoltage` was also incorrect for sure.  However, the `currentCell` appears to be designed to show the voltage of each cell changing every second.  As `currentCell` is static, each time `sendVoltage` is called it would increment to the next cell.  Is it not doing this?  Or, maybe it's hard to tell as the cell voltages should be fairly similar.

In any case, I have a branch that I'll submit with the obvious fixes.  I'm going to leave the `currentCell++` stuff as I believe that's working and even if not it's not doing any harm, or is it?

SPlissken, 2017-10-16:
@teckel12 Nice work!  It's been a while since I made those changes and I'd forgotten about how `currentCell++` worked. It is a hangover from the real sensor which did measure each cell individually.

In my view it is redundant as the cell voltages are identical in this case as they are just vbat divided by the number of cells. I have those two lines commented out and it just shows me a single cell voltage on my TX which is easy to work with in OpenTX, this means I can set one voltage alarm that is independent of how many cells I'm flying which is really handy. I'm not sure how useful it is to be sending the same number multiple times according to the number of cells.

teckel12, 2017-10-16:
@SPlissken Good point.  There's no possible way it's getting multiple cell voltages through INAV so it's impossible to transmit multiple values as well.  What target are you using?  Or do you want to compile it yourself?  The branch can be found here:

https://github.com/teckel12/inav/tree/te_frsky_telemetry

Note, it's a branch from `development` not `master` so it's a 1.7.4 build.  If you want to compile your own build, here's just the changed file:

https://github.com/teckel12/inav/blob/te_frsky_telemetry/src/main/telemetry/frsky.c

SPlissken, 2017-10-16:
@teckel12 compiled, flashed and successfully tested: I compiled it with master (1.7.3) and tested it on a SPRACINGF3 target: Cell voltage is accurate and displayed properly and satellites are correctly reported irrespective of units. Thanks a lot for fixing that.

teckel12, 2017-10-16:
Resolved in #2354

closed 2017-10-17 by SPlissken