iNavFlight/inav #2260: armedFlags is uint32_t in FC, but sends uint16_t in MSP_STATUS_EX

stronnag, 2017-10-05:
It looks like armedFlags has experienced some mission creep since MSP_STATUS_EX was introduced:
````
./src/main/fc/runtime_config.c:uint32_t armingFlags = 0;
./src/main/fc/fc_msp.c:                sbufWriteU16(dst, armingFlags);
````
and `src/main/fc/runtime_config.h`
````
    ARMING_DISABLED_BOXFAILSAFE                     = (1 << 16),
    ARMING_DISABLED_BOXKILLSWITCH                   = (1 << 17),
    ARMING_DISABLED_RC_LINK                         = (1 << 18),
    ARMING_DISABLED_THROTTLE                        = (1 << 19),
    ARMING_DISABLED_CLI                             = (1 << 20),
    ARMING_DISABLED_CMS_MENU                        = (1 << 21),
    ARMING_DISABLED_OSD_MENU                        = (1 << 22),
````
If the outside world only needs 16bits, all fine and dandy (a comment to that might help, or explicit cast). Otherwise perhaps a new MSPV2 STATUS_EX_EX message?

digitalentity, 2017-10-06:
I'm in favor of introducing the new MSP2_INAV_STATUS message

stronnag, 2017-10-06:
Me too, on the other hand, if we just had the 3rd party interesting values in the lower 16bits (there seem to be some undefined bits in `src/main/fc/runtime_config.h`), then the old STATUS_EX would be useful again.

stronnag, 2018-05-13:
Fixed some time ago via MSP2_INAV_STATUS

closed 2018-05-13 by stronnag