iNavFlight/inav #2261: RTFQ Omnibus F4 V3 Mag

0higgins0, 2017-10-05:
Can't get the magnetometer to be recognized on omnibus f4 v3.  Frsky sbus on uart 6 (sbus, 5v,gnd) and gps on uart 1 are working. Mag to scl and sda will not for me.  Any ideas

DzikuVx, 2017-10-05:
try lower `i2c_speed` and recheck if UART3 is not configured. Also you might trying adding 4.7kOhm pullup resistors for SCL and SDA to 3.3V line.
And confirm that hardware is functioning with different FC or Arduino for example

teckel12, 2017-10-06:
@0higgins0 And don't flip scl and sda as you do with a UART.  I can confirm that the Omnibus F4 V3 works with iNav 1.7.3 and a magnetometer.  Also, is the magnetometer working in the configurator?  That's where you want to make sure it works first before over SBUS.  Finally, make sure you detect new sensors on your transmitter.

0higgins0, 2017-10-06:
Thanks for the response. I don't have the scl and sda flipped, but that's a good point to mention.  Without frsky sbus wired or turned on through uart 6, my mag still isn't recognized on inav. Just get the red icon.  Tried lower and higher i2c speeds with no luck.  However, this is an m8n and mag combo from readytoflyquads that has a 3.3v reg/filter built in.  Not sure how that may differ from other setups

jstremmler, 2017-10-06:
With my RTFQ Omnibus F4 V3 it works. Hope, we are talking about the same FC: https://www.banggood.com/Eachine-Wizard-X220S-FPV-Racer-Customized-Version-Omnibus-F4-V3-Betaflight-3_0_1-Flight-Controller-p-1177121.html?rmmds=myorder
Pls note that this FC has a connector J16 at the same place where the Omnibus Pro F4 V2 had connector J10  for SDA,SCL,5V,GND,TX6,RX6. If you connect your magnetometer to J16 you can ruin the FC and/or the magnetometer because J16 on this board has pins for: GND,5V,PWM1,PWM2,PWM3,PWM4. (same order) Don't trust the layout for J16 you can find in the internet! I tested it by controlling the pins against J12,J13,J14 and J15 on the board. So first, what you have to do, is to connect the magnetometer to the connections for SCL/TX3 and SDA/RX3 on the 3pin row at the rear side of the board. Then try to set with the iNAV configurator the hardware for the magnetometer in the Configuration Menu to AUTO and see if it works......

0higgins0, 2017-10-06:
Auto doesn't work either. Tried that many times.  Also, my board's scl/sca are on j1 thru-hole locations, labeled as you are describing (scl/tx3, ect.).  Again though, that's a good point to bring up for others reading this.  Note this board is from readytoflyquads.  I ordered it as a flip32 f4 v3 omnibus.  Im not affraid to be wrong with my wiring or config. I just want to get it to work and maybe help others having similar issues.  Thanks to all .

jstremmler, 2017-10-07:
So my Omnibus F4 Pro V3 (the one from Banggood-see above) works perfectly. 

I connected the GPS to UART6 AND SDA and SCL to UART3. Have a look at the screenshot of my iNAV Configurator [https://www.dropbox.com/s/6ok9fuj4twda2nn/Eachine_OmnibusF4.jpg?dl=0](url) to see that it works!

0higgins0, 2017-10-07:
Thanks. But now my gps won't even work.  Tried all 3 uarts at different baud rates. I had trouble getting the gps to work when I got it, then it suddenly started communicating . Maybe the unit was bad from the start.  Going to get another unit to try.

jstremmler, 2017-10-07:
Shit happens! I crashed a quad some time ago but nothing seemed to be broken (I thought). 

But then I recognized that the GPS did not find satellites anymore. Opened the case of the GPS and checked voltages. Everything was ok, then connected another GPS and that one worked! So the GPS was broken in the crash! Btw. the magnetometer worked with the broken GPS....

0higgins0, 2017-10-07:
I suppose if people like us didn't try, nothing would ever advance.

digitalentity, 2017-10-08:
Happened to me a few times. GPS modules are fragile thingies - once in a while after a crash I have to throw GPS outside a window because it stops working.

teckel12, 2017-10-08:
I once had a bad GPS module, but it decided to return to China (with my poor 1 day old quad attached). I was glad to see the problematic GPS sensor go, but not so glad it took my newborn quad with it.

0higgins0, 2017-10-08:
That does hurt.  Fortunately, my gps was purchased as a single item, so I
won't have that problem.  I just ordered another unit on Amazon,  so we'll
see how that one goes.

Tried testing my "bad" unit on another board.  Didn't work with it either.
Again, tx/rx GPS stopped working aside from the mag never working for me.

I'll update when I receive the new gps/mag

jstremmler, 2017-10-09:
Connections:  
[image]

Bushedo, 2017-10-09:
I have similar issues with external mag.  I have an Omnibus F3 AIO (no magnetometer) FC from RTF Quads. I cannot get it to work with the external compass on my GPS in iNAV. The compass module uses a 5883 mag that I have verified works using an Arduino mega board and Adafruit libraries and code. GPS works fine in iNAV.  When I select this chip under the mag drop down menu and save the config, iNAV quits communicating and tells me I have to update the software. To regain comm with the board through iNAV I have to power down and disconnect the SDA and SCL lines from the mag. Have searched the iNAV forums and so far have not been able to find any similar problems. There were suggestions about adding 3.3 V pull ups to the SDA and SCL lines but this caused more problems until I removed the 1K ohm added resistors.  Any help will be greatly appreciated.  Had same problem with another GPS module (one BN-800 and one NEO M8N).
Any suggestions will be greatly appreciated.
  |

DzikuVx, 2017-11-21:
Ticket seems be be stuck. I'm closing it. If any new data appears feel free to reopen

closed 2017-11-21 by DzikuVx

GS-AutoTech, 2018-01-04:
I've been having alot of problems with my GPS/MAG. I've built a 250mm Nighthawk Quad.
Omnibus F4 V3 Pro : FrSky X4rBS using SBUS : Matek PDB : OneShot 125 BL Heli ESC's : M8N UBLOX w/MAG, round puck with mast. was listed as the PixHawk GPS.

I used the supplied board connection for the GPS, GPS on UART 6 & MAG on UART 3.  5v supply from board thru the same connector. The GPS showed blue in Configurator after I finally got the port properly set. The MAG showed red. I worked for 2 days looking for a solution. I found a thread about the pull up resistors for the MAG. I installed the pull up resistors in the GPS puck & BAM! GPS & MAG now blue. Calibrated MAG & GPS has good fix w/ 9-13 sats indoors.  
I thought i was good to go but now the quad won't ARM.  I have re-flashed all three targets with same results. I have a second GPS that I went thru same process : installed = no comm, wired pull up resistors = got comm & calibrated MAG, good fix.  BUT wont ARM.  

I am new to RC & Quad building. I have quite a bit of tech experience & am a quick study. Although I claim no expertise. I have learned a great deal from the threads I have read here & that has helped tremendously. Thanks in advance for any help here, its very appreciated.

teckel12, 2018-01-06:
@GS-AutoTech Just so you know, it's something on your end as I have similar builds without a problem (but I've been doing this for a long time).

There's a reason it won't arm. There's lots of safe-guards in place with INAV. Consider it a good thing that you couldn't arm as you probably would have lost your model otherwise.

Anyway, logs and CLI dump would tell why it's not arming. Also, I would make sure you've calibrated everything (more than just mag calibration). Also with the mag, make sure it's orientation is correct.

Really, start to end instructions would be helpful. From ESCs to transmitter. But, there's a lot of different hardware out there so it's difficult to compile. Best look to YouTube for help (the late Blue Falcon is a good source).

GS-AutoTech, 2018-01-06:
Thanks for the reply. I have some updates. Both the GPS & MAG went red in iNAV.  Pulled status from CLI & it says failing on both. So I removed the pull up resistors out. Rechecked = same result. I then wired in my other puck with resistors & then removed resistors, same failing status. Trying to isolate the hardware issue, I wired up a new Omni F7V2 board with puck. Status= failing.
So I either got 2 bad pucks out of the box or I smoked them both during my build. 
In any case I don't blame iNAV.  
Every mistake,  error and setback is teaching me a lesson. 
I'd like to thank everyone for all your help.

teckel12, 2018-01-07:
@GS-AutoTech battery connected?

GS-AutoTech, 2018-01-07:
Yes, both battery connected & not. I've tried every combination I could think of.. I ordered new GPS pucks for both my builds. I found that the DJI pucks will do both GPS & MAG functions on one UART. Which I feel.is advantageous,  so that's what I ordered. 
I will keep posting progress. I'm excited in anticipation of getting iNAV up & running. 
Thanks a bunch for your help

jstremmler, 2018-01-08:
If you would have ordered the recommended GPS from banggood (the one shown in my above picture ) everything would have worked immediately. This GPS even doesn't require the pull up resistors like other GPS's. This GPS is relatively cheap and does not produce any i2C errors with the Omnibus F4 Pro V3. 
The next one from banggood works as well  but requires the pull up resistors: https://www.banggood.com/APM2_6-2_8-GPS-Module-High-Precision-Ublox-NEO-M8N-GPS-With-Electronic-Compass-PIXHAWK-p-1240362.html?rmmds=search&cur_warehouse=CN  
Pls note that the pull up resisitors have to be supplied with 3.3V and not with 5V!!! But the shown GPS has as an advantage an internal 3.3.V power supply. So with these GPSs I have managed to install the pull up resistors inside the GPS and power supplied by the GPS.

jstremmler, 2018-01-08:
I forgot that with the Omnibus F4 Pro V3 you have to be careful when connecting a serial RX and a GPS with magnetometer. The inverter for the serial RX is on the Omnibus F4 Pro V3  usually located at UART6  and then of course you cant connect the GPS to UART6. But then you can connect the GPS to UART1, the magnetometer to UART3/i2C and the serial RX to UART6.  
If the inverter for the serial RX would be implemented on UART3  you would be lost because you need UART3 pins for SCL and SDA for the magnetometer. (or you manipulate your RX for an uninverted serial output)
As I have connected a PPM RX to my FC I didn't have any of these problems.

GS-AutoTech, 2018-01-08:
Thanks for the info. I had gotten conflicting information about the 3.3v & 5v input requirements for the GPS. I had wired the pull up resistors to the 5V supply & for a short time both the GPS & the MAG were working. GPS on UART6,, MAG on UART3.  
I should have tapped the regulated 3.3v supply to install the resistors. This, I'm pretty sure is where i went wrong. 
I will report back when I have the new units installed.

teckel12, 2018-01-09:
@jstremmler on the Omnibus F4 *Pro* there's no inverter on UART6, the inverter is on UART1.  J10 is where you should connect GPS & magnetometer which is on UART6 and I2C (re-purposing UART3).  You are confusing the Omnibus F4 Pro v3 with the Omnibus F4 v3 non-Pro, which are NOT the same.  The Pro model has the amperage sensor, the non-pro does not.  There's very little difference between the different Pro versions (v2/v3).  Best to not confuse anyone and just call it the Pro only.

Please see the latest INAV Omnibus F4 documentation for the correct information:

https://github.com/iNavFlight/inav/blob/development/docs/Board%20-%20Omnibus%20F4.md

If you don't need the inverter (not using FrSky) you can be disabled or you can use J5 which doesn't use the inverter.

jstremmler, 2018-01-10:
I own 2 Omnibus F4 Pro clones  (the ones with the current sensor on board) from China  which have the inverter on UART6! With these FC's I could only make the GPS working when connecting it to  UART1......

But your are right, most of the Omnibus F4 Pro FC's have the inverter on UART1 and actually I didn't see anyone with the inverter on UART3. But who knows.....

teckel12, 2018-01-11:
@jstremmler Well, clones you can never guess what you'll get, nor can INAV (or the documentation) be created to every messed up clone.  For **ACTUAL** Omnibus F4 Pro boards, UART1 has the inverter, UART6 is on J10 (along with UART3/I2C).

The documentation also talks specifically about the Pro Corner and there's already a different target for it in INAV.

jstremmler, 2018-01-11:
I did see this documentation before, but it is incomplete because in iNav we cant allocate resources like in Betaflight and I don't know whether an inverter for UART3 and another one for UART6 will be available in iNAV as well. 
But I have ordered an Omnibus F4 Pro corner  (price is actually very low, $27 at Gearbest) and will then see how inverters are implemented in iNAV.

GS-AutoTech, 2018-01-12:
I received 2 DJI GPS LITE pucks in the mail.  Connected to UART6 on my F7
board. Now both MAG & GPS are working.. I will test 2 DJI setup on the F4
pro board on my second build soon.
Thanks for all your hard work and help.
I'll keep in touch.

teckel12, 2018-01-12:
@jstremmler For the F4, inverters are hardware, not software.  So you can't turn on and off inverters like you can with the F3 for example.  With the official Omnibus F4 Pro, the inverter on UART1 can be disabled by changing a solder pad or by using the micro-JST J5 instead of the thru-hole J7 connection.

There's little documentation on clones as they're all over the place, so don't expect documentation to help with unknown clones as they're... unknown.

jstremmler, 2018-01-13:
This is correct for iNAv but what do the above Betaflight commands
resource inverter 3 C09 and
resource inverter 6 C08
for the Omnibus F4 Pro corner are doing then? Have a look here: https://github.com/betaflight/betaflight/wiki/Board---OMNIBUS-F4-Pro-Corners

teckel12, 2018-01-15:
@jstremmler You can pull a pin high or low to turn on/off the inverters.  Basically, the board manufacture uses unused pins to control the inverters.  Hopefully, your Gearbest clone FC included this features and tied it to the right pins.

GS-AutoTech, 2018-01-15:
I have decided to ONLY use original Omnibus flight controllers. So that I
have the best support on my hardware & so that I can help support the
original creators of the product.

teckel12, 2018-01-17:
@GS-AutoTech Me too. Typically like $5 difference, why bother with a clone?

jstremmler, 2018-01-17:
Average price of an original Omnibus F4 Pro corner e.g. is approximately $44. So the price difference between the clone and the original is $17 or 38%.  The truth will prevail!

GS-AutoTech, 2018-01-17:
I look at it a different way, my time is valuable. When I waste several
hours or even days trying to sort out a problem relayed to a clone part, I
could have bought a pile of original boards with full support from the
original creators.. the extra $$ also supports new development from those
creative people.

By the way, can anyone forward me or point me to where I can download the
full factory manuals and doc's for the original Omnibus boards?

dseivwright, 2018-01-22:
Thanks. I was having issues getting a HMC588 MAG connected on an Airbot Omnibus F4 V5 with the UART3/I2C shared pins. It was the i2c_speed CLI command that got it working for me by dropping it to 200KHZ.

teckel12, 2018-01-22:
I almost think 200kHz should be the default as it solves so many problems and 200kHz is plenty fast for devices that connect to i2c (magnetometer, barometer, sonar).

DzikuVx, 2018-01-22:
@teckel12 200khz is not enough sice it will too low for boards with I2C gyro. And we have plenty of those. Like the most popular SPRF3 for example

teckel12, 2018-01-22:
@DzikuVx the SP Racing F3 only works in INAV with the 200kHz setting, at least all of the SPRF3 boards available today.  And is 200kHz really too slow for the gyro?  If so, why is there even a 200kHz option?

GS-AutoTech, 2018-01-26:
I had problems with several GPS pucks on 1 original Omnibus F4V3Pro board &
both F7V2 boards. I performed the pull up resistor fix on 2 different
pucks, Tried on all three boards..  I did get MAG to work for a short time
with the resistors on the F4 board, then the puck went dead. the same hack
on the second puck I got nothing but a second dead puck... I was very
careful with my wiring & checking/double checking connections.
This is not an isolated issue.. My issue was solved by installing the DJI
puck. Worked instantly, no hacks needed.