iNavFlight/inav #2265: Inav issue full throttle after dive quad

teamnanopirate, 2017-10-06:
Hello,
With a friend we both have the same problem, with a 6 " quad, after a dive, with full trottle the drone abruptly go on the side and back.
After long tests, we can say that it is not coposant of the drone, not its structure (we have 2 differents frames).
This morning we flashed Betaflight just to see and as by micracle the problem had disappeared on the 2 quads.
Do you know what this can be ? a bug from Inav? a bad setting? (tested with 1.7.2 and 1.7.3)
We use FRsky R9M SBUS
I am in 5S and my friend in 4S lipo
here vidéo of the issue :
https://www.facebook.com/gemim.nanopirate/videos/1944740352409663/
Thank you for your hep

fiam, 2017-10-06:
@teamnanopirate Video requires registration to watch. Can you post it somewhere else?

DzikuVx, 2017-10-06:
I also can not see the video. @teamnanopirate please post a video on a public resource and provide a blackbox log of that happeing. Without it is really hard to tell what is happening. Until then I'm tagging this as "Support"

Redshifft, 2017-10-06:
@teamnanopirate  Looks to me like the Quad runs out of controlling headroom - Antigrav in Betaflight might help otherwise just use less throttle in Inav. My money is you will also get those "kicks" with rapid full throttle punches also ;)

teamnanopirate, 2017-10-06:
Hi, 
sorry for the link, here is Youtube and log (drone issue in the log's file at 0:57)
[LOG00001.TXT](https://github.com/iNavFlight/inav/files/1362848/LOG00001.TXT)
https://www.youtube.com/watch?v=vzoMMmpjJhg

DzikuVx, 2017-10-06:
@teamnanopirate 
[image]

Check motor#1. Probably ESC requires calibration since FC start to request 100% from it while requesting only around 70% from front motors. Check props, calibrate all ESC and shuould be fine

DzikuVx, 2017-10-06:
And whole quad seems to be little tail heavy

teamnanopirate, 2017-10-06:
@DzikuVx  thanks for your interest but nop, my esc are correctly calibrated and props are ok.
100% rear motors and 70% front look normal compared to my actions.
finaly my quad is well balanced.

DzikuVx, 2017-10-06:
@teamnanopirate with all due respect, blackbox log you provided states something completely different. Motor 1 is at max even before jerk/roll starts and this is clearly the reason of what you are expecting

teamnanopirate, 2017-10-07:
@DzikuVx thank you for your help maybe it's that but all my set up is correct. 
my friend has exatly the same issue and when we flash betaflight all is ok :(

DzikuVx, 2017-10-07:
There are at least few reasone why it works better in BF. For example, DSHOT in BF? Or different max_throttle and so on and so on.

teamnanopirate, 2017-10-07:
I don't say that beta is better or worse, it's not the problem.
I'm just saying that with identical settings we don't have this problem with beta. 
which excludes materials problems.
min throttle set to 1100 and max 2000, esc are correctly calibrates., Tx end points are off course set.

fiam, 2017-10-07:
@teamnanopirate Thing is, betaflight devotes way more resources (both on chip and human) to racing mini quads, while INAV is focused on navigation.

Due to the needs of racing quads, betaflight has way more complicated filtering and can cope better with some mechanical/electrical problems on small quads. In exchange for this, they aren't able to support GPS navigation at all on F3 targets and their F4 navigation is much simpler than what INAV supports.

What I'm trying to say is that is perfectly logical that while a quad flies OK using betaflight, it might need some adjustments and fine tuning in order to make it fly as well on INAV because the latter is not able to mask some problems as well as the former.

digitalentity, 2017-10-08:
I agree with @DzikuVx and @Redshifft - INAV runs out of headroom for executing the required correction. 

Betaflight copes better because has more resources to do sophisticated filtering and some smart things like antigravity, while in INAV we have to sacrifice those things in favor of navigation and other smart things that are not available in Betaflight :smile:

digitalentity, 2017-12-31:
No activity. Closing.

closed 2017-12-31 by digitalentity