iNavFlight/inav #2266: Problem with barometer when OSD enabled on Omnibus F3

MrAlfabet, 2017-10-06:
Steps to reproduce:
Buy the omnibus F3 V1.1
Flash iNav 1.7.3 (Also tried 1.7.1, no difference). I notice that even though full chip erase is enabled, I don't really see it erase anything.
Check barometer (BMP280 onboard) in the sensors tab. This is working fine.
Enable OSD, the barometer now gives weird readings, values jumping up and down. https://i.imgur.com/T4xcH4e.png
With OSD disabled, the barometer works fine
This is with and without things connected to the FC (camera/vtx/lipo/escs etc)
This is the second FC that has this problem (thought the baro was faulty on the first one)
Just tried 1.7.1, same problem.

I also tried flashing the inav hex file with BF configurator (which does do a full chip erase), problem persists. iNav is also having problems with connecting to the board when OSD is enabled. Sometimes it will only give me the CLI, sometimes it just doesnt connect for a few tries. Sometimes it reboots after connecting.

Also tried pulling the iNav configurator from the git repo (why doesnt this include all files? I had to pull the 'build' folder from the extension installed from the chrome store), no dice.

MrAlfabet, 2017-10-06:
Just tried the new 1.7.4, problem solved.

closed 2017-10-06 by MrAlfabet

stronnag, 2017-10-06:
966e97c6a0d6f6fafc14eb85164576e544615a35 to be precise