iNavFlight/inav #2267: Bug - NAV LAUNCH + FS Mode = out of control

Schalonsus, 2017-10-06:
Today i wanted to start my wing but gone terrible wrong.

Start procedure after GPS lock:
go to start position
set switch to activate Alt Hold + Angle
set switch to activate Nav Launch
stick arming
set switch to activate FS
apply throttle (motor starts spinning)
throw the plane

so whats now: plane takes off, but kind of different when normal launching with Nav Launch. Hurry to reach the FPV screen, see that it barley skips the roof of some houses and disarms itself. I had no control and couldn't rearm. So with the little control left landed the plane in the woods.

Fortunately the plane just went 500m so recovery wasn't that bad.
I first thought it was the fault of new LRS receiver but i can reproduce the failure.
I made the exact procedure as listed above and the same occured, just that i didn't throw the plane but hold it in my hand. No control at all, whatever stick i stirr or what mode i selected. Just waiting until launch time is over or disabling Nav Launch exits this mysterium with a Disarm. And rearming is not possible for long time.

So the problem occurs if you use Nav Launch, plane is armed and then you select FS.
[BlueWing_Dump.txt](https://github.com/iNavFlight/inav/files/1363923/BlueWing_Dump.txt)

digitalentity, 2017-10-07:
Ok, the issue is that launch mode takes priority over failsafe RTH. Failsafe code attempts to initiate RTH but fails, so plane falls back to normal failsafe procedure - and disarmed as a result of it. Re-arming is disabled for 30 seconds after failsafe procedure finishes.

If you could record a log - it would help figure out what's going on under the hood.

We also need to discuss what the desired behavior should be in case failsafe activates when launch mode is active.

Schalonsus, 2017-10-07:
Unfortunately its a F1 board without flash memory and all UARTs are occupied.
But maybe i can test with another plane, just need to enable BB.

I think the right procedure would be let nav launch finish and then initiate FS without disarming

oleost, 2017-10-07:
My initial thoughts is if Failsafe is active, either by switch or by system it should immidiality go to failsafe behavior.

In this cause, instant disarm due to close to home position.

We should never mix Failsafe with usage. ( I assume you wanted it to circle after launch ? )

In this cause combining launch mode with pos hold and alt hold or RTH mode would be much more appropiate.

Schalonsus, 2017-10-07:
Your assumption is 100% correct. I changed RTH from this switch to FS. If FS switch is active ebfore arming it doesnt allow to arm, but because i forgot to switch it on before arming, i did it afterwards and this leads to this problem.

closed 2023-10-27 by DzikuVx