iNavFlight/inav #2270: RTH mode

madoboy, 2017-10-07:
Hi. Can we have option in RTH to when plane comes near to launch point to lower the flying altitude? Like if if is in 200m range to change from 500m to 100m. And also analog airspeed sensor when it's gonna be implemented to maintain planes speed? Now on RTH throttle is fluctuating.

digitalentity, 2017-10-07:
Good idea. We'll make that happen

renatobfa, 2017-10-09:
Also, while in RTL, the THROTLE POSITION is showing my manual inputs instead of the real throtle position sent to the ESC by the FC... I think THROTLE POSITION should always show the REAL throtle output to ESC, so we can know how much is being commanded by the FC in ANGLE, HORIZON and RTL modes.

giacomo892, 2017-10-09:
@renatobfa the Throttle display will be solved in the new release. It is already in the development branch.

jelle737, 2017-10-26:
I like this 2 stage RTH and like to see it implemented. Some extra altitude for longer distances would be welcome. Or even a mix of 'highest' on distance and 'fixed' once close.

heady123, 2018-03-14:
What about a "continuous descent rth" until we reach a given min-altitude.
Should save battery during return. combine it with a 2-stage.
For each stage we could have a given mode (either fixed, contin descent, max-height ...) and the needed parameters.
This would allow for something like over 500m distance (tunable) the aircraft stays at the current altitude and from there we continuously descent until we reach a defined altitude over the homepoint and change to loiter.

jelle737, 2018-03-14:
@heady123 what you suggest can be (almost) achieved by setting a small nav_fw_dive_angle. also in rth mode if you cut throttle while flying above the setpoint altitude, the motor should stop running (if allowed through the motor stop feature) and plane should glide (correct me if this behaviour changed). Setting a small nav_fw_dive_angle has it's drawbacks though, if it's too small and therefore does not make the plane descend, it won't descend. But some different type of rth behaviour like your suggestion would be a good add-on.

Fast-Guy, 2018-05-20:
Dual altitude RTH would be great for my flying area. Very hilly so that I have to fly at close to 2000' to clear the terrain by 200' yet at home 2000' would be the planes height above me and very difficult to take over by LOS. Having the plane drop down to 200' or 300' at home would be a great enhancement.

closed 2018-06-26 by fiam