iNavFlight/inav #2278: Make the OSD use METRIC by default

fiam, 2017-10-07:
OSD defaults to IMPERIAL units. I think we should change the default to METRIC. I'm not sure how many INAV users are from countries where they don't use the metric system, but my suspicion is we metric users represent a bigger proportion.

So, thoughts on this one? Feel free to use thumbs up or down.

teckel12, 2017-10-08:
Those using the metric system are smarter and therefore more capable of switching from imperial to metric. If we make metric the default, all the dumb Americans will be confused as to why the numbers are all wrong, resulting in a ton of problem tickets. (Spoken tongue-in-cheek from an American who wants to keep the default imperial).

fiam, 2017-10-08:
We the MPFROW (Metric People From the Rest Of the World) have been oppressed for too long, having to spend a couple of seconds changing the OSD units on every flight controller we have set up. The ruling IMPERIAL class has taken this too far and have pushed us to the limit, so now that we don't have anything else to lose, the uprising has begun: ¡Viva la revolución!

Just to be clear, I have no strong feelings about this one. Please, thumbs down my 1st comment so we can keep track of everyone's opinion. The option with more "votes" just before we cut the first 1.8 beta, wins. Does that sound fair?

digitalentity, 2017-10-09:
I placed a :+1: under the first post because I'm using metric system. However from backwards compatibility perspective and user friendliness it might be wiser to keep default units imperial.

giacomo892, 2017-10-09:
All iNav internal computations are done in metric. Seems weird to me to end up displaying by default converted data.

I know it's just a settings in the CLI/Configurator but it is rather funny.

Having real word stats will for sure give us an answer, but it will surprise me if there will be more IMPERIAL users than METRIC ones 🥇

teckel12, 2017-10-10:
I actually wouldn't mind if it defaulted to metric because everything else on the FC is metric by default and everything on my transmitter is metric by default as well.  It would seem more consistent if everything in INAV was metric.

digitalentity, 2017-10-13:
Ok, metric it is. Done in #2294

PR #2294 (2017-10-13): Make OSD default to OSD_UNIT_METRIC
Fixes #2278
commit 0c2c0f57 (2017-10-13): Make OSD default to OSD_UNIT_METRIC

Fixes #2278
closed 2017-10-13 by digitalentity