iNavFlight/inav #2285: Beitian BN-880 GPS doesn't work with iNav > 1.4

scottyob, 2017-10-08:
Anyone else had issues with the Beitian BN 880 not working with iNav > 1.4?  

I see the I2C error count climb, and when it's in a climbing state, if I disconnect the BN-880 entirely, it still climbs making me think the I2C has got itself in a bad state.

I've tried putting a 1K resistor between the +5V, SCL and Ground as well as one between +5V, SDA and ground.  I've tried swapping those for a 4.7K with no change either...  

Running under a OMNIBUS F4 target (with DYS F4 Pro)

Seems to work great in Betaflight

scottyob, 2017-10-08:
Sorry, title should read >= 1.6.0

digitalentity, 2017-10-08:
Try lowering I2C speed to 200khz and see if it solves the issue for you.

digitalentity, 2017-10-08:
How did you connect the pull-up resistors? They should go from SCL to +3.3V and from SDA to +3.3V

scottyob, 2017-10-08:
Yep. Tried both 1k and 4.7K in the fashion you mentioned.  I tried also changing the speed to no avail. 

Should the errors continue rising if I disconnect scl and sda mid operation? (Seeing them on the inav chrome plugin)

digitalentity, 2017-10-08:
Can you post a `bootlog` output of your board with connected compass?

scottyob, 2017-10-08:
Sure.  I've also noticed the "cycle time" is around 2,000.  Not sure what that should read either.

```
# bootlog
Time Evt            Description  Parameters
   2:  0          CONFIG_LOADED 
   2:  1       SYSTEM_INIT_DONE 
 502: 19   TIMER_CHANNEL_MAPPED  (0, 0, 0, 0)
 502: 18  TIMER_CHANNEL_SKIPPED  (2, 0, 0, 3)
 502: 18  TIMER_CHANNEL_SKIPPED  (3, 0, 0, 3)
 502: 19   TIMER_CHANNEL_MAPPED  (6, 1, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (7, 2, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (8, 3, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (9, 4, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (10, 5, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (11, 6, 0, 2)
 502:  2          PWM_INIT_DONE 
 502:  3       EXTRA_BOOT_DELAY 
1339:  9         GYRO_DETECTION  (6, 0, 0, 0)
1640: 10          ACC_DETECTION  (7, 0, 0, 0)
1663: 11         BARO_DETECTION  (4, 0, 0, 0)
1663: 20        PITOT_DETECTION  (0, 0, 0, 0)
1663: 12          MAG_DETECTION  (2, 0, 0, 0)
3341: 13  RANGEFINDER_DETECTION  (0, 0, 0, 0)
3341:  4       SENSOR_INIT_DONE 
4041:  5          GPS_INIT_DONE 
4092:  8           SYSTEM_READY 

# 
```

teckel12, 2017-10-08:
I'm running Omnibus F4 with Beitian BN-880 running INAV 1.7.3 on 2 models and both work perfectly with no errors (at 400Hz).  My Omnibus FCs both have a barometer, so I believe the pull-up resistors are already installed (I didn't wire up pull resistors). If your FC doesn't have a barometer, maybe then the pull-up resistors are needed?

Pairan, 2017-10-09:
Running the Beitian BN-880 on Matek F405 OSD (has no Baro) and it works perfectly!

giacomo892, 2017-10-09:
are there in the wild some different BN-880 revisions? Maybe some works ok, others not.

DzikuVx, 2017-10-09:
I also have no problems with BN-880 on different boards and firmware versions

scottyob, 2017-10-09:
It might just be a bad board then perhaps?

DzikuVx, 2017-11-17:
Since ticket got into stuck phase, I'm closing it. If any new data appears, fee free to reopen

closed 2017-11-17 by DzikuVx

robcer1122, 2020-05-06:
Does anyone know if the BN 880 has a barometer?

giacomo892, 2020-05-07:
@robcer1122 no