iNavFlight/inav #2286: No motor output on Matek f405 AIO

x3m84, 2017-10-08:
Ive connected servos on s2 and s3, and esc on s1. servos work fine.
i can connect to the esc in blheli, but the esc is unresponsive to throttle inputs from the motors tab. 
i have tested to put the esc on s1-s5, but only connects to blheli on s1.

to me it seems like inav doesent send a throttle signal on s1.

ShawnzAZ, 2017-10-08:
I'm having a similar problem in 1.7.3.

I have the ESC on S1 and the servos on S3 / S4.   The motor works on S1 but only the servo on S3 works.  On my previous FC this is how it worked, this one doesn't seem to put motors on output 1 & 2.

In the motors tab do you see 1000 on just output 1 or 1 and 2?  On mine I just see it on 1.   I suspect this is why my S4 isn't working (the other servo is on S2)

x3m84, 2017-10-08:
I only see one motor output in motors tab, on my old omnibus i had two.

Ive tested alla protocols, different blheli softwares and two different escs.

On mine servos dont work on S4, only on S2 and S3.

ShawnzAZ, 2017-10-08:
I confirmed the same for the servos.  S2/S3 = Outputs 3/4 in the mixer.

I'm using the flying wing mix and a Hobbywing Skywalker 50A ESC with inav set to 'Standard' output for the ESC protocol.  It is working on the S1 output of the Matek F405-AIO for me.

closed 2017-10-08 by x3m84

Turek, 2017-12-29:
Hello, I can't get the servos to work on my setup, where did you connect + and - for the servos? To their corresponding + and - on the FC on to the 5V regulator on the board?
No matter how I connect them servos are not working, main motor is working fine though.

ShawnzAZ, 2017-12-29:
Turek - you do not want to run the servos from the 5v from the Matek.

What I did was use an ESC with a BEC and just connected the +5v from the ESC and the two servos together.   Same for the grounds, except that I added a ground back the FC ground (any ground, I think I used an ESC pad).

If your ESC does not have a BEC, you'll need a 5v BEC and connect it up to the +5v and ground from the servos and ESC.  Don't forget the ground back to the FC.