iNavFlight/inav #2287: Motor / Servo outputs not correct for Flying wing in 1.7.3 beta on Matek F405 AOI

ShawnzAZ, 2017-10-08:
It appears that the flying wing config only setup on one motor when it should setup 2 on outputs 1&2.  Only output 1 shows the 'min throttle' setting.  The rest show 0.

Output 3&4 should be the servo outputs.   I wired it this way and only output 3 moves a servo.   The associated 'servo' is Servo 4 in the GUI,

I'll move the servo signal wire from output 4 to 2 to see if it works.  If so it'll confirm the issue.

dump below (this model is not completely setup yet)
[image]
:

# dump

# version
# INAV/MATEKF405 1.7.3 Sep  9 2017 / 10:51:01 (974440df)

# resources

# mixer
mixer FLYING_WING

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature GPS
feature CURRENT_METER
feature LED_STRIP
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD
[... 471 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

ShawnzAZ, 2017-10-08:
I moved the wire from the S4 pad to S2 and now I have two servos and the motor working.

It appears that this 1.7.3 beta for the Matek F405-AIO does the following:

S1 = Motor 1
S2 = Output 3 in the mixer diagram.  It is also 'Servo 3' in the Servos tab GUI.
S3 = Output 4 in the mixer diagram.  It is 'Servo 4' in the Servos tab GUI.

As-per the diagram, and how it was on the FC before the Matek F405 it should be:
S1 = Motor 1
S2 = Motor 2
S3 = Servo Output 3
S4 = Servo Output 4

[image]

MW2Stambo, 2017-10-23:
I was having issues trying to set up a Nano Talon on 1.7.3 
It may be a problem where the servo mix is not being accepted from the preset.
Your dump has no smix at all.
The smix format seems to have changed.
Old
smix 0 2 0 -100 0 0 0
New
smix 0 2 0 -100 0
You may have to use a custom mix.

closed 2018-04-25 by DzikuVx