iNavFlight/inav #2295: UART1_RX on NAZE32

junwoo091400, 2017-10-08:
Hello, I was trying to manually send serial signals w/ FTDI adapter.
I checked that '# get gps' was being sent through RX line when done via inav-configurator.
So I tried w/ Arduino Serial Monitor with 'same(115200) baud rate' => No Response.

It seems like there is an 'invisible' difference between Two...

My question is "Is communicating with NAZE32 UART1_RX pin even possible??"

junwoo091400, 2017-10-08:
This actually started with troubleshooting Why Bluetooth Signal to UART1_RX pin w/ EZ-GUI app was not working....

The App was sending bunch of MSP commands(checked manually w/ FTDI adapter), but FC was not responding!
-> And when I 'manually' turned the 'inav-configurator' ON w/ Laptop, EZ-GUI finally Received Battery voltage and such information...

: So my guess is that There are some **'invisible'** bytes that EZ-GUI must send, in order for FC to acknowledge the message, as like configurator Smoothly does so!

stronnag, 2017-10-08:
UART1 works just fine for MSP on NAZE. This is not an iNav issue.

junwoo091400, 2017-10-09:
@stronnag OK I will maybe try flashing another NAZE board!

closed 2017-10-09 by junwoo091400