iNavFlight/inav #2297: Change altitude alarm to always use meters internally

fiam, 2017-10-08:
Currently the altitude alarm has different meanings, depending on the OSD units. When using SI units, the altitude alarm is interpreted as meters, while setting the units to IMPERIAL interprets the alarm as specified in feet.

This design requires the code in osd.c to perform unnecessary arithmetic for converting the altitude alarm on demand (since it could be changed at any time) to the necessary unit, wasting a few cycles every time the altitude indicator is drawn. It also complicates the code without any good reason.

My proposal is changing it to behave like the new distance and negative altitude alarms. Always use meters internally and convert between meters and the imperial unit which makes more sense (feet for altitude, miles for distance) in the configurator UI.

PRs with the implementation for these two new alarms can be found at:

INAV: https://github.com/iNavFlight/inav/pull/2296
Configurator: https://github.com/iNavFlight/inav-configurator/pull/276

As you can see, doing the conversion in the UI requires just a bit of code and since we're already doing it for these new alarms, we might as well do it for all of them a save a few bytes of flash and cycles on the FC.

Feel free to just use thumbs up or down.

digitalentity, 2017-10-09:
I'm all in for this. Consistency is good.

fiam, 2017-10-12:
Fixed by #2296

closed 2017-10-12 by fiam