iNavFlight/inav #2299: magnetometer are detected but no oscillation found when tuning. (beerotor F4 & radiolink se100)

chan2328, 2017-10-09:
FC: beerotor F4,  identifier: INAV, version: 1.7.3
Running firmware released on: Sep 9 2017 10:49:35
Board: BRF4, version: 0
GPS: Radiolink SE100 (ublox M8030 with DA 5883 compass)

I can make sure that the wiring between FC and GPS is correct, and the GPS automatically detected and works fine, but the magnetometer 5883 are recognized by inav only it does not work. please see below pic. i have tried several things to do such as;
1. decreasing the i2c speed from 400khz to 200khz or 100khz. 
2. connecting the vcc from 5V to 3.3V.
3. connect magnetometer with vcc, scl, sda, GND only.
the results are the same, no oscillation found by mag. this is a trick when i turn the compass vertically, the oscillation will change from -0 to 0 only, and then the magnetometer will be lost. the SCL and SDA of FC are measure as 3.3V and the internal barometer works fine. i just brought them last week, is it a issue of inav, FC or mag? 

[image]
[image]
[image]

chan2328, 2017-10-09:
# bootlog
Time Evt            Description  Parameters
   2:  0          CONFIG_LOADED 
   2:  1       SYSTEM_INIT_DONE 
 502: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (7, 7, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (8, 8, 0, 2)
 502:  2          PWM_INIT_DONE 
 502:  3       EXTRA_BOOT_DELAY 
1188:  9         GYRO_DETECTION  (7, 0, 0, 0)
1779: 10          ACC_DETECTION  (8, 0, 0, 0)
1800: 11         BARO_DETECTION  (4, 0, 0, 0)
1800: 20        PITOT_DETECTION  (0, 0, 0, 0)
1800: 12          MAG_DETECTION  (2, 0, 0, 0)
3461:  4       SENSOR_INIT_DONE 
5170:  8           SYSTEM_READY 

# status
System Uptime: 725 seconds
Voltage: 0 * 0.1V (3S battery - OK)
CPU Clock=168MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280, MAG=HMC5883
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, GPS=NONE
SD card: Startup failed
Stack size: 2048, Stack address: 0x10010000
I2C Errors: 0, config size: 3318, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = none
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = none, used = none
System load: 0, cycle time: 2004, PID rate: 499, RX rate: 49, System rate: 9

chan2328, 2017-10-10:
i guess, that DA 5883 is QMC5883 compass. in INAV configurator UI there is no option for QMC5883. but found in CLI command.  
when i set FAKE magnetometer in UI, CLI show as below.
mag_hardware = QMC5883
Allowed values: NONE, AUTO, HMC5883, AK8975, GPSMAG, MAG3110, AK8963, IST8310, QMC5883, FAKE

when i set FAKE in CLI. UI show "blank" in magnetometer.

unfortunately, it does not work also.

McGiverGim, 2017-11-13:
Hi! I think that I've the same problem with the same product (Radiolink SE100) and Omnibus F4 pro (V3). I have two units of this product, the first worked without problem, but after two hours of use, something went wrong and the unit was dead.
I bought another, and the board is slightly different, now is like yours, and the MAG does not work too.
Did you find a solution for your problem?

McGiverGim, 2017-11-14:
@chan2328 if it helps, I finally make it working (using Betaflight, but I'm pretty sure that in iNav the solution will be the same). The chip is detected like HMC5883 and the only thing I needed to change in the code is the use of the gain. In the:
https://github.com/iNavFlight/inav/blob/912d131556302cf6ea67d59308e43dfe65ca8026/src/main/drivers/compass/compass_hmc5883l.c#L185-L187
Remove the magGain from here. The calculation of it is zero, and it will work.
A better solution will be to modify the calculation of the gain, but this is outside of my knowledge.

neyluzz, 2017-11-24:
Would be great if this problem is fixed. It's a great GPS and if we can make use of the compass it would be superb.

McGiverGim, 2017-11-24:
The easiest way is, if magGain is zero in some of the axis, then magGain = 1 for all axis. With this I think it will work.

neyluzz, 2017-11-24:
Thanks for your reply mate. How can I compile your code into a build for my Omnibus F4 v2? Thanks a bunch!

McGiverGim, 2017-11-24:
If it's only for you, simply remove the magGain from the above lines, ending like this:
```
 magDev->magADCRaw[X] = (int16_t)(buf[0] << 8 | buf[1]) ; 
 magDev->magADCRaw[Z] = (int16_t)(buf[2] << 8 | buf[3]) ; 
 magDev->magADCRaw[Y] = (int16_t)(buf[4] << 8 | buf[5]) ; 
```
And build it.

Arvycka, 2017-11-24:
Hi all. I recently received Radiolink SE100 unit from Banggood and I have the same problem as above - mag is detected as HMC5883 in Inav but outputs are 0 and calibration not possible. (However model still moves in Inav configurator if I rotate TS100 or put a magnet near it, only not accurately)
I have contacted Radiolink and they told me that black boards - like mine or in the picture from the first post  (they show blue boards without RF shield in their adds) are their latest units:
"Dear Sir
   your SE100 GPS is the newest version ,becuase the chip HMC5883 stoppde production recently, we have to change the chip.  
   the newest SE100 GPS can not use with F3 flight controller,  can use with APM and PIX flight controller."

So they cobbled up something from the QMC5883 chip to mimic HMC5883 chip and this "creation" works as neither of those.

Now I removed gain calculation and mag output is no longer 0, though I wonder if accuracy or something else is not  affected this way?

Other than that GPS works very well in this unit and they also released a smaller version (TS100 mini) which claims to have QMC5883 mag. I wonder if they screwed this one up as well.

neyluzz, 2017-11-24:
@Arvycka would you be willing to help me create a build that fixes this issue? I don't have the knowledge to create a build. Thank you

Arvycka, 2017-11-24:
Which target?

neyluzz, 2017-11-24:
OmnibusF4PRO

Arvycka, 2017-11-24:
Building. Can I upload it here?

neyluzz, 2017-11-24:
You can upload it to my NAS if you can :) https://remote.lijzenga.biz:5001/sharing/WtcBwBKSo

Arvycka, 2017-11-24:
https://drive.google.com/file/d/1mtcN4J8CPb8gDNTWP2XShEhXYqcABBjE/view?usp=drivesdk

neyluzz, 2017-11-24:
@Arvycka thank you so much!!

Arvycka, 2017-11-24:
I also ordered Radiolink TS100 mini (before I found this thread). It's much smaller and lighter. I will report the results once I receive and test it.

McGiverGim, 2017-11-24:
Very interested in your results with the mini. The SE100 works perfect, if the mini is similar, will be great for our minis.

neyluzz, 2017-11-24:
That would be great. It is €13 euro's right? I've tested your build but I seem to have problems with my barometer now.

Arvycka, 2017-11-24:
Yes, thes Radiolink units have very attractive prices.
Hm...I have not changed anything else. My onboard BMP 280 baro works fine on Omnibus F3. 
This was my second build to be honest :-)

neyluzz, 2017-11-24:
Strange... my board has a BMP280 as well

Arvycka, 2017-11-24:
Here are the instructions for building in windows if anyone needs:

https://github.com/Arvycka/inav/blob/40efa4200e3f3cd28764a59b1b4835d89e5df189/docs/development/Building%20in%20Windows%20light.md

chan2328, 2017-11-25:
@McGiverGim I also asked RadioLink. they replied the built-in mag changed from HMC5883 to QMC5883, it works with their flight controller which is called PIXHAWK. I don't know the logic of programming, it is out of my knowledge. Finally, I just bought a new HMC5883 instead of QMC5883(SE100 built in mag), and the mag works.

Thanks for your help, you found out the solution of it, is it work prefect? but i have two questions of it.
1. can QMC5883(SE100 built in mag) be detected by Betaflight now? if yes, it is work without changing the code?
2. how can I change the code for my quad using inav? edit like
 magDev->magADCRaw[X] = (int16_t)(buf[0] << 8 | buf[1]) * magGain[X]; 
 magDev->magADCRaw[Z] = (int16_t)(buf[2] << 8 | buf[3]) * magGain[Z]; 
 magDev->magADCRaw[Y] = (int16_t)(buf[4] << 8 | buf[5]) * magGain[Y]; 
or 
 magDev->magADCRaw[X] = (int16_t)(buf[0] << 8 | buf[1]) ; 
 magDev->magADCRaw[Z] = (int16_t)(buf[2] << 8 | buf[3]) ; 
 magDev->magADCRaw[Y] = (int16_t)(buf[4] << 8 | buf[5]) ; 

@digitalentity could you help to add this issue to 1.9. please? or may be add a new mag like DA5883 in the list of mag_hardware? Thank you very much.

chan2328, 2017-11-25:
actually my quad does not work with GPSHOLD, when i turn GPSHOLD on, the quad will drift, after few seconds become worse, even it has 15 satellites signal. i disappointed and boring now  :(

McGiverGim, 2017-11-25:
@chan2328 no, betaflight needs more changes to make it work, so we decided by the moment not include this changes in the code. I've it working on my quad, with a custom firmware, and the changes needed have been explained in the betaflight issue.
When you say that yours is drifting, are you talking about the original firmware or with a custom firmware with the changes needed (without gain)?

chan2328, 2017-11-25:
@McGiverGim i am using original firmware which is INAV version: 1.8.0 firmware for beerotor F4 board. when I hold the quad by manually 5 meters above the ground, it become solid. then I turn GPSHOLD on, it will start to drift and few second later become worse. I have to turn GPSHOLD off, otherwise it will crash.  . At the moment, I don't know why. :(   altitude hold & heading hold are work very well.

could you help to compile the changes(for QMC5883 mag) into beerotor F4 firmware to be custom firmware for me, please?

McGiverGim, 2017-11-25:
@chan2328 I've tried to build the firmware for you, but I've an error. I've build other targets without problem, but your target gives me an error. I usually work with betaflight, so I don't know too much about iNav builds, maybe other user can help you, sorry.

Arvycka, 2017-11-25:
Try this.

https://drive.google.com/file/d/1D2tpXcL6u8jmZveBksB2gTzaEjmVpngM/view?usp=drivesdk

I have not changed anything except hmc5883 driver (removed gains), so if something doesn't work for you, I have no idea why :-)

hugoget, 2017-11-25:
Hi, I'm having the same problem with a Spracing F4evo card and a GPS RadioLink SE100. Do you have a solution for the F4 evo card to make the magnetometer work ?

Arvycka, 2017-11-25:
@hugoget 
"The chip is detected like HMC5883 and the only thing I needed to change in the code is the use of the gain. In the:
inav/src/main/drivers/compass/compass_hmc5883l.c
Lines 185 to 187 in 912d131
 magDev->magADCRaw[X] = (int16_t)(buf[0] << 8 | buf[1]) * magGain[X]; 
 magDev->magADCRaw[Z] = (int16_t)(buf[2] << 8 | buf[3]) * magGain[Z]; 
 magDev->magADCRaw[Y] = (int16_t)(buf[4] << 8 | buf[5]) * magGain[Y]; 

Remove the magGain from here. The calculation of it is zero, and it will work."

Here are the instructions for building in windows:

https://github.com/Arvycka/inav/blob/40efa4200e3f3cd28764a59b1b4835d89e5df189/docs/development/Building%20in%20Windows%20light.md

hugoget, 2017-11-26:
Thanks for the information. By cons when I type the command "C:\devtools\shF4.cmd"
, I get this message :
"some tools were not found in PATH or expected location : git.exe 
You may want to install them and/or update pa"

chan2328, 2017-11-26:
@Arvycka Thank you for helping me. I will try the firmware later.

Arvycka, 2017-11-26:
@hugoget 
I'm sorry it didn't work for you. I'm not a developer. I followed the tutorial here:
https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows%20light.md

But it's outdated (obsolete arm toolchain version and  doesn't mention where to install Ruby.) I don't know how to update it. 

If you've done everything correctly your devtools directory should look like this:
[image]

I'm on windows 7
and Drive C: like so:
[image]

Feldsalat, 2017-11-26:
Windows light... :
https://github.com/iNavFlight/inav/pull/2280

chan2328, 2017-11-27:
@McGiverGim I tried the firmware by @Arvycka , there is no I2C error, the reading of sensor tab is quite solid and sensible, it did very well. but the reading of heading is little bit difference from sensor tab.  When I turned the quad 90 degree horizontally, sometimes the reading of heading is reaching 90 degree within 1 second. but sometimes the reading become solid within 4 second. what should i do to improve the sensibility of heading? otherwise i don't know what will happen in the sky.

@Arvycka Thanks for introduce the windows light, but what is the next step? I follow the introduction, just cloned the inav. any tutorial is for changing, editing the code, and how can I make the custom firmware by myself? for example how to remove the code as you did, and produce the firmware for me.

chan2328, 2017-11-27:
[image]
[image]

Arvycka, 2017-11-27:
@chan2328 I have noticed such behaviour myself while testing on the bench yesterday - operation of mag is strange, it takes a long time for the heading to settle after a 90 deg rotation. I have a standalone HMC5883 and it works just fine and smooth in the same environment. I suspect that removing gains only solves part of the problem.

Regarding editing source - go to "inav/src/main/drivers/compass/compass_hmc5883l.c" inside your devtools directory and open it with Notepad++ or another advanced text editor, find lines 185-187 and edit them.
It's just for this case, I have no coding knowledge whatsoever.

digitalentity, 2017-11-27:
If heading drifts a lot after a 90 deg rotation it could be caused by several reasons:
1) Compass sensor misaligned
2) Compass sensor mis-calibrated or uncalibrated
3) Compass sensor malfunctioning

@chan2328 in your case it's likely uncalibrated compass. Please calibrate and verify.

chan2328, 2017-11-27:
@digitalentity after some tried, one finding is that when i turned the quad from 0 degree to 90 degree,  the reading of heading would be like this,  0 > 270 >180> 90 degree or sometimes 0 > 270 > 0 > 90 degree. even calibrated. although it takes 5 seconds to point the reading, the final reading is correct and solid if i don't turn any more.
[image]

@Arvycka ok, i understand, it also out of my knowledge.

digitalentity, 2017-11-27:
What's the value of your "mag_align" parameter and where your compass
sensor is located?

chan2328, 2017-11-27:
@digitalentity here is CLI, the align_mag is default.
Entering CLI Mode, type 'exit' to return, or 'help'
# set mag
align_mag = DEFAULT
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP

mag_hardware = HMC5883
Allowed values: NONE, AUTO, HMC5883, AK8975, GPSMAG, MAG3110, AK8963, IST8310, QMC5883, FAKE

mag_declination = 0
Allowed range: -18000 - 18000

magzero_x = 150
Allowed range: -32768 - 32767

magzero_y = -224
Allowed range: -32768 - 32767

magzero_z = -17
Allowed range: -32768 - 32767

mag_calibration_time = 30
Allowed range: 30 - 120

imu_dcm_kp_mag = 10000
Allowed range: 0 - 65535

imu_dcm_ki_mag = 0
Allowed range: 0 - 65535

inav_auto_mag_decl = ON
Allowed values: OFF, ON

attached is the compass & GPS mounted on the frame above 10cm. 
[image]

digitalentity, 2017-11-27:
Most likely you need a different `align_mag` value. In GPS units compass is usually upside down meaning that you'll need `*_FLIP` value.

Arvycka, 2017-11-27:
Looking at the picture above the mag X axis is now pointing forward, Z axis is pointing down, So one would need to use "0 deg and flip" for the mag alignment value.
[image]

I have used 270 and flipped, it is pointing to North correctly, and increased heading degrees when rotated clockwise, still it takes a long time to settle if rotated quickly. 
Does it matter if alignment is set before or after calibration?

chan2328, 2017-11-27:
@digitalentity @Arvycka Oh, yes. after changing the MAG alignment to CW 180 flip the problem is solved. it becomes sensible and pointing to correct direction as fast as turning the quad. I will try to fly later.

chan2328, 2017-11-28:
@McGiverGim @digitalentity @Arvycka Great! it works very well either in POSHOLD mode or RTH mode. Although it is a little bit of drifting on POSHOLD, it is acceptable, also it did very well on RTH mode. so is it possible to add the new selection of mag hardware (which is deleted the mag again) for future firmware as well. Thank you so much for your help.

andrejpodzimek, 2017-12-20:
Thanks to @McGiverGim I've also made it work on iNav, [using this hack](https://github.com/andrejpodzimek/gammaflight/commit/9b02f455c7bf54c69f232c3b008bdc9f7664e4fa). More details are [in a related Betaflight issue](https://github.com/betaflight/betaflight/issues/4531#issuecomment-352956851).

digitalentity, 2017-12-20:
Radiolink SE100 is on the way from Banggood. Once it's here, I'll make a proper fix.

kristoferahlqvist, 2017-12-20:
Crap I just ordered radiolink ts100!
Should I cancel and order another gps?

alanfj, 2018-01-19:
When will this fix and what release will this be on? I too have 2 of these Radiolink SE100's and having the same issues on INAV. I am excited that you guys have hacked the problem. Looking forward to the release.

chan2328, 2018-01-19:
@alanfj I believe @digitalentity  will make a proper fix, the day is coming soon.

digitalentity, 2018-01-21:
@chan2328 @alanfj my SE100 is a few days away from me. Will look into properly fixing this as soon as I get it.

digitalentity, 2018-01-28:
Also #2699 will attempt QMC5883 detection before HMC5883 to avoid possible misdetection

alanfj, 2018-01-28:
digitalentity, this is wonderful news! I have two large FPV photography quad drones waiting for this great new (RadiolinkSE100 compass) update for the OmnibusF4Pro board. Looking forward to the new addition! Once again, thanks!

alanfj, 2018-01-28:
> My Radiolink SE100 GPS arrived. Compass is correctly detected as QMC5883 and works correctly. @chan2328<https://nam04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fchan2328&data=02%7C01%7C%7Cf3cb068919784724e1db08d5661aada1%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636527187118833443&sdata=0bfBndpWm5tnUKAkQFukHHWKxhHCVe%2BnGeVxi%2Fm9cbI%3D&reserved=0> @alanfj<https://nam04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Falanfj&data=02%7C01%7C%7Cf3cb068919784724e1db08d5661aada1%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636527187118833443&sdata=7pJA%2B8ifdlRazNJLwL3wsfV8hMfrgDWvW4nId%2BTBwq0%3D&reserved=0> what board are you using? I'll provide a hex built from latest development branch and we'll check if it works for you.

Konstantin, digitalentity, this is wonderful news! I have two large FPV photography quad drones waiting for this great new (RadiolinkSE100 compass) update for the OmnibusF4Pro board. Looking forward to the new addition! Once again, thanks!

chan2328, 2018-01-28:
@digitalentity this is great news! I am using BeerotorF4 board.

Arvycka, 2018-01-30:
Received my Radiolink TS100 today, only slightly more than 2 months after ordering it :)
It's quite a bit smaller: 
[image]
It has the same magnetometer chip as SE100 (marked DA 5883) but here it doesn't pretend to be HMC5883 - it only works if set to "FAKE" in iNav. It also gets autodetected as "FAKE" and works right away. Only problem - it fails randomly for short periods of time, and sometimes permanently - until a reboot. Sometimes with I2C errors, sometimes without - no correlation. Tried reducing I2C speed to 100kHz - it's still failing once in a while. Mine's probably faulty :(
GPS appears to work OK, haven't tried it in the open field. Sees 8 sats at home. Refresh rate 5Hz.

digitalentity, 2018-01-31:
@Arvycka so the big GPS/Compass mimics HMC5883 and doesn't work, right? I wonder how many of this flawed GPS units are out there...

Arvycka, 2018-01-31:
@digitalentity my SE100 is recognized as HMC5883 and only works with the driver modification discussed above. They've told me they'd done that on purpose since HMC5883 had been discontinued. Maybe later they've changed that since yours is detected normally as QMC5883 and so does my TS100.

McGiverGim, 2018-01-31:
There're two variants of the SE100, one with the real HMC and the other with the QMC trying to mimic the HMC.
In the variant with QMC, it does not work because fails calculating the MAG GAIN. According to RadioLink it works for APM and PixHawk, so I suppose that these two systems didn't use the MAG GAIN or make it in a different manner.

digitalentity, 2018-01-31:
I wonder where I can get a variant with QMC mimicing the HMC chip. I think we need a proper fix for this.

[27 more replies]

closed 2018-02-21 by digitalentity