iNavFlight/inav #2301: Wrong throttle value in NAV LAUNCH mode.  Inav 1.7.3

Feldsalat, 2017-10-09:
Starting the plane with NAV LAUNCH mode and flight mode set to HORIZON + ALTHOLD makes the motor spin only with "nav_fw_cruise_thr " value.
NAV LAUNCH without ALT HOLD mode lets the motor spin with "nav_fw_launch_thr" as it should.

Video:
[https://youtu.be/nFNhnC0H_4s](url)

Diff and log:
[launch fail.zip](https://github.com/iNavFlight/inav/files/1369677/launch.fail.zip)

fiam, 2017-10-09:
Just edited your comment to fix the link.

digitalentity, 2017-10-10:
@Feldsalat from the log I can see that your plane was not in launch mode at the moment of arming. Did you activate launch mode switch before arming?

digitalentity, 2017-10-10:
In your log from first post I see no traces of enabled launch mode. According to `flightModeFlags` field you only had `ARM`, `HORIZON` and `ALTHOLD` box modes enabled on your transmitter.

Feldsalat, 2017-10-10:
?? the motor starts after throwing as expected in launch mode...
Without launch mode the motor would have spinned up directly after applying throttle!?
Now tested a second time and it worked as expected. I don't understand what happened yesterday... ?

Feldsalat, 2017-10-10:
and... STAB was shown in OSD!

digitalentity, 2017-10-10:
@Feldsalat so - most likely a user error. Can we close this?

Feldsalat, 2017-10-10:
INAV behaved like in launch mode ... if there is no explanation I would not just ignore it.

digitalentity, 2017-10-10:
Log says otherwise - it was AltHold from the moment of arming.

digitalentity, 2017-10-10:
#1938 might have helped debugging tho - it would show what actual stick positions were.

Feldsalat, 2017-10-10:
"Log says otherwise..." 
Yes I understand but INAV behaved like in launch mode!  If i made a fault it should not show "STAB" on OSD and behave like in Launch mode. This time it only was a 120g plane...

"it would show what actual stick positions were"
Throttle stick is shown in OSD and the not spinning propellor at 78% throttle can be seen in video too.
When i touch the sticks while in launch mode the flight mode in OSD changes immediately.
In the video you can see the flightmode stayed in "STAB".

Whatever happened... I would like to know why.
I understand that this is difficult to debug... 

"Can we close this?"
If you think so...

digitalentity, 2017-10-10:
It showed `STAB` because AltHold mode on a plane forcibly enables stabilization (ANGLE mode).

digitalentity, 2017-10-10:
You know what's interesting - there's no sign of throw in the logs. It looks like you armed when already flying - after the throw. Also, first half a second after arming throttle command is 1550 (commanded by AltHold) while motor is stopped - this means that throttle stick was at zero.

[image]

What transmitter/recevier are you using?

Feldsalat, 2017-10-10:
"It showed STAB because AltHold mode on a plane"
What I see in OSD:
no flightmode -> AIR
Horizon -> HOR
Horizon+Althold -> AH
Launch -> STAB (before throw and then furthermore five seconds)
RTH -> RTL
Poshold -> PH
...

"What transmitter/recevier are you using?"
Frsky Taranis X9D Plus with  XSR via SBUS

digitalentity, 2017-10-10:
This is odd. Log clearly indicates altitude hold being active. Is it possible that I'm looking ar wrong log?

Feldsalat, 2017-10-10:
Excuse the waste of your time. It was the wrong log :-(   Sorry!

No signal is recorded for the motor in the correct log. I'm surprised a bit ...
I could hear the engine running clearly. Motor turned up (not full speed only about half throttle) and then slowly decreased  speed. In the video you can also see the motor was running.
Video with correct log:
[Video](https://youtu.be/yKWID1lk9nM)

I think this issue can be closed. But why was the motor running, without a throttle signal was recorded in log? Thought that would not be possible...?

digitalentity, 2017-10-11:
@Feldsalat is there a correct log file? I'd like to see complete INAV's internal state.

Feldsalat, 2017-10-11:
I'd forgotten about that...
[LOG00031.TXT](https://github.com/iNavFlight/inav/files/1374541/LOG00031.TXT)

digitalentity, 2017-10-11:
Yes, this looks like a correct log.
We see the propeller spinning in the video without throttle signal probably because of the airflow - it's air spinning the propeller.

The throw was perfect, however the pitch angle of the plane was 48 deg - you pointed plane's nose too high in the air. There is a built-in safeguard that prevents launch from detecting a throw when the plane is tilted more than 45 deg (default for `nav_fw_launch_max_angle`). You can adjust it to something suiting your launching style - maybe 60 deg?

Bottom line - configuration issue. Launch mode behaving as configured.

Feldsalat, 2017-10-11:
If I had not heard the motor run... but I have.
This all seems to be impossible ... :-(  I would also not believe it.
The speed of the propeller is much too high. This was not only by airflow.
It is impossible to let the motor turn only by throwing the plane. I tested it several times today...

Some minutes bevor with a spinning propeller by airflow:
[airflow](https://youtu.be/4xRf-S3p4ho)

and at higher ground speed:
[motor stops](https://youtu.be/YqiX5u8760g)
All within some minutes on a calm day with low winds.

I'm afraid we will not find the problem. 
I have now tested it several times with the airplane in the hand and it always worked.

However, I have changed a setting today.
Old:
set align_board_roll = 16
set align_board_pitch = 900
set align_board_yaw = 2700 
New:
set align_board_roll = 900
set align_board_pitch = 870
set align_board_yaw = 1800 

The problem was that I could not adjust the pitch orientation for Horizon mode.
In Setup Window nothing changed for "pitch deg" and in OSD crosshair stayed above Horizon.
In the plane the board is pointing up a bit, 3-5 degree I think.
This was certainly part of the problem -> "pitch angle of the plane was 48 deg - you pointed plane's nose too high in the air."
Now roll cannot be adjusted, but this is not a problem.

digitalentity, 2017-10-24:
@Feldsalat I believe the original issue is resolved - we confirmed by logs that launch mode behaved exactly as configured.

Feldsalat, 2017-10-24:
I can not contribute anything new. So, from my point of view, we can close the issue.
@digitalentity 
Since the wrong pitch angel contributed to my problem ... and before I open unnecessary a new issue:
Am I wrong with the setting of the board alignment? See end of last post...

digitalentity, 2017-10-24:
As long as firmware is correctly calculating attitude - your board alignment is set correctly. There's always more than one solution, especially if your board is mounted upside down.

closed 2017-10-24 by digitalentity

Feldsalat, 2017-10-24:
> As long as firmware is correctly calculating attitude - your board alignment is set correctly.

In the Horizon mode, the aircraft flew not straight but clearly downward. In this case, the alignment can not be changed is really a bad thing. 
I know this problem now, but others might also get problems with it.