iNavFlight/inav #2302: Target Confusion Omnibus

domhan, 2017-10-09:
Hello,
sorry for the Question. I'm really confused about the Omnibus Targets. The Wiki, the Doc and the realese are different. In my Situation. 

Omnibus F4 Pro AIO v3... what is the right target? Omnibusf4v3 or Omnibusf4pro?

Please help me - and if possible - please update the doc.

Thank you and best regards
Gregor

DzikuVx, 2017-10-10:
@domhan please understand that we are not in possesion of any new version of any flight controller. And versions of Omnibus F4 and created very frequently. V3 should use V3 target. OmnibusF4Pro is reserved for v2 hardware.

https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20Omnibus%20F4.md explains everything

Plus, there is no "official" v3 Pro. V3 Pro is only an another clone-with-changes that was created by banggood without any support

closed 2017-10-10 by DzikuVx

OhmEye, 2018-06-13:
8 months later the target confusion isn't getting any simpler. :) Just to confirm, now that there actually is an official Omnibus F4 Pro (V3) is the correct target for it still OMNIBUSF4PRO ?

shellixyz, 2018-06-13:
These targets are basically the same. I think the main difference is the presence of an inverter for UART6 on F4V3. If when you are using the RC input with SBus it is not working switch to the other target.