iNavFlight/inav #2304: BlackBox data downloading takes too much time : 10+ minutes ...

junwoo091400, 2017-10-10:
Hello, I have NAZE32 for quadcopter. And I use '2MB Memory On Board' to record my data. 
The Problem is, even with only about 630kB of logged Data, it takes more than 4 minutes to download it via inav-configurator ...

When I have FULL 2MB data, it takes more than 10 minutes,,, 
I think that with '115200(bits/s) = 14.0625[KiloBytes/s] or Slower' , it should only take 
2*1024/14.0625 = 146 Seconds... Even with FULL Memory usage!

=> I know this assumes Direct-Reading from onboard chip, but it seems odd that result is 4~5x slower...

So, why is it so Slow, and how can it be improved?

digitalentity, 2017-10-10:
Which version of the firmware and Configurator are you using?

junwoo091400, 2017-10-10:
@digitalentity 
on QUAD : 1.7.2 // using this because 1.7.3 didn't show onboard 2MB memory, but 1.7.2 did.
on Configurator : 1.7.3

Just looking back at it now,,, maybe 1.7.3 on configurator side didn't match my quad's, and this also explains Wild Behavior of my Blackbox Replay video
https://www.youtube.com/watch?v=ao-gsP8AkRs
(Quad Graphic on the Top Left is WILD... probably blackbox was not correctly interpreted by configurator...)

DzikuVx, 2017-10-10:
@junwoo091400 I'm affraid this will be a no-fix. Naze target reached end-of-life and will not be available in next release on INAV. Users of Naze will have to stick to 1.7.3 or older versions.

This is why I'm closing this issue

PS. 1.7.3 should be fully functional on Naze.

closed 2017-10-10 by DzikuVx