iNavFlight/inav #2305: When doing AltHold+Poshold, and Throttle goes Low, motor goes IDLE and Drop.

junwoo091400, 2017-10-10:
Hello, I was performing AltHold + PosHold.
1) AltHold is Perfect. When I put middle throttle, it holds altitude really well. (I have set 'use_midthr_for_althold')
2) PosHold, on the other hand, especially in this case, I was using 'On-board magnetometer', Yes I know it is bad, but it worked when it was facing North, probably because magnetic field strength was strong enough to not mess-up it's Heading Values. So it in fact worked quite flawlessly.

=> But the problem I encountered was this:
I was lowering altitude with (ALTHOLD+POSHOLD) activated, by 'lowering my throttle stick'.
Then, quad just fell out of the sky.
VIDEO (https://www.youtube.com/watch?v=72na7eh8WTo)

After some investigation, it got down to 2 possible reasons
1) POSHOLD got messed up because quad was slowly yawing towards East, which might have caused false Heading Values
2) My Throttle Command for lowering the altitude was so low that it caused Disarm.
----------------------------------------------------------------------------------------------
But Later, I found this : 
[image]

When 'blackbox log' was opened with Hex Editor, it said "DisArm Reason = 2"

I traced this message all the way back into 'fc_core.c'. And it was evident from declaration from 'fc_core.h' that 2 meant DISARM_STICKS. 

So the Flow could've been (must have been)

'rc_controls.c' on line 212, activated mwDisarm(DISARM_STICKS) => set lastDisarmReason too.
'fc_core.c' on line 317, activated blackboxFinish()
'blackbox.c' on line 1051, activated blackboxLogEvent(FLIGHT_LOG_EVENT_LOG_END, NULL)
'blackbox.c' on line 1472, activated blackboxPrintf("End of log (disarm reason:%d)", getDisarmReason()),
'fc_core.c' on line 329, Returned 'lastDisarmReason', which was : 'DISARM_STICKS'.

And there , is where the Last Line of my Blackbox  (disarm reason:2) came from.
-----------------------------------------------------------------------------------------------
[Crashed_Yes.TXT](https://github.com/iNavFlight/inav/files/1370371/Crashed_Yes.TXT)
I am uploading my Blackbox File too. Please note that probably because of difference in version of my quad & configurator, attitude & gyro and some other things were not interpreted properly. But You can clearly see the last Line and some other things.

**So to summarize,
 I think the program initiated 'DISARM', because my throttle command was too LOW, which of course was intention to Lower the Altitude.
 If quad was really high up in the sky, and user wanted it to go down pretty fast, he/she will command very low value on throttle channel. This will cause copter to DISARM. 

=> Can this be fixed or can you guys come up with other reasons for my situation????

Thanks.

digitalentity, 2017-10-10:
From your log. 
Flight #1: ACRO mode. Copder disarmed by stick combo (throttle low + yaw left). Perfectly valid.
Flight #2: 3D poshold mode (Alt + Pos). No disarm reason, log is truncated - the disarm event is not recorded in the log.

Also, in your video blackbox log probably doesn't match the video:
[image]

junwoo091400, 2017-10-10:
Oops ;;
You are correct ... I didn't know I could select Log(2/2) in BlackBox Explorer .... Damn :(

But just a question, is it possible for quad to DISARM in midair, in ANY circumstances, if Throttle Command stays low for some time? (I found out that 2nd Log was cut in during hover.. so no crash log)

=> If so, wouldn't it be more safe to disable that? (Don't allow disarm while ALTHOLD)

digitalentity, 2017-10-10:
If you are using stick arming & when throttle is staying low disarm is possible in one of two situations:

1. You use MOTOR_STOP and throttle stays low longer than `auto_disarm_delay` seconds
2. You move YAW sitck to the left thus executing disarm stick combo

digitalentity, 2017-10-10:
>If so, wouldn't it be more safe to disable that? (Don't allow disarm while ALTHOLD)

Disabling possibility to disarm is a BAD idea. Imagine something going wrong - you'll have to firstly disable AltHold and only then attempt disarm. This is not a good situation from safety perspective.

junwoo091400, 2017-10-10:
@digitalentity 
So does that mean : if MOTOR_STOP is disabled, which is my case, I can't disarm the quad just because I left throttle-command LOW for a long time? <Clears out One Possibility of my Crash>

digitalentity, 2017-10-10:
No, if MOTOR_STOP is disabled, low throttle alone won't disarm. You have to attempt left yaw. Probably this is what happened - you need to hold yaw left only a for a few moments (~400ms) to disarm.

junwoo091400, 2017-10-10:
Thanks! Helped me alot!

closed 2017-10-10 by junwoo091400