iNavFlight/inav #2310: Servo Throws Restricted In Non-Stabilized Modes

CradyLab, 2017-10-10:
Does anyone know why servo throws are restricted even in Passthru/Acro mode?  Starting with the Z-84 Wing template, I built a Sky Shadow around an Omnibus F3 AIO and a pair of Hitec digital servos and after getting everything calibrated (_gyros, Rx et al_) I notice the control surfaces weren't deflecting nearly as far as I'd assumed they should when not in a stabilized mode (_ie Passthru/Acro_). 

To confirm the outputs were being limited I wrote a short routine for Arduino that compares the PWM signals being sent to the servos.  Even with INAV indicating the Rx Pitch & Roll channels sweep the full range from about 980 to 2010 (indicated in the Receiver tab), the actual range of the servo outputs are reduced to roughly 1200 to 1800.  These values were measured directly from the FC pins with the servos removed.

giacomo892, 2017-10-10:
Higher the D part of ROLL/PITCH and look at the elevons

CradyLab, 2017-10-10:
Thanks.  I didn't realize the PIDs had any effect when in Passthru.  Is that documented because I never saw any mention of that

digitalentity, 2017-10-10:
PIDs don't have any effect in PASSTHROUGH. However mixer does. Since 1.7.3 default flying wing mixer reduced pitch/roll weights to 50% to prevent servo saturation in some maneuvers. If you want old behavior - please install custom smix for a flying wing.

DzikuVx, 2017-10-10:
@CradyLab in mixer each axis has only 50% authority on a servo in flying wing mixer.
Than means, full roll stick input will result only in 50% or servos movement. 
Full servo deflections are possible only on max roll and max pitch. This is intended. 
What I wrote applies to all  all modes, including Passthrough

DzikuVx, 2017-10-10:
Or increase Servo rate to 125%, but standard mixer works just fine

CradyLab, 2017-10-10:
**@DzikuVx** Out of curiousity I tried increasing the servo rate to 125% but that only add about 100 to each end - still well short of the full range.  That said, I won't mess with anything if the limits are there by design.  I haven't flown it yet and just wanted to be sure what I was seeing was normal ; )

Thanks!

digitalentity, 2017-10-10:
Closing then

closed 2017-10-10 by digitalentity