iNavFlight/inav #2312: Omnibus F7 (Banggood) onboard barometer (BMP280) not detected

swemonstro, 2017-10-10:
I recently got a F7 omnibus from banggood, where there should be (and physically are mounted on the board) an BMP280 barometer, but I cant get any values from it. 

Im very new to iNav, but I think I managed to flash the board to version 1.7.3 

Any suggestions where to start? and what kind of information could I include here (and how) to make it easier to find the problem.

Sincerely yours

Carl from sweden

DzikuVx, 2017-10-10:
I falshed INAV to my Omnibus F7 (original, not BG clone) and it worked fine. Baro detected and functional.

digitalentity, 2017-10-10:
@swemonstro please provide a link to the board you purchased.

swemonstro, 2017-10-10:
Should have bought a real one (wasnt aware it was a clone when I bought it) 

No other changes, norhing under ports or such?

DzikuVx, 2017-10-10:
https://quadmeup.com/hands-on-omnibus-f7-flight-controller/

No changes were required, this baro is SPI connected. Either we again have a clone that changed something or board is damaged

swemonstro, 2017-10-10:
The link to the board:
https://www.banggood.com/BF3_2-Omnibus-F7-Flight-Controller-STM32-F745-Integrated-OSD-Built-in-5V-2A-BEC-Dual-Sensors-p-1177839.html

I tried both set i manual and auto, but no luck

michaelvanderhoek, 2017-10-10:
thats weird. maybe try flashing an older version of Inav, did you flash it in DFU mode?

swemonstro, 2017-10-10:
I did flash it in DFU, trying right now to do that, but now it get stuck at 'erase'

trying to reinstall driver thru zadiq

swemonstro, 2017-10-10:
and after that I cant reconnect at all. the computer dont even recognize the board as stm32, just Unknown USB Device (Device Descriptor Request Failed)

think my board is toast...  :(

DzikuVx, 2017-10-10:
@swemonstro it's not that easy to fly a STM32. Probably it's only a driver problem

giacomo892, 2017-10-10:
Try to put it into bootloader and flash again with a different firmware and using "no reboot sequence"

swemonstro, 2017-10-10:
In back again after some problems getting back into DFU.

there are only one firmware for omnibus F7. so not much to choose from unfortunately. or do you suggest trying another board?

michaelvanderhoek, 2017-10-11:
try to flash it again in DFU mode. make sure you ticked the box: no reboot sequence. 
Maybe it will work. if not then i think its a broken board

swemonstro, 2017-10-11:
I got the card back in DFU mode last night, and flashed it again. (but not sure if I ticked the 'no boot sequence' - what does that do?)

So now the flight controller works as Before, with acc. and so, but still no barometer.

digitalentity, 2017-10-11:
My best bet is that it's a faulty board. Original Omnibus F7 works fine

swemonstro, 2017-10-12:
Banggood asked be to document the behavior on video, so I did. Perhaps anybody can see if I do anything wrong.

https://youtu.be/UiZz_-FuWUs

DzikuVx, 2017-10-13:
Like I wrote in the other ticket, I flashed Omnibus F7 yesterday, everything seemed to be working fine, including barometer. So sorry, but looks like INAV has nothing to do with your problems and it seems to be hardware issue.

BTW, I watched the video, and I do not see anything wrong with your setup

closed 2017-10-13 by DzikuVx

swemonstro, 2017-10-13:
Thank you. no I also believe that it's the board that's the problem.

Thank you for the support!

Arvycka, 2017-10-19:
I've just received F3 Pro from AliExpress - Barometer on it doesn't work either. Probably doesn't work on any of the clones.

giacomo892, 2017-10-19:
@Arvycka Try 1.8RC

Arvycka, 2017-10-19:
Thanks, tried that - still no go for baro

digitalentity, 2017-10-22:
@Arvycka please provide a link to a board. I'll try to get one and tackle the baro issue.

Arvycka, 2017-10-22:
@giacomo892  I have since received an identical board from another supplier and baro worked right away on it - so it's a hardware issue. That's what I get for being a cheapskate. Thanks for your help and sorry for posting to a closed thread.

swemonstro, 2017-11-12:
Just an update, I recieved another board from bangood (same omnibus F7) and the Barometer worked instantly, so it was a faulty board.

thanks for the support!!

digitalentity, 2017-11-12:
@swemonstro thanks for the update!