iNavFlight/inav #2318: Suggesting a new RTH mode : SHORTEST / DIRECT / STRAIGHT

OlivierC-FR, 2017-10-11:
What about a RTH mode that would make the aircraft come back using the shortest possible path.

My practical case is as follow : With my flying wing, I'm 5000m away, 1000m higher.
In case of failsafe, if nav_rth_alt_mode is set to AT_LEAST or MAX or CURRENT or EXTRA, the plane will come back very high above me, and I may not be able to see it / hear it / visually take control back.
Auto-landing would then bring the plane to ground level but that may take quite a long time.

In such case, the only mode that actually works is the FIXED altitude one, with a rather low allowed dive angle for nav_fw_dive_angle, in the 3-8° range as example, so he plane does not hit the mountain on its way back by diving too hard. That's why I suggest a RTH mode where the plane would directly aim for the point above home. Straight line between current position and "home + nav_rth_altitude".

Could make the RTH sucessfull in case of motor failure, because the plane would glide. Also in case of battery low, because flying down a constant slope is way more energy efficient than flying high in the sky and then later spiral down. For the rare cases where the plane is lower than the pilot, slowly climbing is also more efficient than climbing hard to reach a given altitude and then continue horizontal.

junwoo091400, 2017-10-11:
That would be a good Enhancement for Sure!
Your Idea can be combined with #2270 Issue.

Dronek, 2017-10-11:
> ... the plane will come back very high above me, and I may not be able to see it / hear it / visually take control back.

I ran into a similar situation lately, so I totally support this idea.
However, with a constant dive angle the slope is heavily influenced by wind:
a) With headwind the plane will lose altitude too quickly as a function of distance.
b) With tailwind the plane will lose altitude too slowly relaitve to home distance.

In a) it might descent to a defined "home altitude" way before it reaches home while in b) it might come home at an altitude way above the defined "home altitude".

In MWOSD there is a very useful function called "glidescope". It's basically a representation of the vertical angle between homepoint and the aircraft. For example, if plane travels home at a constant altitude, this angle will obviously increase.

Based on your idea I suggest to implement a mode that keeps this angle constant. It can be easily calculated from home coordinates and current aircraft coordinates and translated into a target altitude. This mode would then make sure that the plane travels on a defined slope relative to home and it would be independent of any wind influence.  Moreover, the mode would be very helpful for longrange flights in order to automatically keep a healthy ratio of distance to altitude. We could have a parameter like "fw_nav_dist_alt_ratio" for this new mode. If set to e.g. 10 the target altitude would always be 1/10 of the current home distance. For example 1000 m high at 10 km distance.

Actually, now that I'm writing it, it's not needed to calculate the angle. It would be probably enough to maintain a constant distance/altitude ratio.

OlivierC-FR, 2017-10-11:
About the wind,  if the plane is constantly aiming for the point above the pilot, I guess the navigation PIDs would deal with it, no ? The "I" correction should fight the wind I think.

Dronek, 2017-10-11:
> About the wind, if the plane is constantly aiming for the point above the pilot, ....

Sure, that's the intended behavior. But it is not achieved by a constant flight attitude (dive angle). In order to reach the mentioned point the flight attitude of the plane can be very different with different wind situations. 

The method, I tried to discribe would be like RTH with CURRENT height while recalculating the value of CURRENT height as a function of home distance.

junwoo091400, 2017-10-11:
And of course, Target home altitude should be higher with user specified rth altitude :)

Redshifft, 2017-10-11:
There can never be a perfect single failsafe/RTH procedure for every situation :(

Best I can think off is  Failsafe > climb 10mtrs > if above set RTH height, desend at some ratio relative per km from home as Olivier said.

My thoughts for always having that initial climb is hopefully it will unblank say an RC link signal loss :)

Dronek, 2017-10-11:
> There can never be a perfect single failsafe/RTH procedure for every situation :(

... so let's quickly and permanently dismiss the term "perfect" in conjunction with RTH/FS ;)

The way I see it, this here could be rather a mode which can be also an option for RTH (same as ALTHOLD being an option for RTH right now, aka CURRENT). It would be really useful to have a mode for general flying where the ratio of home distance and altitude is kept constant. Not only for RTH.

OlivierC-FR, 2017-10-11:
I sooo agree with the initial climb, always necessary, even if the aircraft is well above the set RTH altitude, some extra 20-40 meters is often enough to get the video feed or RC link back, most of the failsafes at medium-long range are because you went slightly behind the top of a hill.

Now, to be honest my actual setting of RTH = FIXED at altitude 50m, plus the max dive angle set to 5° is working quite well, it makes a nice glide in like 99% cases, it's just that a straight RTH would be neater :)

Redshifft, 2017-10-11:
Yes, max dive angle set to 5° + fixed 50mRTH altitude will get the job done at your 5k distance very well (about 1m down every 11.43m covered) I suppose straight to a home point 50m high should not get many in trouble as we are all going to be basically LOS

Dronek, 2017-10-12:
@Redshifft @OlivierC-FR 
Just take into account that the "dive angle" in iNav is actually the pitch angle of the aircraft.
This will guarantee a constant negative vertical speed at most. The fixed angle can however result in very different slopes depending on wind and even depending on aircraft.
For example: with headwind the actual slope will be steeper than without wind and much steeper than with tailwind.

That's why I think, a constant distance/altitude ratio would be nicer and pitch angle would dynamically adapt due to NAV PID as in ALTHOLD etc.

[image]

OlivierC-FR, 2017-10-12:
I agree, limiting the dive and climb angles to low values is not ideal.

EnglishTurbines, 2017-10-19:
I always set my RTH Altitude to 100 mtrs which means the plane arrives Home at a suitable height you can see LOS. My dive angle is limited to just 5 degrees, this is ideal as it turns altitude into speed while returning Home in a long and gradual descent (usually downwind if you are watching what you are doing).  People who fly at heights above 400ft are not being very responsible IMO. If you are using UHF and 1.2G video, you dont need excessive altitude. If you are using 5.8G Video, then using excessive height to compensate for a poor video signal is not really very responsible IMO.  All these suggested RTH scenarios are endless as different people want different ways of doing it for different reasons. A pick list of how you want it done is probably the only way. But I think it should be very limited, not expanded add infinitum....That's how Mission Planner got to be WAAAAAAY to complicated.....LESS IS MORE...!!!! Right now, people like iNAV **_because its not complicated_**, lets keep it that way please.

shellixyz, 2018-05-28:
I think this issue should be reopened

fiam, 2018-10-09:
@shellixyz should we reopen?

shellixyz, 2018-11-22:
@fiam Yes, didn't see your message, I've been drowning in mails ;)

shellixyz, 2019-06-07:
Implemented in 2.2, closing

closed 2019-06-07 by shellixyz