iNavFlight/inav #2319: Omnibus F7 (Banggood) problem with I2C

swemonstro, 2017-10-11:
I have as mentioned in Another thread some problem with my Omnibus clone from Banggod (wasnt aware it was a clone when procured unfortunate) 

I have connected a gps to the card with success, but when I connect my compass (an external xxx) to the SCL/TX3 and SDA/RX3 (and 5V and GND) the card don't connect to my PC (normally USB4 appears when I plug in the USB)

do I need to start I2C somehow? or do I need to decide between UART3 (RX and TX) or SDA/SCL somehow?

any suggestions

Sincerely yours

Carl

digitalentity, 2017-10-11:
This may happen if you connected something wrong and I2C bus got stuck.

swemonstro, 2017-10-11:
And there is nothing I should do on the board to enable i2c since its on the same pins as UART3?  or change anything at 'ports'?

swemonstro, 2017-10-12:
Banggood asked be to document the behavior on video, so I did. Perhaps anybody can see if I do anything wrong.

https://youtu.be/UiZz_-FuWUs

giacomo892, 2017-10-12:
Have you checked that UART3 is all disabled from Ports tab?

swemonstro, 2017-10-12:
This is how the ports are configured:  
[image]

giacomo892, 2017-10-12:
Do you mind adding 4.7k resistors from SDA,SCL to 3.3v? and see if baro is detected?
Btw the fact that the board is not powering up via USB is not a good sign.

DzikuVx, 2017-10-13:
Yesterday I flashed my Omnibus F7 (original from Airbot) with INAV and everything worked fine: HMC588L mag and onboard baro were detected and working. 
So, either FC is damaged or external MAG is damaged and not working correctly. 
@swemonstro check different I2C hardware.a
Closing.

closed 2017-10-13 by DzikuVx

swemonstro, 2017-10-13:
I tried an added the resistors as suggested, tried two versions of magentometers and totaly three different sensors but the same result every time. 

I'm now convinced its the F7 board that's faulty