iNavFlight/inav #2320: in horizon mode servo wrong direction

sniperxnl, 2017-10-11:
guys,

new build, passtrugh wing flys perfect, ground test horizon and the servos are the wrong direction. I have my taranis stock so nothing is mixed

digitalentity, 2017-10-11:
Most likely wrong board alignment or wrong servo setup.

sniperxnl, 2017-10-11:
board aligment is good. servo are both on -100 to fly in passtrough

digitalentity, 2017-10-11:
The fact that the board is moving servos differently in HORIZON and PASSTHRU modes is in 100% cases configuration issue. Lilely causes:
1. Board alignment is incorrect
2. Stick input is reversed
3. Servo direction is incorrect.

Please upload your Configuration dump. Also, please connect your plane to Configurator and record some videos for us:

1. Configurator - Setup page (where we can see a 3D model). Record a video of you physically tilting the plane (nose up/down, bank left/right). Both Plane and Configurator screen should be visible in the video.
2. Configurator - Receiver tab. Please record a video of Receiver tab and your transmitter (both should be visible) - move pitch up/down, roll left/right, yaw left/right.

Pairan, 2017-10-11:
had the same ... but then found, I've swapped Servo 3 with Servo 4 and was able to fly it in passthr! Going to horizon was almost fatal :)

So ... nothing wrong with iNav ... the setup is mixed!

closed 2017-10-13 by DzikuVx

KnuckleUpFPV, 2018-01-20:
I am having the same issue. My wing follows the model. If I put it in horizon mode tipping it on its nose, it thinks its banking. If I roll the wing left or right the ailerons either both go up or both go down. Ive tried swapping the servos, reversing directions back and forth. Its driving me bananas. I cannot understand whats going on. I am on inav 1.8 with an omnibus f4 v3

shellixyz, 2018-01-20:
Check how the model moves in the setup tab. Moving nose up should make the front of the 3D model go up... repeat for other axes. If that's good make sure the left servo is connected to the third output and the right one to the fourth output.

Last step is making the servos go in the right direction. Put the FC in angle mode. If you tilt the model to the right the left elevon should go up and the right one should go down. The opposite if you tilt it to the right. Reverse the servo(s) direction accordingly in the servos tab in the configurator if the elevon movements are not correct.

If you followed this everything should be good. If not check everything from the start you did something wrong.

KnuckleUpFPV, 2018-01-20:
Yeah as I said before it matches the model in inav. It moves exactly the same. Servos respond correctly to stick inputs. Use the high five method to confirm. Not my first wing. I've tried the servos in both spots. It's the same regardless. Ive reversed them in configurator. I've reversed them at the transmitter. I've done all possible combinations. Roll thinks it's pitch, and pitch thinks it's roll. Unless i give it stick inputs and then it behaves as it should. If i let auto level do its thing, it's wrong. Can't change fc orientation. Because then it doesn't match the model.

If I use the v3 firmware it will freeze up and completely lose its programming. I use the pro firmware and soft serial actually works. It doesn't freeze up at all. Both firmwares have the servo direction error though. I followed the diagram for accelerometer calibration exactly. More than 4 times. Never changed. 

Only thing that seems to partially work is changing channel mapping to EATR. But then the left servo nose facing away or number 3 doesnt move all the way. Doesn't matter if I change the travel or not. Still moves the same distance. Not even close to what servo 4 is capable of moving.

shellixyz, 2018-01-20:
You shouldn't have to guess the channel mapping roll stick should move the roll channel in the receiver tab, pitch move the pitch channel and so on. Moving the roll stick right should increase the roll channel. Moving the pitch stick up should increase the pitch channel.

If that's right and the model is moving like it should then it is only a matter of connecting the right servo to the right output and choosing the right direction for the servos in the servos tab.

Shouldn't be more complicated than that.

shellixyz, 2018-01-20:
For the servo travel if it is alright in passthrough mode then you need to follow this for the stabilized modes to work like you intend them to: https://github.com/iNavFlight/inav/wiki/Tune-INAV-PIFF-controller-for-fixedwing

KnuckleUpFPV, 2018-01-20:
It's not a guess on channel mapping. Aetr the pitch moves pitch, roll moves roll. 

Eatr pitch moves pitch, roll moves roll.

 I've done both and switched the servos back and forth. It only changes the direction they move until i change it in servos tab. 

It shouldn't be more complicated, but it is. I dont have this issue on any other fc. My matek and sp racing fc's are fine. Again not the first wing Ive built. 

I don't know whats causing it. I've tried every possible combination of settings. Only one that's close to right is with it set to eatr. 

But then one servo barely moves while the other still has full travel. Adjusting the gains applies to both sides evenly or should. It's not. Servo 3 is restricted in movement with this channel mapping to get auto level modes to match fc movement. If i go to aetr the movement looks like the fc is rotated. But I have full throw on both servos. I can even adjust the travel. And the travel adjustment works. But back to the issue of the servos thinking the fc is rotated. It's either something in the firmware or this is a different variation of the v3. I couldn't find a definite picture of the v4. I did find the v3 and v5. My board layout matches the v3 perfectly. Uart6 doesnt work at all. I think it's got to be something different with the board itself and the firmwares available do not match. I've tried all of them. Regular omnibusf4 won't open in configurator. All the others work. I can't soft serial with the v3 firmware. I can soft serial with the pro firmware. I like the extra osd options from the pro firmware better anyhow. 

This is such a pain. I should have just used another sp racing f3. All of my other wings work fine. Even the 3d printed wing with the sp racing f3.

shellixyz, 2018-01-20:
You said the same sticks move the same channels in AERT and EATR. That can't be right.

I don't know what is going on but you should test things incrementally:
1. Test the channel mapping
2. Test passthrough. If it doesn't work check the outputs and also the servos independently.
3. Test `ACCRO` because it only tests the gyro. If you get good travel in passthrough and you configured the FF terms there is no reason it shouldn't work in this mode.
4. Test `ANGLE` because it test the accelerometer

Your issues are very odd. Maybe your FC's MCU is dead but I doubt it. It would probably just not work but because you are saying you know how to setup iNav and it is the only one posing problem I can't see anything else.

KnuckleUpFPV, 2018-01-20:
Yeah aetr and eatr have the same movements on the receiver screen. Doesn't make any sense to me. There is no mixing in the transmitter. I appreciate all the suggestions. I've done them all. Maybe something happened when the fc completely lost its programming. First time I've ever seen  flight controller lose all of its settings on a save and reboot. Wasn't just that page. It was the setup on every page that was wiped. I'll toss the sp racing f3 in or get an f3 omnibus. I dont really want to trust it at this point.

ISubrosaI, 2018-01-26:
I am having the same issue, with the same board. Omnibus f4 pro v3. However this is my first wing, and I didn't catch the problem until I almost gave up to go fly my quad. Gonna try an older firmware version and see if that changes anything...

CPTjoe76, 2018-02-01:
...and it is the same with my F4 V3 board (non-pro version).
Also my 4th flying wing with a F4 board. Must be a bug in the hardware?!
I think that if we could change direction of the pitch or roll channel in the FC independently from servo reversing, then it would work (and would have to reverse the TX non-standard as well). Servo reversing won't help because one channel is always right and the other wrong (pitch or roll). Reversing servos will shift the problem to the other axis because each servo is used for pitch AND roll.
Do you know what I mean? 
I would throw the V3 to rubbish, but I want to connect RSSI and an Airspeed Sensor to the board.
Does anyone know, if - and to which port I can connect the analog RSSI AND Airspeed sensor at a F4 V2 pro?

digitalentity, 2018-02-04:
Wrong servo direction in some of the modes is in 100% cases a setup issue. Either mixed up left and right servos (or mixer problem), incorrect channel directions on TX, incorrect servo reversavls on FC or incorrect board alignment. 

Check the following:
1) Model in Configurator should mode same way as actual aircraft when you tilt it by hand; If not - check board alignment
2) On RX tab, pitch stick down should be 1000, roll left - 1000; If not - check channel inversion on your radio
3) Check that servos move correct in `PASSTHROUGH` mode; If not - try swapping servos and changing servo inversions.

KnuckleUpFPV, 2018-02-04:
You say its a setup issue when we have built other wings with inav without issue. I've also had inav completely wipe my flight controller on a save and reboot. Something is buggy in the firmware. My other wings all respond correctly. I know its hard to admit there may be something wrong on the programming side. But that's the case.

radnav, 2018-02-04:
I got it - and apologize. I think the failure in my case was the following:
My board is mounted 90 degree offset. Since I made the alignment after mounting the board, I aligned it according to flight direction and not according to the arrow on the board.
However this failure did not show up on the configurator, since the model was moving 100% correct.
After I realized the failure 2 days ago, I made a correct alignment and the failure was still present.
I was sure, I swapped the servo connectors again to evaluate! Swapping it today suddenly made it work. I could not believe it!!! So sorry for any inconvenience, but I was trying for days now.
@KnuckleUpFPV : do you have an offset mounting of your board?
By the way, my other boards are mounted offset as well, but I aligned them before installing.

KnuckleUpFPV, 2018-02-04:
No offset on mine. After about the 8th or so firmware flash it started working correctly and hasn't dumped the configuration since then. I had to use the pro firmware. With soft serial. It worked after that. Thank God.

digitalentity, 2018-02-05:
@KnuckleUpFPV as you said, if you have build other wings with the same firmware and they work fine - there's nothing wrong with the firmware. Please provide full details about your setup - board placement (preferrably photos), Connections, full config `dump`.

KnuckleUpFPV, 2018-02-05:
See my last post AGAIN.  After 8th flash the setup I was using worked fine. It was completely wiping out all programming of the flight controller after save and reboot at times. My buddy Richard had the same issue on his new matek f405 crt. He got his all setup and hit save and reboot and it was like he never set anything up at all. He was also having the same issue with servos not matching. He's good friends with Sampson Guo from matek. If you dont want to believe me you can get it from him or Sampson. The exact same setup and the servos weren't matching before. And now they do.

digitalentity, 2018-02-05:
If it started working after 8-th clean reflashing I see two explanations:
1. Hardware is faulty. Even if it works now I wouldn't risk flying this board in this case.
2. Something was different about 8-th reflash. In this case we should figure out what it was.

Before you can test you need to do minimal serup - mixer, receiver etc. It's possible that you did something differently. To further investigate I need to reproduce the issue and for that I need your description and settings for a non-working setup.

KnuckleUpFPV, 2018-02-05:
It's happened on a ombibus f4 v3 and a matek f405 crt and my sp racing f3. The sp racing f3 I did nothing but change it to flying wing and set motor protocol just to test. It wiped the changes and changed it back to quadx. Just like the others. Two of the boards are mine. The matek belongs to Richard Hamilton. I'm going to go ask on the inav facebook if anyone else has had this happen. The flight controller that wasn't working has been completely fine because I haven't plugged it in. After I got it to respond correctly with the exact same setup I used every time I set it up. I used the exact same setup on my other wings. Settings are all identical. 

Now that I think about it I never had this issue on the chrome configurator. I started using the stand alone. Richard was also using the stand alone. Last time I connected and pulled the dump I used the chrome gui. 

I have the dump file if you want to see it. I'm not going to risk plugging it in again though. All of my functions are working and its flying perfect. I can plug the sp racing f3 in if you like. It's just an empty board. Not connected to anything. At the time I had it hooked to a gps and a barometer to test it for a tiny wing im building. Once I saw it wiped out I boxed it and grabbed another. 

I'm happy to help but I do not want to plug any of my wings in.

albrechtf, 2018-03-10:
I had the same with my latest fixed wing (second plane powered by iNav for me). Finally I just set the rate for both Aileron Servos on the Servos tab to -100%, and inverted the channel in my TX (to match this, as it was correct with TX before, but wrong auto-correction in Angle mode). Now it works fine. Perhaps this may help others stumbling upon this...

subiblue2002, 2018-03-24:
I'm also experiencing this problem. 
Going to attempt to reflash firmware. 
I'm currently using the latest version. Can't remember off the top if my head what version it is.
Any movement on this ?

digitalentity, 2018-03-25:
Incorrect servo directions is in 99% cases a setup issue. 

1. Verify that your RC channels are correct (move both sticks to top-right and you should see all channels go to max).
2. Verify that flight controller is aligned correctly (model in the Configurator should reflect motion of your plane)
3. Verify that servos are connected to correct outputs (left to left, right to right)
4. Reverse servos that move in the opposite direction
5. Verify that servos now move correctly both in Passthrough and Angle modes. If they don't - you missed something of the above.

colelandreneau, 2018-04-02:
I'm having the same problems on my Matek F405 std (standard). Servos are good in manual/passthrough but in angle and horizon, they are wrong. I had my board aligned correctly but I decided to turn it 90deg and not touch anything in configuration, this fixed the roll axis issue but not pitch. The servos go down when they should go up. I'll continue to mess with settings but so many people with the same issue make me wonder whats going on.

digitalentity, 2018-04-02:
Difference between servo motion in `PASSTHROUGH` and stabilized modes is always a configuration issue. It has been explained multiple times - see my previous comment above for latest explanation.

colelandreneau, 2018-04-02:
Thanks, hopefully I can get it sorted out. I see what you mean.

KnuckleUpFPV, 2018-04-02:
It's because the servos are wrong in the inav diagram. Look at the diagram as if its the bottom of the wing and not the top. It's not all fc's that do this though.  There has been at least 5 people this week in the facebook group with the same issue. Plugging the right servo into the first servo position with nose facing away from you and left servo into second servo spot will work. Which is wrong from the diagram. If you do it the other way you have to reverse servo directions in the configurator and your transmitter. And also adjust your flight controller direction. Another fun fact for you. The matek f405 aio on 1.9 has servo lockouts from full throw in the stick. Go bacl down to 1.8 and servos are fine. Can move them rapidly to full stick in 1.8 back and forth and all is well. Try moving it in 1.9 to full stick they lock out. Only for a second. But its long enough to put you in the ground when flying low and fast. 150.00 lesson learned. Switched between firmwares twice to test this.

After I replaced my new wing I moved my inav setup into a fresh build. Fresh flash on the f405 aio to 1.8. This time nose away from me I hooked right servo to s2 pad and left servo to s3 pad. Put fc same orientation as my other builds arrow forward. Added no adjustments to servo direction. Works perfect in all modes. Responds appropriately.

colelandreneau, 2018-04-02:
Maybe I can clarify what's going on. In acro, manual, pass through, angle horizon and so on, the servos correctly respond to MY INPUT, however when I move the plane around on it's axis in angle/horizon the servos don't respond correctly to level the plane out if it we're in the air. Am I experiencing the same thing as you guys? All inputs are fine but the gyro won't level the wing correctly on it's own. The board is correctly oriented and is moving correctly in the setup diagram. Hopefully that clears it up.

KnuckleUpFPV, 2018-04-02:
Try switching your servos to the opposite sides. Then make sure your channel mapping is correct

dassdrones, 2018-07-05:
I have an omnibus F4 clone with the same issue. Had the firmware flashed using OMNIBUSF4PRO LEDSTRIPM5, reflashed it to OMNIBUSF4PRO and the issue was fixed

jdjeff58, 2018-07-22:
I'm using a Matek F405 and my issue is that no matter how many different ways I configure my transmitter or the FC in Inav, pitch and roll are swapped.  All I'm trying to do is set up the FC for flying wing to see the proper inputs vs. the proper outputs.  In receiver tab, when I set up the channel mapping, I can get the proper movement on the color bars.  But then the servos behave backwards in pitch and roll.  If I swap inputs in the transmitter, the servos behave correctly but then the receiver tab pitch and roll gets swapped.  I've been chasing my tail all day long with this.  I swapped servo cables and that doesn't help.  I'm not even testing any modes yet...this is all manually deflecting the control stick.  I'm on outputs S3 and S4 of the FC and losing my mind.

Jancis, 2019-06-26:
I had the same issue - manual works as expected, but the horizon was doing something weird. Turns out i had pitch and roll mixed up in receivers tab and one of servos was reversed in servos tab. Again - config issue in my case. 
I have Matek F405-AIO / inav 2.2.0

digitalentity, 2019-06-27:
@Jancis see https://youtu.be/fVJYodLimD8 and make sure you have everything set up correctly.

a555bob666, 2019-07-16:
I had the same issue. worked perfectly in passthrough, but the servo directions were reversed in angle and horizon. my quick fix was to reverse the direction in my transmitter, and then reverse the direction in the gui, that way it was correct in stablized mode and in passthrough. idk if that helps yall or not.

jcameron, 2020-08-07:
I found this thread when I ran into the same issue on my wing with iNav. Turns out that the pitch and roll channels were reversed on the TX, but the mix in iNav was configured in a way that undid the reverse and allowed the wing to fly normally in manual mode! 

Switching to a stabilized mode caused an immediate crash, and after bench testing to check that the servos responded in the right way when the aircraft was moved I tracked it down to the incorrect TX channels.