iNavFlight/inav #2321: Omnibus F4 v2 usb came off

sniperxnl, 2017-10-11:
Guys,

In the field today after some testing inflight the usb came off, and also the pads.
Someone knows away to get it back online?

DzikuVx, 2017-10-11:
@sniperxnl I do not think this is the correct place to ask that specific hardware questions. But if pads are gone, it might be undoable and getting a new board might be the cheapest solution

closed 2017-10-11 by DzikuVx

sniperxnl, 2017-10-11:
That is not true, when it happen in betaflight then there is a special betaflight firmware that activate on a other uart the msp, you need to load that with a st-link v2 tool. So there is a inav firmware needed that use a diffrent msp port

digitalentity, 2017-10-11:
MSP is enabled by default on all boards with VCP