iNavFlight/inav #2322: Setting max altitude in ALTHOLD mode?

acazorso7, 2017-10-11:
Hello, 
It's possible to set a maximum altitude when the quadcopter is flying in ALTHOLD mode? I'm searching because I think that it's very useful when we fly indoors. Right now I can't find something similar in iNAV.

For example, when quadcopter arrives to 3 meters it maintains the 3 meters and only obey the remot control when it indicates to lose altitude.

Could someone let me know if it exists?
Many thanks!

fiam, 2017-10-11:
@acazorso7 Not possible at the moment. See #1510

acazorso7, 2017-10-12:
@fiam Many thanks for your quick answer. Do you know if this development is in progress or planned for the near future?

fiam, 2017-10-12:
@acazorso7 Not that I'm aware of.

We do have an altitude alarm in the OSD and most people seem to be happy with that. Artificially blocking climbing could prevent you from avoiding and obstacle and cause a crash, that's why we're a bit hesitant to add such a feature.

Discussion and PRs are always welcome, so feel free to tell us about your use case.

closed 2018-05-27 by stale[bot]