iNavFlight/inav #2323: Magnetometer issues with Omnibus F3

Bushedo, 2017-10-11:
I have issues with external mag under iNAV Config 1.7.3. I have an Omnibus F3 AIO (no magnetometer) FC from RTF Quads. I cannot get it to work with the external compass on my GPS in iNAV. The compass module uses a 5883 mag that I have verified works using an Arduino mega board and Adafruit libraries and code. GPS works fine in iNAV. When I select this chip under the mag drop down menu and save the config, iNAV quits communicating and tells me I have to update the software. To regain comm with the board through iNAV I have to power down and disconnect the SDA and SCL lines from the mag. Have searched the iNAV forums and so far have not been able to find any similar problems. There were suggestions about adding 3.3 V pull ups to the SDA and SCL lines but this caused more problems until I removed the 1K ohm added resistors. Any help will be greatly appreciated. Had same problem with another GPS module (one BN-800 and one NEO M8N).
Any suggestions will be greatly appreciated.
  |

digitalentity, 2017-10-11:
Is you GPS module powered? When connected only by USB Omnibus doesn't provide power to external devices.

Bushedo, 2017-10-12:
I also had the problem with battery power connected.  Not sure if that was the case when I tried the pull-up resistors.  Do you think I should retry them using battery power?

swemonstro, 2017-10-12:
Sounds like the same problem I have with my Omnibus F7 board. #2319

Bushedo, 2017-10-12:
On my F3 board the baro works so the I2C bus is working.  Whenever I set the mag parameters in config and have SDA and SCL connected, the config program tells me I have to update the flight controller software.

swemonstro, 2017-10-12:
Ah, not quite the same problem then...

giacomo892, 2017-10-12:
Bare in mind that on OMNIBUS F3 BARO is connected to SPI bus. Not I2C.

Bushedo, 2017-10-12:
Oh, I didn't know that.  Since the BME280 can be either I2C or SPI is it for sure the case that all baros are using SPI on the F3 boards?

jflyper, 2017-10-13:
For Omnibus (F3), onboard BMP280 (not BME280) is on SPI.

Bushedo, 2017-10-13:
Well, it seems the issue is likely a faulty I2C bus on this FC.  Anyone know how common this is?

jflyper, 2017-10-13:
Never heard of the case.

You have to find out what's going on by scoping. Logic analyzer, then DSO.

KiwiDavid, 2017-10-15:
As DE indicated the GPS needs to be powered off a battery (not USB) but the GPS module usually has a 5V to 3.3V regulator which provides power to both the GPS and the on-board mag sensor.  If the GPS is unpowered then you also have an unpowered mag chip sitting on the I2C lines and substrate diodes may start to conduct causing problems with the bus.
I can't access my SP F3 board with external mag unless I power with both USB and main battery or I disconnect the external mag SDA and Clk lines.

Bushedo, 2017-10-15:
Well,  I had battery power applied during some of the tests where the mag failed.  I also checked voltage at the 5v pins of J11-13 and measured 4.6-4.7 volts with just USB power applied.  With battery connected these pins showed 5v.  I will confirm the results of mag tests with battery power tonight.

KiwiDavid, 2017-10-15:
Failing on battery power does sound like an issue.  I and a few others a while back had temperature related problems with mag detection.  Mine worked fine in the house but always failed at the field - basically any temperature below about 20C it failed to detect the mag chip and the I2C logs show it was never addressed by the micro - i.e. not a sensor fault.
Your problem may be totally different but be aware.

Bushedo, 2017-10-15:
How can I see the I2C logs?

Sent from my Verizon, Samsung Galaxy smartphone

KiwiDavid, 2017-10-15:
Sure - they're a little lengthy but I've cut them to the minimum.

Bad start up.
Time [s],Packet ID,Address,Data,Read/Write,ACK/NAK
2.6875e-005,0,h (0x68),u (0x75),Write,ACK
7.8375e-005,1,h (0x68),h (0x68),Read,NAK
0.000134625,2,h (0x68),'0' (0x00),Write,ACK
0.000186125,3,h (0x68),'255' (0xFF),Read,NAK
0.0002425,4,h (0x68),'6' (0x06),Write,ACK
0.000294125,5,h (0x68),'19' (0x13),Read,ACK
0.000317,5,h (0x68),l (0x6C),Read,ACK
0.000339875,5,h (0x68),'243' (0xF3),Read,ACK
0.000362875,5,h (0x68),'25' (0x19),Read,ACK
0.00038575,5,h (0x68),'5' (0x05),Read,ACK
0.00040875,5,h (0x68),'30' (0x1E),Read,NAK
0.0004835,6,h (0x68),k (0x6B),Write,ACK
0.000507875,6,h (0x68),'128' (0x80),Write,ACK
0.10115375,7,h (0x68),k (0x6B),Write,ACK
0.101178125,7,h (0x68),'3' (0x03),Write,ACK
0.101234125,8,h (0x68),'25' (0x19),Write,ACK
0.1012585,8,h (0x68),'0' (0x00),Write,ACK
0.116402875,9,h (0x68),'26' (0x1A),Write,ACK
0.11642725,9,h (0x68),'1' (0x01),Write,ACK
0.116484,10,h (0x68),'27' (0x1B),Write,ACK
0.116508375,10,h (0x68),'24' (0x18),Write,ACK
0.1165645,11,h (0x68),'28' (0x1C),Write,ACK
0.116588875,11,h (0x68),'16' (0x10),Write,ACK
0.116644875,12,h (0x68),7 (0x37),Write,ACK
0.11666925,12,h (0x68),'2' (0x02),Write,ACK
0.1167255,13,h (0x68),8 (0x38),Write,ACK
0.116749875,13,h (0x68),'1' (0x01),Write,ACK
0.12715325,14,w (0x77),'160' (0xA0),Write,ACK
0.127204875,15,w (0x77),'0' (0x00),Read,NAK
0.12726075,16,w (0x77),'30' (0x1E),Write,ACK
0.127285125,16,w (0x77),'1' (0x01),Write,ACK
0.13014925,17,w (0x77),'160' (0xA0),Write,ACK
0.130200875,18,w (0x77),'0' (0x00),Read,ACK
0.13022375,18,w (0x77),'152' (0x98),Read,NAK
0.13028025,19,w (0x77),'162' (0xA2),Write,ACK
0.130331875,20,w (0x77),'193' (0xC1),Read,ACK
0.13035475,20,w (0x77),'18' (0x12),Read,NAK
0.13041125,21,w (0x77),'164' (0xA4),Write,ACK
0.130462875,22,w (0x77),'198' (0xC6),Read,ACK
0.13048575,22,w (0x77),'248' (0xF8),Read,NAK
0.130542875,23,w (0x77),'166' (0xA6),Write,ACK
0.1305945,24,w (0x77),w (0x77),Read,ACK
0.130617375,24,w (0x77),\r (0x0D),Read,NAK
0.130673875,25,w (0x77),'168' (0xA8),Write,ACK
0.1307255,26,w (0x77),j (0x6A),Read,ACK
0.130748375,26,w (0x77),' (0x27),Read,NAK
0.130804875,27,w (0x77),'170' (0xAA),Write,ACK
0.1308565,28,w (0x77),{ (0x7B),Read,ACK
0.130879375,28,w (0x77),+ (0x2B),Read,NAK
0.130935875,29,w (0x77),'172' (0xAC),Write,ACK
0.1309875,30,w (0x77),l (0x6C),Read,ACK
0.131010375,30,w (0x77),'132' (0x84),Read,NAK
0.13106825,31,w (0x77),'174' (0xAE),Write,ACK
0.131119875,32,w (0x77),[ (0x5B),Read,ACK
0.13114275,32,w (0x77),'11' (0x0B),Read,NAK
0.745652125,39,h (0x68),; (0x3B),Write,ACK
0.74570375,40,h (0x68),'1' (0x01),Read,ACK
0.745726625,40,h (0x68),\t (0x09),Read,ACK
0.7457495,40,h (0x68),'255' (0xFF),Read,ACK
0.745772375,40,h (0x68),'164' (0xA4),Read,ACK
0.74579525,40,h (0x68),'16' (0x10),Read,ACK
0.745818125,40,h (0x68),o (0x6F),Read,NAK
0.786451875,41,h (0x68),C (0x43),Write,ACK
0.7865035,42,h (0x68),'255' (0xFF),Read,ACK
0.786526375,42,h (0x68),'205' (0xCD),Read,ACK
0.78654925,42,h (0x68),'0' (0x00),Read,ACK
[... 135 lines cut ...]
0.706575125,56,'30' (0x1E),'6' (0x06),Read,NAK
0.706633,57,'30' (0x1E),'2' (0x02),Write,ACK
0.706657375,57,'30' (0x1E),'1' (0x01),Write,ACK
0.777088375,58,'30' (0x1E),'3' (0x03),Write,ACK
0.77714,59,'30' (0x1E),'2' (0x02),Read,ACK
0.777162875,59,'30' (0x1E),2 (0x32),Read,ACK
0.77718575,59,'30' (0x1E),'2' (0x02),Read,ACK
0.777208625,59,'30' (0x1E),'31' (0x1F),Read,ACK
0.7772315,59,'30' (0x1E),'2' (0x02),Read,ACK
0.777254375,59,'30' (0x1E),'8' (0x08),Read,NAK
0.77731375,60,'30' (0x1E),'2' (0x02),Write,ACK

The thing to note is the time jump in the first (bad) log-
0.131119875,32,w (0x77),[ (0x5B),Read,ACK
0.13114275,32,w (0x77),'11' (0x0B),Read,NAK
0.745652125,39,h (0x68),; (0x3B),Write,ACK
0.74570375,40,h (0x68),'1' (0x01),Read,ACK

Now the micro is the master and initiates all the traffic and the sensors only respond when addressed so what is going on here?  The bad start never addresses the mag sensor - ever. It should be addressed right after the baro but instead there is a long pause then it's back to acc/gyro again.  
It should be noted also that the baro code does have a branch for temperatures below 20C and another lower than that again but it can't be a code bug otherwise everyone in Northern Europe and the Northern US would be complaining over winter.   It may be a mitigating factor but there is something else involved - bus speed or clock stretching maybe.
Don't get too involved on this one because there is no evidence your problem is related to mine at all.
I post only for your academic interest.....

Cheers,
David

KiwiDavid, 2017-10-16:
Sorry - if you want to record I2C traffic yourself it requires a logic analyzer - probably not worth the effort unless you have other needs for it.

Bushedo, 2017-10-16:
Oh, thought maybe the black box did it somehow.

Sent from my Verizon, Samsung Galaxy smartphone

wiwel, 2017-10-31:
I have had a similar issue with the SPRacingF3 Deluxe.  No magnetometer on power on.  I have found that the magnetometer  is sensitive to the Vdd rail ramp rate.  My DC/DC has a very clean ramp from 0 to 5v in 1mS.  It appears this is too slow.  To fix this I put a power cut momentary push button in line to cycle the main 5V once the DC/DC comes on.  With the button, the rail comes up in about 1uS.  This allows the magnetometer to work every time.  The slow ramp seems to latch up the magnetometer.  Once this happens even a system reboot does not fix it.   Google has not turned up any known power rail ramp issues with this magnetometer.   I have not attempted to use an external mag because  I do not want to permanently de-solder the on board mag.  There could be some code entanglement here but since it is working for me now, I do not plan to scope anything.

KiwiDavid, 2017-11-01:
A most interesting find - many chips are Vdd rate sensitive.  I would go back and confirm if my problem was caused by the same effect but I had to change to external mag anyway and have since replaced the "dodgy" unit.
If someone can confirm this finding it may solve a variety of issues relating to the mag sensor.  Mine (and a few others) was temperature sensitive but this could be in the BEC.  Capacitors also change with temperature.  Nice one - E-beers to you!

TimMW1, 2017-11-01:
This could be SW related in that the MCU could be accessing the mag before it gets done "booting".  This could put it's I2C state machine into a wacked state.  I have seen this many times before with I2C devices with no reset line.  To truly root cause this would require some I2C decode of bus traffic.  Just time.  For now the added 5 gr of weight from the push button switch is nothing.

KiwiDavid, 2017-11-02:
As you can see a few posts back - the mag never gets called in a bad start so it's not like it never responds.

Pokornz, 2017-11-21:
Hi, is it true that the I2C on this board uses 3.3V logic? I need to connect a sonar via Arduino to get I2C into the fc, and Arduino uses 5V logic I think..

DzikuVx, 2017-11-21:
@Pokornz all STM32 are 3.3V logic. So either 3.3V Arduino or Logic Level Converter in between

jflyper, 2017-11-21:
It works with 5V logic.

However, I recommend using bidirectional level shifters for safety of the boards.

DzikuVx, 2017-11-21:
@jflyper It might or might not. I had to replace STM at least twice because of 5V logic attempts

jflyper, 2017-11-21:
@DzikuVx 
That's weird. Connecting 5V logic to I2C pins should do no harm even if it doesn't work, as they are FT.

Pokornz, 2017-11-21:
Thanks guys, I already got the level shifter, I just wanted to confirm. But I guess I have to supply the 3.3V (for the shifter) from the Arduino, since the Omnibus F3 has only 5V.

DzikuVx, 2017-11-21:
There is a theory and there is a practice, like always ;) 
The problem is probably the fact that on Arduino you can not disable builtin pullups when acting as I2C slave in an easy way. I gave up after few tries and just switched to 3.3V everywhere

closed 2018-02-05 by DzikuVx