iNavFlight/inav #2327: MSP_RTC can return 0 time (affects BBlog ctime and date stamp as well).

stronnag, 2017-10-12:
This is really a record of an unexpected occurrence rather than a bug (maybe). Apologies that this is a bit of a saga. 

Today, in 1 of 7 BB logs, RTC was not set; note: During this whole episode there was sub 1.2 HDOP, between 19-27 visible satellites and solid 3D fix.

Since we introduced `MSP2_COMMON_SET_TZ` and `MSP_RTC` mwp has been logging RTC in mwp's stderr log as (for example) ---- here I'd forgotten to enable ntpd on the laptop :).
````
2017-10-06T15:41:57+0100 RTC local 2017-10-06T15:41:57.752, fc 2017-10-06T15:43:58.310
````
mwp's logic sets the TZ and requests RTC during the pre-arm, testing for an internal `rtcsecs` variable being zero, and setting it from the `MSP_RTC` response. In the period I've been doing this (since
2017-09-11), I had never seen MSP_RTC return 0, until today. This resulted in blackbox logs with 'start of epoch' ctime and internal time stamp. Here's the logs from today, the `LOG*.TXT` blackbox logs and
mwp's `mwp_*.log`. Both of these start on arming, and there is a 1-to-1 correlation between the mwp log file name and the LOG* timestamps, apart from LOG00003.TXT, which also contains:
```
H Firmware revision:INAV 1.7.4 (2710c172) SPRACINGF3EVO
H Firmware date:Oct 12 2017 11:55:55
H Log start datetime:0000-01-01T00:00:00.000+00:00
````
Log folder (note `LOG00003.TXT`)
````
$  -l  /t/inav/2017-10-12_quad/
total 36792
-rw-r--r-- 1 jrh jrh 2691764 Oct 12 14:03 LOG00001.TXT
-rw-r--r-- 1 jrh jrh 2321459 Oct 12 14:08 LOG00002.TXT
-rw-r--r-- 1 jrh jrh 6508502 Dec  1  2015 LOG00003.TXT   #### Time stamp ####
-rw-r--r-- 1 jrh jrh 3752341 Oct 12 14:34 LOG00004.TXT
-rw-r--r-- 1 jrh jrh 2793474 Oct 12 14:40 LOG00005.TXT
-rw-r--r-- 1 jrh jrh 7228993 Oct 12 14:49 LOG00006.TXT
-rw-r--r-- 1 jrh jrh 3112796 Oct 12 15:01 LOG00007.TXT
-rw-r--r-- 1 jrh jrh  806287 Oct 12 16:31 mwp_2017-10-12_140338.log
-rw-r--r-- 1 jrh jrh  743094 Oct 12 16:31 mwp_2017-10-12_140827.log
-rw-r--r-- 1 jrh jrh 1971004 Oct 12 16:31 mwp_2017-10-12_141448.log
-rw-r--r-- 1 jrh jrh 1274780 Oct 12 16:31 mwp_2017-10-12_143452.log
-rw-r--r-- 1 jrh jrh  944486 Oct 12 16:32 mwp_2017-10-12_144029.log
-rw-r--r-- 1 jrh jrh 2423017 Oct 12 16:32 mwp_2017-10-12_144939.log
-rw-r--r-- 1 jrh jrh 1073912 Oct 12 16:32 mwp_2017-10-12_150103.log
````
Immediately prior to LOG00003.TXT, in mwp's event (not flight) log, which contains the unarmed events:
````
2017-10-12T14:14:34+0100 Saving mission
2017-10-12T14:14:34+0100 To WP1: 74.0m, nav_wp_safe_distance 100
2017-10-12T14:14:34+0100 Confirmed mission save
2017-10-12T14:14:36+0100 RTC local 2017-10-12T14:14:36.190, fc 1970-01-01T01:00:00.000
2017-10-12T14:14:37+0100 RTC local 2017-10-12T14:14:37.244, fc 1970-01-01T01:00:00.000
2017-10-12T14:14:38+0100 RTC local 2017-10-12T14:14:38.176, fc 1970-01-01T01:00:00.000
2017-10-12T14:14:39+0100 RTC local 2017-10-12T14:14:39.225, fc 1970-01-01T01:00:00.000
2017-10-12T14:14:40+0100 RTC local 2017-10-12T14:14:40.141, fc 1970-01-01T01:00:00.000
2017-10-12T14:14:41+0100 RTC local 2017-10-12T14:14:41.079, fc 1970-01-01T01:00:00.000
2017-10-12T14:14:42+0100 RTC local 2017-10-12T14:14:42.116, fc 1970-01-01T01:00:00.000
2017-10-12T14:14:43+0100 RTC local 2017-10-12T14:14:43.032, fc 1970-01-01T01:00:00.000
2017-10-12T14:14:43+0100 RTC local 2017-10-12T14:14:43.946, fc 1970-01-01T01:00:00.000
2017-10-12T14:14:44+0100 RTC local 2017-10-12T14:14:44.987, fc 1970-01-01T01:00:00.000
2017-10-12T14:14:45+0100 RTC local 2017-10-12T14:14:45.918, fc 1970-01-01T01:00:00.000
2017-10-12T14:14:46+0100 RTC local 2017-10-12T14:14:46.824, fc 1970-01-01T01:00:00.000
2017-10-12T14:14:47+0100 RTC local 2017-10-12T14:14:47.864, fc 1970-01-01T01:00:00.000
2017-10-12T14:14:48+0100 LTM/Mavlink mode
2017-10-12T14:14:48+0100 Armed 0
2017-10-12T14:14:48+0100 Logging to mwp_2017-10-12_141448.log
````
At the end of the mission, still no RTC, disarmed but powered:
````
2017-10-12T14:23:41+0100 Distance = 2169.5, max speed = 10.0 time = 528
2017-10-12T14:23:41+0100 Disarmed msp
2017-10-12T14:23:42+0100 RTC local 2017-10-12T14:23:42.509, fc 1970-01-01T01:00:00.000
2017-10-12T14:23:43+0100 RTC local 2017-10-12T14:23:43.483, fc 1970-01-01T01:00:00.000
2017-10-12T14:23:44+0100 RTC local 2017-10-12T14:23:44.501, fc 1970-01-01T01:00:00.000
...
2017-10-12T14:31:38+0100 RTC local 2017-10-12T14:31:38.987, fc 1970-01-01T01:00:00.000
...
2017-10-12T14:32:06+0100 RTC local 2017-10-12T14:32:06.378, fc 1970-01-01T01:00:00.000
````
Change the LIPO, for another go (valid RTC returned at first request).
````
2017-10-12T14:32:57+0100 Start poller
2017-10-12T14:32:58+0100 RTC local 2017-10-12T14:32:58.194, fc 2017-10-12T14:32:58.673
2017-10-12T14:32:58+0100 Arming flags:  (0000), load 5%
2017-10-12T14:33:21+0100 Setting autoland for RTH
2017-10-12T14:33:27+0100 Saving mission
2017-10-12T14:33:27+0100 To WP1: 55.2m, nav_wp_safe_distance 100
2017-10-12T14:33:28+0100 Confirmed mission save
2017-10-12T14:34:52+0100 LTM/Mavlink mode
2017-10-12T14:34:52+0100 Armed 0
2017-10-12T14:34:52+0100 Logging to mwp_2017-10-12_143452.log
````
Yea, we have RTC again (and for the rest of the day). in fact, all the other 6 BB logs have consistent time stamps.

So why, for twenty minutes or so, did MSP_RTC return 0 in otherwise excellent GPS conditions?

Maybe in future I will set RTC from the laptop time (now iNav has kindly reminded me to enable ntpd).

stronnag, 2017-10-12:
Here's the number of satellites / HDOP immediately into the mission. Good quality coverage.
Beitian BN880 (with Galilleo), UBLOX7 iNav defaults.

[image]

stronnag, 2017-10-13:
As this is not a flight safety issue and is currently of indeterminate cause, I'm closing it. It can always be resurrected if more information emerges.

stronnag, 2017-10-19:
I'm now of the theory that this might be caused by RF interference, on the basis that the only instance I've seen this I was using HC-12 rather than 3DR for telemetry, and perhaps the HC-12 are noisier, and this affects time reception/

stronnag, 2017-10-19:
Reopening, as there seem to be other examples: https://www.rcgroups.com/forums/showpost.php?p=38478984&postcount=16202

fiam, 2017-10-19:
@stronnag That seems like a different issue, since it seems he gets some time displayed but it's not the correct one. Note that `tz_offset` is in minutes, so setting it incorrectly could cause the hours and minutes to be way off.

In any case, we'd need him to provide more details like a config dump and the `status` command output after a GPS fix has been acquired.

stronnag, 2017-10-26:
Latter case looks like #2409

closed 2017-10-26 by stronnag