iNavFlight/inav #2330: inav 1.7.3 buzzer and baro

StariyHopeer, 2017-10-13:
after flashing no beeping boozer on naze32 now. After the L2c port connected 3dof Compass HMC5983 BMP180 barometer. on of them is automatically started only a compass, a barometer is not defined. When trying to put the barometer in the manual, nothing happens and on the Board the led blinks.

stronnag, 2017-10-13:
The BMP180 (BMP085) is not compiled by default on NAZE target. As F1 will not be supported after 1.8, your will have to compile your own firmware. See https://github.com/iNavFlight/inav/wiki/Building-custom-firmware if you need some assistance.

StariyHopeer, 2017-10-13:
Thanks for the help, but for me it's too difficult, have no barometer.

giacomo892, 2017-10-13:
Here you are 1.7.3 with BMP180 enabled.
@StariyHopeer 

[inav_1.7.3_NAZE_BMP180.hex.zip](https://github.com/iNavFlight/inav/files/1382368/inav_1.7.3_NAZE_BMP180.hex.zip)

DzikuVx, 2017-10-13:
And since Naze will not be supported in next version of INAV, I'm closing this one. Thanks everyone

closed 2017-10-13 by DzikuVx

StariyHopeer, 2017-10-13:
thank you very much for the help!