iNavFlight/inav #2332: How we test RTH on the ground

Jozx900, 2017-10-14:
We put F4 v2 in the FT mini arrow. GPS gets 9 and disarmed moto.  When turn to RTH model and  set throttle to the center, The moto was start. We taken the FT with bicycle and rode 10 meter. the moto suddenly stop when RTH is on. We think is too dangerous the moto stop in the sky. Is that possible when the RTH turns on in the sky then all moto will stop?  
We are afraid to fly with Inav now,
How to set RTH  just like MyFlyDream AutoPilot or Arkbird? We can carry plane run fast to test the RTH.
Is that throw it to the sky is the only way to test RTH?

oleost, 2017-10-14:
https://github.com/iNavFlight/inav/wiki/Failsafe

MW2Stambo, 2017-10-23:
I think there is a minimum distance for RTH to work, about 50 meters.
Perhaps you can change this in CLI, or ride bicycle further before switching on RTH. :-)

closed 2018-05-27 by stale[bot]

nkorls, 2018-06-07:
same question