iNavFlight/inav #2339: SP Racing F4 Evo Flashing problem

awcherry2, 2017-10-15:
Unable to flash inav to SP Racing F4.  Board connects in DFU mode with boot button pressed.  Firmware is loaded online successfully.  Upon hitting "flash firmware", board hangs at "Erasing".

Log information displayed at top of screen, upon hitting "flash firmware" is "USB device successfully opened with ID: 1" then "Detected device with total flash size 1024 kiB" then after about 4 seconds, "USB device successfully closed".  Firmware does not flash.  Board no longer works in iNav or Cleanflight at this stage.  I can then connect in DFU to Cleanflight and flash firmware with no issue.

"Erasing" progress bar at bottom of screen never progresses.  Tried multiple times, various ways, same result.  Tried with Chrome config as well as Windows.  Loaded firmware online and locally.  

At any stage, I am able to connect in DFU in Cleanflight and flash firmware with no problems.  Have done this many times now and always works.

Any ideas?  Thanks.

digitalentity, 2017-10-15:
Unckeck "Full erase" switch. Issue is already fixed in new configurator (which is not released yet)

awcherry2, 2017-10-15:
Thank you for the prompt reply.  I had actually tried this already, but with similar results to other times I flashed; board did not connect, device unrecognized by Windows.  

I now realize that I did in fact flash successfully, but I am having the problem mentioned in a prior Issue regarding COM ports / USB 3.0 ports and this board.

We can consider this closed, and I will switch over to that thread to see if a concrete solution has been found for the connection issues the other users were experiencing.

Thanks again.

closed 2017-10-15 by awcherry2