iNavFlight/inav #2342: Omnibus F4 V2 not in pwm mode

michaelvanderhoek, 2017-10-15:
Hi all!

Im setting up an omnibus F4 V2 for my Radjet and want to use my old FrSky V8frII receiver which is only pwm out. 
When i check the box for PWM in (single wire) and click save it reboots and then it is still on PPM in. 
Doesnt matter which firmware I choose. 

Im on 1.7.3 now, downgraded to 1.7.2 and 1.7.1 but no luck so far.

Any ideas?

DzikuVx, 2017-10-15:
This board does not support PWM input (1 channel per wire). There are not enough hardware inputs, sorry

closed 2017-10-15 by DzikuVx

michaelvanderhoek, 2017-10-15:
Thats weird right? It does have pwm 1-4 and gnd + 5V on a connector. Or is that PWM out at J11? 

Ill order some ppm decoders then! Thanks fir the quick reply