iNavFlight/inav #2344: Allow landing after RTH just during failsafe

fiam, 2017-10-15:
I know we're pushing it a bit for the 1.8 release, but I'd like to get one more small thing done which could avoid some crashes. Since planes can't (still) land vertically on INAV and we don't have any way to avoid obstacles that might be present around the home position (e.g. trees), I prefer to disable landing during RTH, take control when the plane is close to me and land it manually.

This behaviour is controlled by the `nav_rth_allow_landing` setting. An unfortunate side-effect is that since RTH as a failsafe mechanism uses this setting too, an RX failure (e.g. broken cable) will result in the plane loitering around home until it runs out of battery and falls down from its RTH altitude, which will usually be a few meters.

In case of a FS-initiated RTH I'd prefer to allow the plane to land rather than exhaust its battery in the air circling around me. While automatic plane landing is at this time not totally safe, it's certainly way better than a dead fall from e.g. 70 meters. For this reason, I'd like to introduce a way to tell INAV to allow landing after RTH only when RTH has been triggered by a failsafe.

We can implement this in two ways:

a) Introduce a 3rd state for `nav_rth_allow_landing` called `FS`. This would allow users to restore config dumps without errors.
b) Introduce a new setting called `nav_rth_allow_landing_fs` with ON and OFF values which is only used iff `nav_rth_allow_landing` is false and the failsafe system is active. I dislike this one because we would be introducing a setting which is only used when another setting has an specific value and might also cause confusion (e.g. what does nav_rth_allow_landing = ON and nav_rth_allow_landing_fs = OFF do during a FS?).

giacomo892, 2017-10-15:
Nice idea. I think solution A is the best IMHO. Maybe the third state can be called FS_ONLY

digitalentity, 2017-10-16:
@fiam I think that solution A is the best at the moment. Let's change names to `NEVER` `ALWAYS` `FS_ONLY`

fiam, 2017-10-16:
@digitalentity While `NEVER`, `ALWAYS` and `FS_ONLY` are certainly better names, that introduces the problem that dumps won't restore cleanly. However, that seems like a acceptable tradeoff to me. I'll submit a PR for it later today.

fiam, 2017-10-16:
Fixed by #2351

PR #2351 (2017-10-16): Add a 3rd state for nav_rth_allow_landing
New state FS_ONLY allows landing only when RTH is enabled at the
same time as the failsafe mode. This allows the aircraft to land
itself when there's no way to regain control.

Fixes #2344 

Configurator PR at https://github.com/iNavFlight/inav-configurator/pull/281
commit 26938321 (2017-10-16): Add a 3rd state for nav_rth_allow_landing

New state FS_ONLY allows landing only when RTH is enabled at the
same time as the failsafe mode. This allows the aircraft to land
itself when there's no way to regain control.

Fixes #2344
closed 2017-10-16 by fiam