iNavFlight/inav #2353: ACRO or STB mode, throttle will mix with servos.

Jozx900, 2017-10-16:
https://www.youtube.com/watch?v=v0Zdn3U-BGg&feature=youtu.be
In ACRO and STB mode, just like video shows, when I change the throttle, the servos will change also. they seems mixed together. I didn't set any mix in the transmitter, so it must be some mix in the INAV, is that normal?

Jozx900, 2017-10-18:
all mode will happen this problem.here is my setting:
# dump# version# INAV/OMNIBUSF4PRO 1.7.2 Jul 20 2017 / 11:59:41 (81b85de)# resources# mixermixer FLYING_WINGmmix reset# servoservo 0 1000 2000 1500 100 -1 servo 1 1000 2000 1500 100 -1 servo 2 1000 2000 1500 100 -1 servo 3 1000 2150 1480 110 -1 servo 4 1000 2150 1555 -110 -1 servo 5 1000 2000 1500 100 -1 servo 6 1000 2000 1500 100 -1 servo 7 1000 2000 1500 100 -1 # servo mixsmix reset# featurefeature -RX_PPMfeature -VBATfeature -RX_SERIALfeature -MOTOR_STOPfeature -SERVO_TILTfeature -SOFTSERIALfeature -GPSfeature -TELEMETRYfeature -CURRENT_METERfeature -3Dfeature -RX_PARALLEL_PWMfeature -RX_MSPfeature -RSSI_ADCfeature -LED_STRIPfeature -DASHBOARDfeature -BLACKBOXfeature -CHANNEL_FORWARDINGfeature -TRANSPONDERfeature -AIRMODEfeature -SUPEREXPOfeature -VTXfeature -RX_SPIfeature -SOFTSPIfeature -PWM_SERVO_DRIVERfeature -PWM_OUTPUT_ENABLEfeature -OSDfeature RX_PPMfeature VBATfeature MOTOR_STOPfeature GPSfeature CURRENT_METERfeature PWM_OUTPUT_ENABLEfeature OSD# beeperbeeper RUNTIME_CALIBRATIONbeeper HW_FAILUREbeeper RX_LOSTbeeper RX_LOST_LANDINGbeeper DISARMINGbeeper ARMINGbeeper ARMING_GPS_FIXbeeper BAT_CRIT_LOWbeeper BAT_LOWbeeper GPS_STATUSbeeper RX_SETbeeper ACTION_SUCCESSbeeper ACTION_FAILbeeper READY_BEEPbeeper MULTI_BEEPSbeeper DISARM_REPEATbeeper ARMEDbeeper SYSTEM_INITbeeper ON_USBbeeper LAUNCH_MODE# mapmap AETR1234# namename -# serialserial 20 1 115200 38400 0 115200serial 0 1 115200 38400 0 115200serial 2 1 115200 38400 0 115200serial 5 2 115200 115200 0 115200# ledled 0 0,0::C:0led 1 0,0::C:0led 2 0,0::C:0led 3 0,0::C:0led 4 0,0::C:0led 5 0,0::C:0led 6 0,0::C:0led 7 0,0::C:0led 8 0,0::C:0led 9 0,0::C:0led 10 0,0::C:0led 11 0,0::C:0led 12 0,0::C:0led 13 0,0::C:0led 14 0,0::C:0led 15 0,0::C:0led 16 0,0::C:0led 17 0,0::C:0led 18 0,0::C:0led 19 0,0::C:0led 20 0,0::C:0led 21 0,0::C:0led 22 0,0::C:0led 23 0,0::C:0led 24 0,0::C:0led 25 0,0::C:0led 26 0,0::C:0led 27 0,0::C:0led 28 0,0::C:0led 29 0,0::C:0led 30 0,0::C:0led 31 0,0::C:0# colorcolor 0 0,0,0color 1 0,255,255color 2 0,0,255color 3 30,0,255color 4 60,0,255color 5 90,0,255color 6 120,0,255color 7 150,0,255color 8 180,0,255color 9 210,0,255color 10 240,0,255color 11 270,0,255color 12 300,0,255color 13 330,0,255color 14 0,0,0color 15 0,0,0# mode_colormode_color 0 0 1mode_color 0 1 11mode_color 0 2 2mode_color 0 3 13mode_color 0 4 10mode_color 0 5 3mode_color 1 0 5mode_color 1 1 11mode_color 1 2 3mode_color 1 3 13mode_color 1 4 10mode_color 1 5 3mode_color 2 0 10mode_color 2 1 11mode_color 2 2 4mode_color 2 3 13mode_color 2 4 10mode_color 2 5 3mode_color 3 0 8mode_color 3 1 11mode_color 3 2 4mode_color 3 3 13mode_color 3 4 10mode_color 3 5 3mode_color 4 0 7mode_color 4 1 11mode_color 4 2 3mode_color 4 3 13mode_color 4 4 10mode_color 4 5 3mode_color 5 0 9mode_color 5 1 11mode_color 5 2 2mode_color 5 3 13mode_color 5 4 10mode_color 5 5 3mode_color 6 0 6mode_color 6 1 10mode_color 6 2 1mode_color 6 3 0mode_color 6 4 0mode_color 6 5 2mode_color 6 6 3mode_color 6 7 6mode_color 6 8 0mode_color 6 9 0mode_color 6 10 0# auxaux 0 0 1 1300 2100aux 1 1 0 1150 1700aux 2 8 0 1700 2100aux 3 27 2 1825 2100aux 4 18 0 1900 2100aux 5 0 0 900 900aux 6 0 0 900 900aux 7 0 0 900 900aux 8 0 0 900 900aux 9 0 0 900 900aux 10 0 0 900 900aux 11 0 0 900 900aux 12 0 0 900 900aux 13 0 0 900 900aux 14 0 0 900 900aux 15 0 0 900 900aux 16 0 0 900 900aux 17 0 0 900 900aux 18 0 0 900 900aux 19 0 0 900 900# adjrangeadjrange 0 0 0 900 900 0 0adjrange 1 0 0 900 900 0 0adjrange 2 0 0 900 900 0 0adjrange 3 0 0 900 900 0 0adjrange 4 0 0 900 900 0 0adjrange 5 0 0 900 900 0 0adjrange 6 0 0 900 900 0 0adjrange 7 0 0 900 900 0 0adjrange 8 0 0 900 900 0 0adjrange 9 0 0 900 900 0 0adjrange 10 0 0 900 900 0 0adjrange 11 0 0 900 900 0 0# rxrangerxrange 0 1000 2000rxrange 1 1000 2000rxrange 2 1000 2000rxrange 3 1000 2000# masterset looptime = 2000set gyro_sync = ONset gyro_sync_denom = 2set align_gyro = DEFAULTset gyro_hardware_lpf = 20HZset gyro_lpf_hz = 60set moron_threshold = 32set gyro_notch1_hz = 0set gyro_notch1_cutoff = 1set gyro_notch2_hz = 0set gyro_notch2_cutoff = 1set vbat_adc_channel = 2set rssi_adc_channel = 3set current_adc_channel = 1set airspeed_adc_channel = 0set acc_notch_hz = 0set acc_notch_cutoff = 1set align_acc = DEFAULTset acc_hardware = MPU6000set acc_lpf_hz = 15set acczero_x = 140set acczero_y = -31set acczero_z = -345set accgain_x = 4115set accgain_y = 4105set accgain_z = 4117set align_mag = DEFAULTset mag_hardware = NONEset mag_declination = 0set magzero_x = 0set magzero_y = 0set magzero_z = 0set mag_calibration_time = 30set baro_hardware = BMP280set baro_use_median_filter = ONset pitot_hardware = NONEset pitot_use_median_filter = ONset pitot_noise_lpf =  0.600set pitot_scale =  1.000set mid_rc = 1500set min_check = 1100set max_check = 1900set rssi_channel = 0set rssi_scale = 30set rssi_invert = OFFset rc_smoothing = ONset serialrx_provider = SPEK1024set sbus_inversion = ONset spektrum_sat_bind = 0set rx_min_usec = 885set rx_max_usec = 2115set serialrx_halfduplex = OFFset blackbox_rate_num = 1set blackbox_rate_denom = 0set blackbox_device = SERIALset sdcard_detect_inverted = ONset min_throttle = 1100set max_throttle = 1950set min_command = 1099set motor_pwm_rate = 400set motor_pwm_protocol = STANDARDset failsafe_delay = 5set failsafe_recovery_delay = 5set failsafe_off_delay = 200set failsafe_throttle = 1000set failsafe_throttle_low_delay = 100set failsafe_procedure = RTHset failsafe_stick_threshold = 50set fails [... cut ...]  = 100set nav_mc_vel_z_i = 50set nav_mc_vel_z_d = 10set nav_mc_pos_xy_p = 65set nav_mc_pos_xy_i = 120set nav_mc_pos_xy_d = 10set nav_mc_vel_xy_p = 180set nav_mc_vel_xy_i = 15set nav_mc_vel_xy_d = 100set nav_fw_pos_z_p = 50set nav_fw_pos_z_i = 0set nav_fw_pos_z_d = 0set nav_fw_pos_xy_p = 75set nav_fw_pos_xy_i = 5set nav_fw_pos_xy_d = 8set rc_expo = 0set rc_yaw_expo = 20set thr_mid = 50set thr_expo = 0set roll_rate = 11set pitch_rate = 14set yaw_rate = 20set tpa_rate = 33set tpa_breakpoint = 1300#

fiam, 2017-10-18:
@Jozx900 Please, next time paste your configuration as it is, not as a single line. It's a pain to read it like this.

You don't have AIRMODE enabled. Upgrade to 1.7.3 and type the following in the CLI:

```
feature AIRMODE
save
```

That should make the servos behave as you expect them.

Jozx900, 2017-10-19:
Sorry about that. I copy from my mobile phone.
# dump

# version
# INAV/OMNIBUSF4PRO 1.7.2 Jul 20 2017 / 11:59:41 (81b85de)

# resources

# mixer
mixer FLYING_WING

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2150 1480 110 -1 
servo 4 1000 2150 1555 -110 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature RX_PPM
feature VBAT
feature MOTOR_STOP
feature GPS
feature CURRENT_METER
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
[... 456 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 0
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 11
set pitch_rate = 14
set yaw_rate = 20
set tpa_rate = 33
set tpa_breakpoint = 1300

#

Jozx900, 2017-10-19:
Actually we don’t want this mix exist anymore.

digitalentity, 2017-10-19:
There is no mix between throttle and servos, however there IS a "mix" between motor vibration and servos throught the gyro and PID. Go fly. This is normal.

DzikuVx, 2017-12-07:
As well as a "mix" between throttle and Iterm that becomes active when throttle is up. I'd say everything is fine here, so closing

closed 2017-12-07 by DzikuVx