iNavFlight/inav #2357: Matek F405 OSD Home direction seems inverted

Pairan, 2017-10-17:
We've been out yesterday and the Home arrow was pointing into the wrong direction (felt inverted).

While flying directly to home (direction) the arrow pointed downwards allthough I would have assumed it to show "upwards".

can somebody confirm?

digitalentity, 2017-10-17:
Did you upload the OSD fonts in Configurator? Default fonts seem to come from Betaflight and are incompatible with INAV OSD.

Pairan, 2017-10-17:
nope ... just flashed 1.7.3 on, copied the setting via CLI and saved it! this time I didnt change the font.

Pairan, 2017-10-17:
Thanks, Pawel, as digitalentity wrote "font" I was tempted to check on that. So, okay ... will do so later and report in.

Pairan, 2017-10-18:
just a short heads up - i did pic the standard font from configurator and saved it. But the weather was unflyable. I will try again today

Pairan, 2017-10-18:
Did Check it today! The OSD works as it was supposed to- Home arrow shows the right direction after having flashed the standard font on, which came with the release!

Thanks @DzikuVx and @digitalentity

closed 2017-10-18 by Pairan